Koenkk/zigbee2mqtt #1555: Unable to start the latest dev

milakov, 2019-05-22:
Hi, I am getting this error in the loop with the latest dev branch:

```
[Info] Configuration file found in data path, but no backup file found in data path. Backing up existing configuration to /share/zigbee2mqtt/.configuration.yaml.bk
2019-05-22T10:29:49: PM2 log: Launching in no daemon mode
2019-05-22T10:29:49: PM2 log: App [npm:0] starting in -fork mode-
2019-05-22T10:29:49: PM2 log: App [npm:0] online
> zigbee2mqtt@1.4.0 start /app
> node index.js
  zigbee2mqtt:info 5/22/2019, 10:29:54 AM Logging to directory: '/share/zigbee2mqtt/log/2019-05-22.10-29-54'
  zigbee2mqtt:info 5/22/2019, 10:29:56 AM Starting zigbee2mqtt version 1.4.0 (commit #unknown)
  zigbee2mqtt:info 5/22/2019, 10:29:56 AM Starting zigbee-shepherd
/app/lib/zigbee.js:56
        this.shepherd = new ZShepherd(settings.get().serial.port, shepherdSettings);
                                                           ^
TypeError: Cannot read property 'port' of undefined
    at Zigbee.start (/app/lib/zigbee.js:56:60)
    at startupLogVersion (/app/lib/controller.js:163:25)
    at utils.getZigbee2mqttVersion (/app/lib/controller.js:189:13)
    at git.getLastCommit (/app/lib/util/utils.js:88:9)
    at /app/node_modules/git-last-commit/source/index.js:44:5
    at /app/node_modules/git-last-commit/source/index.js:14:4
    at ChildProcess.exithandler (child_process.js:301:5)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Socket.stream.socket.on (internal/child_process.js:381:11)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm 
ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-05-22T07_29_56_567Z-debug.log
2019-05-22T10:30:00: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]

```

It also made my z2m configuration file to contain settings defined in hass.io plugin only, no devices at all.

Koenkk, 2019-05-22:
Are you running 2 instances of zigbee2mqtt? One bare metal and one hassio?

milakov, 2019-05-22:
I am running just hass.io add-on. Could it be that https://github.com/danielwelch/hassio-zigbee2mqtt/pull/144 leaked into edge version of add-on?

Koenkk, 2019-05-22:
Yes that's in edge (https://github.com/danielwelch/hassio-zigbee2mqtt/commit/ec72af47b059a95cac3120140f9107cdb6b8b2de)

milakov, 2019-05-22:
Any guidance on how to migrate from old config file to new one?

danielwelch, 2019-05-22:
Sorry @milakov , I have pushed some fixes to `.travis.yml` and the distribution script that should prevent this from happening. The `schema.json` could serve as a guide to migrating, but my plan is to trigger a "re-build" of the master branch in travis to roll-back the accidental leak of the dev branch into the edge version.

milakov, 2019-05-22:
@danielwelch NP, thanks a lot!

danielwelch, 2019-05-22:
@milakov I manually triggered the last master build on travis, which I believe will re-push images created from master branch (https://travis-ci.org/danielwelch/hassio-zigbee2mqtt/builds/535462521)

Koenkk, 2019-05-22:
Thanks @danielwelch !

closed 2019-05-22 by Koenkk