Koenkk/zigbee2mqtt #1556: NOT ISSUE - Question

vexter0944, 2019-05-22:
# Bug Report
## What happened
In the OSRAM Lightify App - it allows you to set the default color of the bulb if I turn the switch off to the bulbs and turn the switch on (using normal power light switch).  Can we do similar with zigbee2mqtt somehow?

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2019-05-22:
Yes it's documented here: http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#osramsylvania-led-various

closed 2019-05-22 by Koenkk

vexter0944, 2019-05-22:
Thank you!  I'll look into that and see if I can get it going.

vexter0944, 2019-05-22:
OK - I looked at that - so is that a call my Home Assistant?  As in I want
them to come on to 100% brightness @ 5000k when power is applied via a
light switch power on.  So if I have a BR30 Osram bulb, do I just make a 1
time call to the bulb from like the dev console with the color/settings and
the "osram_remember_state": true and the it's done?  So I'd do this for
each bulb I assume.  Sorry for all the questions just trying to figure it
all out.

Thanks!

Shannon

Koenkk, 2019-05-22:
Yes, and you can send the commands with an mqtt client (e.g. mqtt.fx https://mqttfx.jensd.de/)

vexter0944, 2019-05-22:
Makes sense - I think I got it - thanks again!

fabioxp7980, 2020-02-14:
> Makes sense - I think I got it - thanks again!
> […](#)

Can you share how to publish mqtt message in a Mqtt client to have "osram_remember_state" working for this type of Osram light?

https://www.zigbee2mqtt.io/devices/AC03642.html