Koenkk/zigbee2mqtt #1557: Faster reaction time

drbios, 2019-05-22:
Hi
Experimenting with the latest dev and z-stack 3 on a cc2531 I notice that using the home assistant integrations, every time when lights or wall plugs are activated some how, zigbee2mqtt generates 3 or even 4 log messages. as a result of this behavior the "reaction time" is not "instantanius" but last night because a problem paring a bulb I try not using home assistant integration, instead i use direct home assistant configuration.yaml for every single dispositive in my house. Using this approach the "reaction time" is noticiable faster and my little raspberry pi 3b+ is more quiet, log messages are reduce to just 2 for every action and the CPU hits much less even when I use available timeout option with 10 seconds in the zigbee2mqtt configuration.
Talking about this. The problem with that bulb was that I try reset the light bulb, that was already in the network,  with zigbee2mqtt running. Strange behavior because first I succede removeing the bulb from the network but when I immediately turn on the bulb, zigbee2mqtt recognized again and integrates it to the network without repairing process as result of this ...zigbee2mqtt couldn't find the network path and cousing a error 205 or something. So to fix this:
1. Remove the bulb from the network
2. Turn off zigbee2mqtt service
3. Reset bulb procedure (turn on and off in a specific convination)
4. Ones the bulb is reset I turn on zigbee2mqtt service
5. Repair the device 
6. And Voila! ...perfectly working again

Sorry about my English 
Best regards from Chile

Koenkk, 2019-05-23:
Could you post the log of these 3 or 4 messages and describe what you expected? Could you also clarify which reaction time you are talking about?

drbios, 2019-05-23:
**_this is part of the log of 1 light bulb using home assistant integration (on / off action)_**

May 23 13:04:10 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:10 PM Zigbee publish to device '0x00158d000225dde6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
May 23 13:04:10 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:11 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:11 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:11 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:15 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:15 PM Zigbee publish to device '0x00158d000225dde6', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
May 23 13:04:15 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:15 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":13,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":13,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":13,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM Zigbee publish to device '0x00158d000225dde6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":13,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:16 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:17 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:17 PM Zigbee publish to device '0x00158d000225dde6', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
May 23 13:04:17 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:17 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:17 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:17 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":13,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:17 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:17 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":13,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:17 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:17 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"OFF","linkquality":13,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:18 hassbian npm[29965]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:04:18 PM Zigbee publish to device '0x00158d000225dde6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
[... cut ...]
May 23 13:10:46 hassbian npm[3478]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:10:46 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001c9affc', payload '{"state":"ON","linkquality":63,"brightness":254}'
May 23 13:10:47 hassbian npm[3478]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:10:47 PM Zigbee publish to device '0x00158d0001c9affc', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
May 23 13:10:48 hassbian npm[3478]: #033[32m  zigbee2mqtt:info#033[39m 5/23/2019, 1:10:48 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001c9affc', payload '{"state":"OFF","linkquality":63,"brightness":254}'

drbios, 2019-05-23:
as reaction time i mean, as example,  how long takes to a  light bulb to actually turn on or off after an action is trigerred.

Koenkk, 2019-05-23:
I don't understand why there would be a difference in reaction time when using discovery instead of manual configuration in `configuration.yaml`. This should do exactly the same thing, namely publishing to `zigbee2mqtt/0x00158d0001c9affc/set` `{"state": "ON"}`. 

Could you point where the delays are in the log?

drbios, 2019-05-23:
my theory is that this delay happens when zigbee2mqtt, usig the homeassistant integration,  writes 3 or 4 messeges in the log file, and that makes the reaction time a little slow in comparing using direct configutation in homeassistant configuration.yaml

these 3 or 4 messeges that  i mean are almost identical

`May 23 13:04:10 hassbian npm[29965]: #33[32m zigbee2mqtt:info#033[39m 5/23/2019, 1:04:10 PM Zigbee publish to device '0x00158d000225dde6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
May 23 13:04:10 hassbian npm[29965]: #33[32m zigbee2mqtt:info#033[39m 5/23/2019, 1:04:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:11 hassbian npm[29965]: #33[32m zigbee2mqtt:info#033[39m 5/23/2019, 1:04:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:11 hassbian npm[29965]: #33[32m zigbee2mqtt:info#033[39m 5/23/2019, 1:04:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'
May 23 13:04:11 hassbian npm[29965]: #33[32m zigbee2mqtt:info#033[39m 5/23/2019, 1:04:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000225dde6', payload '{"state":"ON","linkquality":15,"brightness":127,"color_temp":327,"color_mode":2,"color":{"x":0.431,"y":0.402}}'`

Koenkk, 2019-05-24:
Could you post the log with `log_level: debug`  (http://www.zigbee2mqtt.io/configuration/configuration.html) on pastebin.com?

drbios, 2019-05-29:
@Koenkk I am really sorry about this post ...I trully believe that found the problem of the "reaction time"... the guilty was raspberry wifi power management.
Z2M works really well but sometimes after a while when I trigger a wifi button that turns on a zigbee light the Mqtt server on the raspberry who connect to the network through the internal 5g wifi adapter was a little sleeppy because the wifi adapter in low power mode. when I turn off power management everything was flawless. Until now everything works very snappy

Koenkk, 2019-06-03:
@drbios sounds good, please close this if the problems are solved.

closed 2019-06-03 by drbios