Koenkk/zigbee2mqtt #1558: Xstic firmware

martin9115, 2019-05-23:
Hi. i use this kit https://www.digikey.com/product-detail/en/digi-international/XK-Z11-T-W/602-1598-ND/3482851 . I update the coordinator's firmware with XCTU, but when im trying to do the same with the Xstick, it gives me an error (it dont recognize the firmware  ). I've downloaded all the programs from here
https://www.digi.com/support/productdetail?pid=3564

Can someone help me to update the xstick firmware?

Koenkk, 2019-05-23:
How is this related to zigbee2mqtt?

martin9115, 2019-05-23:
Maybe not. If there is a more suitable place, forward it. And still. can you give me some advice?

Koenkk, 2019-05-23:
I don't have knowledge about XBee products. Closing as this doesnt seem to be related to zigbee2mqtt.

closed 2019-05-23 by Koenkk