Koenkk/zigbee2mqtt #1560: create custom device (5 Switch & 5 Relay) with mqtt discovery

AjitKIPL, 2019-05-23:
hi,
I am using hassio with zigbee2mqtt addon. 
Want to develop new device smart electric switchboard having 5 switch and 5 relays with ZigBee HA profile. 
Control device from home assistant and also update device switch state to home assistant. Also discover 5 switches when device join the network.
I am new to zigbee2mqtt, so please guide me to add this device in zigbee2mqtt hassio addon.

I follow TI0001 device in devices.js but it has only left and right buttons and it cant update device switch state when zigbee node device switch on/off. only change state when operate from hassio.
if I am wrong then tell me which frame need to send from zigbee node to coordinator to update on/off state.

please help me to go ahead.

Koenkk, 2019-05-23:
The only requirement for the device is that it exposes 5 endpoints. Each endpoint can then be controlled individually through home assisstant.

AjitKIPL, 2019-05-24:
hi,
1. actually I tried with 5 eps with modelid TI0001 but discovers only left and right buttons into hassio MQTT. 
So i think only one ep with cid = 'genLevelCtrl'; cmd: 'moveToLevelWithOnOff', level: state, transtime: channel, add postfix for switch1, 2, 3, 4, 5 and transtime used as switch number 1.0, 2.0, 3.0, 4.0 & 5.0.
In this case postfix are fixed so i add state = (Number)message.swnum; 
I can receive :- 01 01 04 108 1 0 for switch 1, 01 01 04 108 2 0 for switch 2, 01 01 04 108 3 0 for switch etc 
but not able to discover 4 switches into mqtt it shows left, right and linkquality.

in zigbee2mqtt addon i have not found which file have code that discovers device into MQTT broker.
Is i am doing correctly?

2. when i turn on switch manually on the router device(node) it's not change state of either right or left switch into hassio.
i tried to send cmd from zigbee router when switch toggle but fails. please tell me which cmd frame i want to send to reflect status into hassio. 

thanks in advance for your help

Koenkk, 2019-05-24:
You need to specify it like: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L460 and map the eps like https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L1113

AjitKIPL, 2019-05-24:
ok. 
I can change devices.js file added in /share/zigbee2mqtt/devices.js with "zigbee_shepherd_devices": true.
but I am not able to find homeassistant.js file in hassio-zigbee2mqtt addon to edit const mapping for 5 switches ep.
please let me know how to add homeassistant.js in hassio-zigbee2mqtt addon.

Koenkk, 2019-05-24:
This `homeassistant.js` file can indeed not easily be added. You could try to ask them to add this feature here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

Another option would be to officially support this device, but note that that requires a guide on how to build such a device.

AjitKIPL, 2019-05-24:
hi,
ok. i will include that issue in hassio-zigbee2mqtt.

Is it possible to add 'switch1' instead of 'bottom_left' postfix in devices.js model : DNCKATSW004? like below
ep: (device) => {
            return {'switch1': 1, 'switch1': 2, 'switch3': 3, 'switch4': 4};
        },
or postfixs are fixed non changeable.

Is device discover in mqtt broker is possible in homeassistant.js only? or can i place some code in devices.js.

Koenkk, 2019-05-24:
All postfixes need to be in this list: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/utils.js#L31

Discovery is only possible in `homeassistant.js`.

closed 2019-07-30 by stale[bot]