Koenkk/zigbee2mqtt #1561: No converter available for 'SN10ZW-1CO' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"欧瑞","modelId":"585fdfb8c2304119a2432e9845cf2623","powerSource":3}}'

realJayNay, 2019-05-23:
# Bug Report
## What happened
Trying to add support for Orvibo SN10ZW-1CO Motion Sensor (EndDevice), see https://www.orvibo.com/en/product/bodysensor.html.

## What did you expect to happen
Following the "how_to_support_new_devices" guide and creating an issue as mentioned in the logs.
The device joins, but uncertain how to proceed with creating a new converter.

## How to reproduce it (minimal and precise)
```xml
    // Orvibo                                                                                                            
    {                                                                                                                
        zigbeeModel: ['585fdfb8c2304119a2432e9845cf2623'],                                                                       
        model: 'SN10ZW-1CO',                                                                                         
        vendor: 'Orvibo',                                                                                           
        description: 'ZigBee HA motion sensor',                                                                                                                                                  
        supports: 'occupancy',                                                                             
        fromZigbee: [],                                                                                                           
        toZigbee: [],                                                                                              
    },
```

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version:

realJayNay, 2019-05-23:
When I trigger a motion event on the device, the following log occurs:

> [33m zigbee2mqtt:warn [39m 5/23/2019, 4:53:16 PM No converter available for 'SN10ZW-1CO' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"欧瑞","modelId":"585fdfb8c2304119a2432e9845cf2623","powerSource":3}}'
> [33m zigbee2mqtt:warn [39m 5/23/2019, 4:53:16 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2019-05-23:
Can you share your `database.db`?

realJayNay, 2019-05-24:
@Koenkk  - Sure, here you go: [database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/3215347/database.db.zip)

Koenkk, 2019-05-24:
Please try with:

```js
    // Orvibo                                                                                                            
    {                                                                                                                
        zigbeeModel: ['585fdfb8c2304119a2432e9845cf2623'],                                                                       
        model: 'SN10ZW-1CO',                                                                                         
        vendor: 'Orvibo',                                                                                           
        description: 'ZigBee HA motion sensor',                                                                                                                                                  
        supports: 'occupancy',                                                                             
        fromZigbee: [],                                                                                                           
        toZigbee: [],               
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),           ];
            execute(device, actions, callback);
        },                                                                               
    },
```

realJayNay, 2019-05-24:
@Koenkk - Same result, log.txt after a clean startup and database.db attached.

The device did not seem to join, the LED on the device did not indicate motion events. I pressed the pairing button again at 8:42:42 AM, after which I got the `No converter available` again.

Koenkk, 2019-05-26:
Could you:

- Remove device from `database.db`
- Restart zigbee2mqtt
- Re-pair device

tyler-vincent, 2019-07-24:
@realJayNay Did you get any further with this? I have an Orvibo motion sensor as well and I'd love to see it work.

realJayNay, 2019-09-23:
Hi @tyler-vincent, unfortunately not. Will comment here when I try again and when succesful.

coolguymatt, 2019-09-23:
I believe this was added in zigbee2mqtt 1.6. I added mine (it’s branded living wise but zigbee2mqtt sees orvibo) and it works fine

Koenkk, 2019-09-23:
@coolguymatt thanks!

closed 2019-09-23 by Koenkk