Koenkk/zigbee2mqtt #1562: No mqtt connection attempt

gempir, 2019-05-23:
I run zigbee2mqtt in a docker container from the offical image https://hub.docker.com/r/koenkk/zigbee2mqtt/

A few days ago I moved some stuff and redid the container. Now zigbee2mqtt won't connect to my mosquitto server. 

Thing is there isn't even an error about mqtt connection issue. I jumped into the container and pinged my mosquitto server it's available just fine. My other container running homeassistant connects just fine to the mosquitto server.

I had this exact same setup running before, but now suddendly the container won't do anything anymore.
Previously I had atleast some sort of error message in the logs. But these are my logs:

[image]

Only when stopping the container a single 
`zigbee2mqtt:error 5/23/2019, 7:48:51 PM Not connected to MQTT server! | stderr`
message shows up. Is there any way to tell why it won't connect to mqtt?

## Debug Info
zigbee2mqtt version:
[image]
CC253X firmware version:
https://github.com/Koenkk/Z-Stack-firmware/commit/e54e4edd22ed58ca5c9c783daa3a49535be6663d

Koenkk, 2019-05-24:
Can you post the complete log?

gempir, 2019-05-24:
This is the entire log of it starting.

Koenkk, 2019-05-24:
It doesn't show more then https://user-images.githubusercontent.com/9765622/58281986-c2ee0c80-7da4-11e9-85ec-4622186880e8.png?

gempir, 2019-05-24:
Correct. Had it running for hours but it never printed again until I restart it.

gempir, 2019-05-24:
I correct that* 

It did print this every 5 mins or so.
`zigbee2mqtt:debug 5/23/2019, 9:18:05 PM Saving state to file /app/data/state.json | stderr`

Koenkk, 2019-05-24:
The reason for this is that zigbee2mqtt never starts at all (zigbee-shepherd hangs it seems). You could try the troubleshooting from here: http://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html

gempir, 2019-05-24:
I went through all the steps with no success though. Something is detecting the device. Because my config says

```
serial:
  port: /dev/ttyACM0
  disable_led: true
```

and disable_led actually works already. I also tried changing the port, then it will actually log an error.

Koenkk, 2019-05-24:
Could you post the log when startingwith `DEBUG=* npm start`.

gempir, 2019-05-24:
Any idea how to do that easily when running docker?

Koenkk, 2019-05-24:
Set `DEBUG=*` as environment variable (https://docs.docker.com/compose/environment-variables/)

gempir, 2019-05-24:
Oh yea sorry, I'm stupid i was trying to change the entrypouint :D

This is the full startup log https://codepen.io/danielps1/full/dEmQMq

Koenkk, 2019-05-24:
Can you try reflashing the firmware?

gempir, 2019-05-24:
I did that 2 times already yesterday and the day before :/

gempir, 2019-05-25:
Okay I figured it out. Somehow the USB port was causing issues. Maybe because I plugged a usb stick into the port next to it but using the front USB port of my NAS (where I run zigbee2mqtt) fixed the issue.

I'll close this issue, @Koenkk thank you so much for responding quickly and helping me debug, that's really helpful! I love your project

closed 2019-05-25 by gempir