Koenkk/zigbee2mqtt #1563: CC2530 router connected when plugged in then says offline, does not work

Brandon4466, 2019-05-24:
Hello, I just followed the instructions on how to make and flash a CC2530+CC2591 to a router and everything went well. When I plug it in, I get the device incoming, paired, then the router information (model, etc.). But then nothing happens. When I look at the status it says offline, even in the network map. It is not working at all either, as the range is the exact same and the sensors even show in the network map that they are connecting directly to the coordinator.

Any ideas what this could be?

I am on Zigbee2MQTT version: 1.3.1
the coordinator version is: 20190223
and the router sw_version: Zigbee2Mqtt 1.3.1
Here is my network map:
[image]

I have already reflashed the router again (been using the -std version because i think that's the one im supposed to use?) but with no luck.

Thank you for any help! I am really at a loss on this one.
- Brandon

Ton1965, 2019-05-24:
Hi,
It looks to me the map shows the state incorrectly. I have the same problem with my router that I installed just yesterday. It is also shown as offline on the map but shows connected in HA (check corresponding binary sensor though I am not sure if it is available/correct in z2m 1.3.1).
Generally zigbee devices do not actively look for new routes. I managed to catch the moment yesterday when one of my sensors got connected to the router but in the moring today I noticed that all devices were connected directly to the coordinator again.
The only suspicious thing about your map is it show 0 link quality to your router. Mine shows ~80 and the number changes if the map is updated.
BTW, your map shows all devices as offline.

Brandon4466, 2019-05-24:
@Ton1965 Yeah, the other devices are door sensors so they always report as offline. But the router should report as online, shouldn't it? It also says disconnected in HA.

0 link quality on the network map, but in HA it says 144... It's really weird, but I know for sure it is not extending the coverage. I have a sensor right next to it that is out of range from the coordinator completely and it doesn't work, even when it is within half a foot from the router. I have tried repairing them to the router as well, but even if I completely unplug the CC2531 from the server, it will not pair to the router.

I don't know if it has to pair to it itself or if it will just re-route, but as of right now it is not re-routing I don't think :/

Ton1965, 2019-05-24:
@Brandon4466 
I believe pairing without the coordinator will not succeed anyway, the coordinator is at least needed to tell the router that incoming devices are allowed.
Currently one of my sensors is shown as connected through the router again but it is not the one closest to it at all. The closest ones are still shown as connected directly to the coordinator. Looks like they are happy with the link they currently have.

Brandon4466, 2019-05-24:
@Ton1965 Yeah, mine just shows the router is off on it's own and 0 link which is weird...

I have no idea.

Brandon4466, 2019-05-30:
Anybody have any idea? After a few days everything is still the exact same...

Filipe-Figueiredo, 2019-05-31:
Hello,
I have the same problem. The binary sensor is shown as offline. I installed the router yesterday, and I can't use it because of that

Ton1965, 2019-05-31:
Looks like zigbee map shows routers offline although they are connected and work. Mine was always shown offline but link quality got updated every minute and some devices migrated to it after a couple of days.

dr-apple, 2019-06-01:
At the new Dev Brand it works !!!

Brandon4466, 2019-06-05:
@dr-apple dev branch for the router or the coordinator?

dr-apple, 2019-06-05:
@Brandon4466 dev brand from the zigbee2mqtt software for Linux...

Brandon4466, 2019-06-06:
I just switched to the dev branch and still not working...

closed 2019-08-12 by stale[bot]

Swiftnesses, 2019-09-02:
Same issue here, using dev branch.