Koenkk/zigbee2mqtt #1564: [Question] Zigbee2Mqtt compatible UK wall socket

Woodpeckercz, 2019-05-24:
Hi guys,
Not sure where to post a question like this.
Does anyone know if there is a wall socket/outlet or preferably double outlet that works with zigbee2mqtt?

Thanks

james-fry, 2019-05-24:
You can see here:
https://www.zigbee2mqtt.io/information/supported_devices.html

Hive
Salus
Ikea tradfri might also work, but its not verified.

Woodpeckercz, 2019-05-24:
The thing is I want a built in socket not a smart plug, having a hard time finding one. The like lightwave but zigbee controlled not 433mhz.

james-fry, 2019-05-24:
Ah right. got you.
An alternative is a wifi socket that can be flashed with tasmota.
AFAIK there is only one: https://blakadder.github.io/templates/vigica_VGSPK00815.html
You can get them on amazon uk and they have mixed reviews.

timstanley1985, 2019-05-28:
I use sonoffs flashed with tasmota

Woodpeckercz, 2019-05-28:
Sonoff wall socket? Never saw one.
I'm after double in-wall sockets, not wall plugs.

closed 2019-08-04 by stale[bot]

james-fry, 2019-09-17:
> The thing is I want a built in socket not a smart plug

@Woodpeckercz - just found these:

https://aone.auroralighting.com/ProductDetails/AU-A1ZBDSS

not supported by Z2M... yet. But they are zigbee.