Koenkk/zigbee2mqtt #1566: Question: network stability and t-shooting

std10k, 2019-05-24:
I've been running z2m since roughly July 2018 without many issues until now. 
The old cc2531 firmware was fine (v 0.1 from memory or something like that - it wasn't reporting its version), the old version of z2m was fine too.  
I run in on a Pi, raspbian, no docker.  
I had about a dozen devices, Xiaomi buttons and body sensors, and a ziaomi switch. 
Then about a month ago i started growing the network on z2m and things started to turn pretty ugly. When Ikea finally brought Tradfri to my region i bought a bunch og light globes to automate what i couldn't automate before. I had to upgrade 2531 firmware to the Feb19 version, and z2m to the latest available as they wouldn't pair otherwise.

So now I have upwards of 40 devices, a bunch of new Ikea lights (and a couple of controllers and dimmers which are paired but not quite working yet),  Xiaomi switches (round and square ones, about a dozen), sensors (also about a dozen). 
The home is a 2 story townhouse, not too big. cc2531 could reach much farther than xiaomi hub from my observations, so range shouldn't really be an issue. 
As all light globes are routers, the network map is looking a bit Necronomicon. 

Stability is lost for good. Some stuff is working fine, some other stuff either not working or working intermittently, or with large delays (10-20 sec). 
I primarily use body sensors to turn lights on/off, switches to toggle scenes (1-2-3-4-5  and long press for off).  Sensors would not trigger when they clearly should, buttons work after 10-20 presses. Not fun at all. 

I suspect there is some mess going on in the network. Routers, as i understand, must join directly to the coordinator? if so, would they push each other out if there are more than 15 devices connected to the coordinator?  Would they conflict with xiaomi end devices (which i believe don't like switching hosts) for a 'slot' with the coordinator? 

I'm looking for t-shooting suggestions as i am pretty much on the verge of throwing all this stuff into a bin. 

I wouldn't want to have multiple native hubs (xiamoi proved somewhat unreliable + regional restriction on software are very annoying) as i suspect they'd conflict with each other on the radio. 

I'm thinking to move most lights out of z2m to the native Ikea hub (just got my hands on it today -IKEA was out of stock),  but ideally, i would like to understand the nature of the issues. I doubt it is HASS, seems like z2m or ZigBee network itself is going crazy.  Ah, tried adding 2530 router - it seems to make things even worse.

there are some random errors in the logs, but nothing too conclusive. 
5/24/2019, 5:18:25 AM - error: Zigbee publish to device '0xd0cf5efffe2c7525', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
5/24/2019, 5:18:25 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0xd0cf5efffe2c7525","type":"device","friendlyName":"0xd0cf5efffe2c7525"},"message":"{\"state\": \"ON\", \"brightness\": 255}"}}'

Koenkk, 2019-05-24:
I would recommend starting troubleshooting with the networkmap (http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgenetworkmap)

std10k, 2019-05-24:
Thanks Koenkk , i have the map working with the plugin. Should have attached it to the first post, but here anyway. Actually looking through some other issues here, my map doesn't look particularly bad. 
Could i be just facing 1536?  

BTW, thanks for this awesome project. despite some issues it is a great piece of software and i admire the amount of efforts you're putting in making it happen. 

[image]

Koenkk, 2019-05-24:
Do you have any reliability issues in controlling devices? (e.g. turn on bulbs)

std10k, 2019-05-24:
Yes, sometimes a light would not respond to any commands (be it via button or hass app/web gui directly) but then sudently (about 20-30 sec later) would start executing them all at once, and then may go unresponsive again. Not all lights, and not always. Buttons and sensors seem to be more affected. Usually, when sensors and buttons won't turn the lights on/off i still can control the lights via hass app/gui.  Buttons usually start working after ~30-60 sec and/or multiple presses, but quite unpredictable. 

It it were a normal IP network i'd suspect some kind of loop or broadcast storm. this is why i suspect that a large number of routers may have a role in all this. but i know little about zigbee.

Koenkk, 2019-05-24:
Which lights are having this 30 sec delay issue? I have the same with some chinese led controllers (Gledopto, currently investigating why it happens) but never experienced it with TRADFRI.

std10k, 2019-05-24:
All lights are Ikea tradfri. The coloured ones and the ones with cool/warm light seem to be more affected. I honestly don't think the light make matters (i know ikea work fine), it does feel like some kind of network issue (though can't quite prove it).  Could it be something wrong with PI's USB? i read it can get a bit funny. 
It would probably make sense to re-pair all my stuff as i've been running since old hass and z2m versions, but it is verytime consuming and if i were to re-pair it i'd want to make sure the network topology is OK.

Other thing i noticed that is since i added ikea lights the response time from button push or sensor trigger to the point of time when lights turn on has increased.  Previously the light would turn on almost instantly, now i see ~2 sec delay after i enter an area with a xiaomi sensor it it before the light turns on (that is in areas where everything is generally OK). I've had one light in the corridor since the beginning and it didn't change (neither hass nor zigbee part), this particular light is controleld by broadlink swtitch (so no zigbee there) so only the motion sensor is zigbee - the delay must be in the sensor.

mabi, 2019-05-24:
I am having a similar problem since I replaced a dimmable white OSRAM Smart+ bulb with a multicolor OSRAM Smart+ bulb. I get lost or delayed burst commands with a lot of 233 errors in the log. I noticed that after a successful command is sent, I get a lot of status messages with the same info from the bulb.

johnnyletrois, 2019-05-25:
I had recurring trouble when I had ikea outlets.  Removed the outlets and everything is super stable.

std10k, 2019-05-26:
Well, i got the Ikea gateway and moved a few light globes, dimmer and remote to it from 2531. 
Somehow things got even worse - virtually nothing on z2m is working, though it seems to be online. 

I'm still looking into it (may not have much time as i'm going on a business trip tomorrow), but i noticed these messages in the logs - usually they appear (form my subjective observations) when the Xiaomi switches are not working; once the switches start to work these messages seem to disappear or become less frequent.  I also (to make things messier i guess) updated the firmware to the latest dev with re0enables source routing - it worked without any noticeable difference until i started removing ikea globes. 

5/26/2019, 12:31:27 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.plug' (0x00158d000117b87d) with groupID undefined of endpoint 1
5/26/2019, 12:31:27 PM - debug: Skipping re-transmitted Xiaomi message
5/26/2019, 12:31:30 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.plug' (0x00158d000117b87d) with groupID undefined of endpoint 1
5/26/2019, 12:31:30 PM - debug: Skipping re-transmitted Xiaomi message
5/26/2019, 12:31:30 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.plug' (0x00158d000117b87d) of endpoint 1
5/26/2019, 12:31:44 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.plug' (0x00158d000117b87d) with groupID undefined of endpoint 1
5/26/2019, 12:31:44 PM - debug: Skipping re-transmitted Xiaomi message
5/26/2019, 12:31:44 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.plug' (0x00158d000117b87d) with groupID undefined of endpoint 1
5/26/2019, 12:31:44 PM - debug: Skipping re-transmitted Xiaomi message
5/26/2019, 12:31:44 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.plug' (0x00158d000117b87d) of endpoint 1

Koenkk, 2019-05-27:
The ikea bulbs act as routers, when e.g. a Xiaomi devices is connected to this router it needs to find a new parent (but this can take some time for Xiaomi end devices in my experience).

std10k, 2019-06-02:
OK, some updates on my issue (sorry i was away from home for a few days).

After removing most of ikea light globes, simmer and remote control, things seem to have improved substantially. 

At first i wasn't getting any response from most of Xiaomi end devices - i think it had indeed something to do with the issue with Xiaomi not switching between routers very well. I'd love to understand more about this, e.g. if manual re-pairing is advisable etc. But eventually, they seem to have settled (that is with cc2530 router also removed from the picture). I kept a couple of Tradfri globes in remote parts of the house for netwrok extension for now, they seem to be fine. 

The lags also seem to have mostly disappeared, i.e. lights turn on much faster when i enter an area with a motion sensor (used to take up to 5 sec). 

Ikea globes are working via IKEA gateway quite well so far, and as a bonus, i can actually use IKEA dimmers and controller now, though in somewhat limited, while still being able to control the scenes via HASS. 

It all feels like i was having some kind of loop in the network, or a buggy router that was messing things up. But can't confirm without having any logs. 
New network topology is pretty much just a star, can't see any full-mesh anymore (which is what i think was messing things up for me)

I the previous screenshot a few posts above, there are four double links, between 3 tradfri globes and the coordinator. Is that sort of picture normal? 

As a conclusion, i'm going to keep using Tradfri gateway but probably will keep most of Xiaomi devices on z2m as I am not a big fan if Xiaomi software (and z2m seems a bit more stable than xiaomi hub and range feels better too even with just 2531)

[image]

rickkdotnet, 2019-06-03:
I'm kinda in the same boat. 40+ devices. Tradfri, Xaiomi, Gledopto and Zemismart.

I'm starting to narrow most of my issues down to the Tradfri bulbs:

- Xaoimi sensors would dissapear after a few hours when routed through the Tradfri bulbs, needed re-pairing to get back online. Disconnected all Tradfri bulbs in the house, re-paired them and they have been online ever since. 
- I've lost 2 out of 3 Gledopto RGBW drivers once, and couldn't get to re-pair them with the bulbs disconnected. Disconnected the bulbs and they re-paired straightaway. 
- I've got 4 tradfri bulbs, 1 E14 and 3xE27. Each of the E27 bulbs randomly stops responding to zigbee2mqtt after a few days. A z2m restart fixes that. 
- last night one bulb also stopped responding to it's remote, but was still controllable from z2m. The  remote was also still connected to z2m, I could see the commands still coming in. I re-paired it to the bulb and it has been ok since. 
- I haven't seen issues with the tradfri transformers, outlets and motion sensors (yet)

Other observations: 
- The network map is sometimes a star, and at other times a full mesh. I don't know why and how it relates to stability yet. 
- Sending commands to HASS groups seems to destabilize things quickly. Either the stick crashes or some devices become unavailable after sending the commands. If that happens, most devices come back by themselves after a few minutes, but I think the bulbs don't.

daxy01, 2019-06-03:
It seems the combination Tradfi and CC2531_20190425 is not the best match. I have had the same issue with losing connections. Somehow it seems (seems, no proof) that the Tradfi is interfering somehow. I've tried the older 20190315 (max_stability) version and all is rock-solid again, even without repairing. Perhaps worth a try: 
https://github.com/Koenkk/Z-Stack-firmware/blob/c36ba37ebb10690245ba29d1360a21b5abfa9c43/coordinator/Z-Stack_Home_1.2/max_stability/CC2531/CC2531_MAX_STABILITY_20190315.zip

It's not ideal as you can't upgrade anymore unless it will be addressed, but it will give you a quick fix for now. Let me know if it works in your network as well, just to build some facts together.

std10k, 2019-06-03:
I had the problem with February firmware. Actually, speaking of groups, i think it may have something to do with either groups of binding. 
I had one tradfri dimmer bound with one globe, and once i've done it z2m went down spectacularly: took about half an hour and a dozen of reboots to bring it back up.  The binding worked for a few seconds before the crash (i.e. the light responded to the dimmer), and then never again. 
I also had a group set up which never worked. Did everything by the manual, created a group and joined devices successfully, but tradfri remote still couldn't control the light in any way. I wasn't 100% sure if i just got the group ID wrong (field in wireshark where i took the number from had a different name, but everything else looked the same as in docs) or it was z2m.
Unfortunately i also did all this at about the same time as when i added tradfri lights, so skewed the sequence of events.

rickkdotnet, 2019-06-03:
> It's not ideal as you can't upgrade anymore unless it will be addressed, but it will give you a quick fix for now. Let me know if it works in your network as well, just to build some facts together.

Thanks, but I'm on the max_stability since last friday. I was on 20190223 until then, when a couple of group commands killed the firmware ( https://github.com/Koenkk/zigbee2mqtt/issues/1284#issuecomment-497842752 ).

I tried the latest recommended for a few hours but I started losing devices quickly, so I went to max_stability and that seems to be at least as stable as 20190223.

On the plus side, the tradfri group stuff seems to be working for me.

Koenkk, 2019-06-03:
@rickkdotnet what do you mean with losing devices?

rickkdotnet, 2019-06-03:
@Koenkk good question.  I've been using the term loosely, I'll try to be more specific in the future. 

In the case of the 20190425 firmware it meant the devices stopped responding to (hass) commands until z2m was restarted.

If I look back in the logs I think the corresponding errors are:
PM - error: Failed to bind for reporting for 0x90fd9ffffe2a57d4 - genOnOff - 1 - (Error: Timed out after 10000 ms)
5/31/2019, 10:15:14 PM - warn: Failed to configure woonkamer_lantaarnpaal_licht (0x000d6ffffee390e8) ('Error: Timed out after 10000 ms') (attempt
 #1)

(multiple occurences of these errors on multiple devices) 

I'm used to seeing these for the remotes but now there were lights in there too.

Koenkk, 2019-06-03:
@rickkdotnet what error do you get when controlling the light? (turn on or off)

rickkdotnet, 2019-06-03:
@Koenkk hmm interesting.. when I track back through the logs I don't see any "Zigbee publish" messages to the failed devices on 20190425. It just looks like z2m doesn't receive any commands, so maybe something else went wrong, perhaps higher in the stack while on that firmware? I'll do some proper testing soon. 

For what it's worth, this morning, on the 20190315 firmware, all 3 E27 bulbs were unresponsive and it looked like this: 

6/3/2019, 4:55:01 AM - error: Zigbee publish to device '0x000d6ffffe188ab2', lightingColorCtrl - moveToColor - {"transtime":3000,"colorx":22302,"colory":21469} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
6/3/2019, 5:02:49 AM - error: Zigbee publish to device '0x000d6ffffe0d229c', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
]6/2/2019, 5:15:22 PM - error: Zigbee publish to device '0x000d6ffffe179127', lightingColorCtrl - moveToColor - {"transtime":10,"colorx":17642,"colory":6645} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.

rickkdotnet, 2019-06-03:
Came home tonight and noticed both the Xaomi motion sensor and temperature sensor in the kitchen stopped reporting. Even if I press the buttons, long or short, the logs remain completely silent until I delete them from the database, restart z2m and re-pair them again.
 
After restarting I notice the RGB controller in the same area -which worked fine minutes before- stopped responding to hass commands. I restart z2m and it works again. 

Logs show: 
6/3/2019, 6:32:02 PM - info: Device incoming...
6/3/2019, 6:32:03 PM - error: Failed to setup reporting for 0x00124b001d04db08 - genOnOff - 1 - (Error: AF data request fails, status code: 233. MAC no ack.)
6/3/2019, 6:37:02 PM - info: MQTT publish: topic 'zigbee2mqtt/keuken_ledstrip_licht/availability', payload 'online'
6/3/2019, 6:52:13 PM - error: Failed to ping 0x00124b001d04db08
6/3/2019, 6:52:13 PM - info: MQTT publish: topic 'zigbee2mqtt/keuken_ledstrip_licht/availability', payload 'offline'

And after a restart of z2m it's back immediately, as if nothing happened. 

I'm at a loss how to troubleshoot this. I'm hoping my second C2531 comes soon, so I can do some sniffing.

rickkdotnet, 2019-06-03:
Ok, interesting night tonight. I couldn't resist and tried flashing the "special" 20190315 build from the https://github.com/Koenkk/zigbee2mqtt/issues/1284 thread. After the flash and a reboot I was receiving status updates from the various devices just fine, but couldn't control anything. 

I either didn't see anything at all in the log, or at most this: 
6/3/2019, 8:24:56 PM - debug: Received MQTT message on 'zigbee2mqtt/woonkamer_smiley_licht/set' with data 'ON
6/3/2019, 8:25:08 PM - error: Failed to bind for reporting for 0x90fd9ffffe2a57d4 - genOnOff - 1 - (Error: Timed out after 10000 ms)

I've attached the file here. 
[20190315-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3249519/20190315-log.txt)

After that I went to 20190425 which behaved exactly the same. I tried powercycling a few devices to no avail, and then -desperately- I just flipped the breaker and powercycled about half of the devices. 
Lo and behold, after a few minutes everything started responding again. 

To top it off I added a tradfri outlet close to the coordinator and suddenly I'm getting non-zero link quality reports from a number of devices again. 

This really makes me think I have a misbehaving device somewhere in there.

rickkdotnet, 2019-06-04:
Tried to isolate a problematic device to no avail. As for the linkquality:0 reports, it seems that they are generated by the devices directly connected to the coordinator. The tradfri outlet close to the router reports a linkquality of 80 however. 

Left it overnight, lots of flapping availabilities, including the outlet close to the router, and in the morning the networkmap was back to a star topology with random devices not responding with error 205.  No network route.
 
After a minute or two they respond again, for a while. This is on the 20190425 firmware. 

Maybe there is actually something fried in the stick relating to the radio, although the overall stability is the same as before, on the februari firmware and without the linkquality:0 reports.

Koenkk, 2019-06-04:
Can you try this firmware? https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/CC2531_20190523.zip

rickkdotnet, 2019-06-04:
Sure. 
FWIW, I already reverted to 20190315 and that seemed to have solved the error 205, the availability flapping and the networkmap is a nice mesh again. I'll give the 20190523 a go now.

rickkdotnet, 2019-06-04:
I'm on the 20190523 for the last hour now. First impressions:
- it takes A LOT of time for the network to reconverge, and I had to power off a lot of devices before z2m would be able to control anything.
- The networkmap is stil a star
On the plus side: 
- Haven't seen error 205 nor availability flaps for devices once they have finally joined

I'll give it some more time.

rickkdotnet, 2019-06-04:
@Koenkk still on 20190523: 
- I lost two Xaiomi motion sensors already (go silent until re-pairing), like on the old fimrware. I guess this is #1408 and this firmware doesn't change that behavior. (sidenote: the door sensors have been rock solid so far, unperturbed by any change)   
- after restarting z2m, it subjectively takes longer before all devices are back under control, a few minutes of "Device incoming..." messages before everything reacts to commands. I don't know if this makes sense. 
- the networkmap remains a star 
- some devices report linkquality 0, some don't. Haven't found a pattern here
- apart from the Xaiomi sensors everything remains connected and responsive, no availability timeouts
- zero 205 errors so far, so I guess this fw does fix #1536

Koenkk, 2019-06-04:
- Besides the motion and door sensor, do you have more Xiaomi sensors?
- Can you move the motion sensors closer to the coordinator and see if they also unpair?
- How big is your network?

rickkdotnet, 2019-06-04:
@Koenkk 
- I have 5 door sensors, 5 motion sensors and 3 temp/humidity/pressure sensors. All Aqara. The temp and motion sensors both tend to dissappear.
- There is a motion sensor close to the coordinator, no issues with that
- 39 devices , about half of them endpoints

Koenkk, 2019-06-05:
Do you have OSRAM plugs?

rickkdotnet, 2019-06-05:
Hi, no, no OSRAM plugs. These are my devices: 

      6 {"mfg":"LUMI","model":"lumi.sensor_magnet.aq2"}
      5 {"mfg":"LUMI","model":"lumi.sensor_motion.aq2"}
      3 {"mfg":"LUMI","model":"lumi.weather"}
      3 {"mfg":"IKEA of Sweden","model":"TRADFRI motion sensor"}
      3 {"mfg":"IKEA of Sweden","model":"TRADFRI control outlet"}
      2 {"mfg":"IKEA of Sweden","model":"TRADFRI transformer 10W"}
      2 {"mfg":"IKEA of Sweden","model":"TRADFRI remote control"}
      2 {"mfg":"IKEA of Sweden","model":"TRADFRI on/off switch"}
      2 {"mfg":"IKEA of Sweden","model":"TRADFRI bulb E27 CWS opal 600lm"}
      2 {"mfg":"GLEDOPTO","model":"GL-C-007"}
      1 {"mfg":"SZ","model":"Lamp_01"}
      1 {"mfg":"LUMI","model":"lumi.remote.b286acn01\u0000\u0000\u0000"}
      1 {"mfg":"LUMI","model":"lumi.remote.b186acn01\u0000\u0000\u0000"}
      1 {"mfg":"IKEA of Sweden","model":"TRADFRI wireless dimmer"}
      1 {"mfg":"IKEA of Sweden","model":"TRADFRI bulb E27 WS clear 950lm"}
      1 {"mfg":"IKEA of Sweden","model":"TRADFRI bulb E14 W op/ch 400lm"}
      1 {"mfg":"icasa","model":"ICZB-IW11D"}
      1 {"mfg":"GLEDOPTO","model":"GL-C-009"}
      1 {"mfg":"3A Smart Home DE","model":"LXN56-DC27LX1.1"}

For what it's worth: Setup seems very stable on this firmware, apart from one lost xaiomi motion sensor this morning. I think it went AWOL after a restart of z2m. It's very well possible that (looking at last_seen) z2m was restarting just when it wanted to send it's hourly report.

Koenkk, 2019-06-05:
@rickkdotnet so when pressing the reset button on the motion sensor, does it report something?

Smiggel, 2019-06-05:
I have a reliability issue that I posted in an other topic:
Ok, I was busy with switching all my Zigbee devices to Hass.io, coming from Domoticz. I setup the Zigbee map and noticed that all my sensors did no not communicate with my two routers. So, I thought I would connect my 4 sensors on the ground floor to the router on the ground floor by repairing them close to the stick.

I noticed after doing that, my sensors connected to the router. I could see it on the map. However, I also noticed that 1 of my door sensors and PIR sensors signal did not come through to the coordinator at all. The link quality of the router is 50+. The link quality to my sensors was also above 20. But the signal of opening and closing a door does not come through at all. I had to repair it close the the coordinator in order to make it work.

I use the 2018-09 router firmware. My Zigbee2Mqtt version on Hass.io is 1.4.0. And my router firmware is CC2531_20190523.

Koenkk, 2019-06-07:
@rickdotnet let's keep things in this issue, can you post the log of 3b on pastebin.com and link here? Please also clearly indicate where the problem can be seen.

rickkdotnet, 2019-06-07:
Ok, I've grepped out the lines corresponding to this outlet, and added some comments to: 
https://pastebin.com/7j0xC2dR

I don't seen anything else that's relevant in the logs, but If you want to full log, it's here: https://we.tl/t-fHFgF2CJqm

Koenkk, 2019-06-07:
@rickkdotnet I see a lot of `Starting network scan...`, probably something might go wrong when the device doesn't respond sometime and the job keeps hanging in the queue -> no future jobs are executed.

Do you also have this problem when **never** requesting the networkmap?

rickkdotnet, 2019-06-07:
@Koenkk I don't know, the issue (fortunately) happens too infrequently to say for sure, but I'll keep an eye on it in relation to the networkmap scans and let you know. Extra logging would be good I think. 

Thanks for your help.

Koenkk, 2019-06-07:
Good, a bit offtopic but as you were having issues with `20190425` firmware, can you try the `20190608` firmware? It won't fix this issue, but I need some users who have problems with `20190425` to confirm that they are fixed in `20190608`  (See https://github.com/Koenkk/Z-Stack-firmware/issues/86#issuecomment-500056683)

rickkdotnet, 2019-06-08:
@Koenkk : sure, I've flashed it about an hour ago. It took a while before devices started to rejoin ands. So far things are unstable. 205 errors, availability timeouts. Star topology in network map. 

But I'll give it some more time, because it's always messy after a flash. Why would it take so long for the network to reconverge after a flash? Is the routing table stored on the stick?

Koenkk, 2019-06-08:
Yes routing tables are stored on the stick, if devices still give 205 error, disconnect them from mains power for a few seconds.

rickkdotnet, 2019-06-08:
@Koenkk ok, thanks, that makes sense. 

I tried to stabilize the network on 20190608 by unplugging, bringing things back up slowly etc, but I gave up. In 4 hours it never stabilized. Behavior was similar to 20190425 in my book.

I'm now back on 20190315 and everything was back online and stable within 15 minutes.

Koenkk, 2019-06-08:
Why did you switch back to `20190315`, isn't `20190523` working properly for you?

Koenkk, 2019-06-08:
I've added the required logging to debug https://github.com/Koenkk/zigbee2mqtt/issues/1566#issuecomment-499894692 in the dev branch. Please set `log_level: trace` in `configuration.yaml`.

rickkdotnet, 2019-06-08:
@Koenkk: To see if it would behave different from 20190523. So far, I'd say they behave the same. 

I'll try the dev branch with trace logging.

std10k, 2019-06-09:
I've been off the tread for a while.  My issues mostly disappeared after i removed most of tradfri lights from z2m network, but not completely.  I have one Xiaomi zigbee plug which acts as a router and two Tradfri lights in different parts of the house; there should be no other routers. 
I noticed that from time to time Xiaomi switches (buttons) would not work for about 30-60 seconds (subjectively) when I press them for the first time after not using them for a while. Same probably happens with motion sensors, but they are somewhat harder to notice as they obviously trigger on their own when there are people in the house. I.e. with motion sensors, the lights that they are supposed to turn on would not turn on sometimes, and then it works fine again. 

When the buttons do not work it is super annoying as i either have to use the hard switches to turn the lights on (and they are under smart switches that are stuck to them so quite difficult to reach) or i have to wait for those 30-60 seconds near the switch....

I captured some logs from the time when a Xiaomi button was not working (and started to work later).
This button is near the Xiaomi smart plug( router) and controls two lights via hass: 1 ikea (on ikea gateway) and another one connected to this Xiaomi plug.  When the button doesn't work, i mean it won't turn either of the two lights on or off, so must be a button problem, not a plug problem. Unless it pairs to the plug, which i suspect may be the case. 

I'm unsure how to interpret the logs. Any clues? 

0x00158d0002109f86  is my xiaomi button (round one),  0x00158d000117b87d is the Xiaomi plug (router) near that button.  Also attached the new network map. 
I noticed this error. 

error: Zigbee publish to device '0x00158d000117b87d', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 17

`
root@raspberrypi:/opt/zigbee2mqtt/data/log/2019-06-02.11-40-42# tail -f log1.txt
6/9/2019, 1:34:08 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002109f86', payload '{"linkquality":42,"duration":2493,"battery":100,"voltage":3002,"click":"single"}'
6/9/2019, 1:34:08 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch' (0x00158d0002109f86) of endpoint 1
6/9/2019, 1:34:09 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d000117b87d/set' with data 'ON'
6/9/2019, 1:34:09 PM - info: Zigbee publish to device '0x00158d000117b87d', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
6/9/2019, 1:34:09 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000117b87d', payload '{"state":"ON","power":5.77,"voltage":null,"consumption":7.18,"temperature":36,"linkquality":47}'
6/9/2019, 1:34:09 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"61440":117440725,"onOff":1}}' of device 'lumi.plug' (0x00158d000117b87d) of endpoint 1
6/9/2019, 1:34:09 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000117b87d', payload '{"state":"ON","power":5.77,"voltage":null,"consumption":7.18,"temperature":36,"linkquality":44}'
6/9/2019, 1:34:09 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"61440":117440725}}' of device 'lumi.plug' (0x00158d000117b87d) of endpoint 1
6/9/2019, 1:34:09 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.plug' (0x00158d000117b87d) of endpoint 1
6/9/2019, 1:34:09 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000117b87d', payload '{"state":"ON","power":5.77,"voltage":null,"consumption":7.18,"temperature":36,"linkquality":47}'
6/9/2019, 1:34:38 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.plug' (0x00158d000117b87d) with groupID undefined of e                                                                                                             ndpoint 1
6/9/2019, 1:34:38 PM - debug: Skipping re-transmitted Xiaomi message
6/9/2019, 1:34:38 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.sensor_motion' (0x00158d0001dc1365) of endpoint 1
6/9/2019, 1:34:38 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001dc1365', payload '{"occupancy":true,"linkquality":44,"battery":97,"voltage":2995}'
6/9/2019, 1:34:38 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_switch' (0x00158d0002109f86) of endpoint 1
6/9/2019, 1:34:38 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_switch' (0x00158d0002109f86) of endpoint 1
6/9/2019, 1:34:38 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.plug' (0x00158d000117b87d) with groupID undefined of endpoint 1
6/9/2019, 1:34:38 PM - debug: Skipping re-transmitted Xiaomi message
6/9/2019, 1:34:38 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.plug' (0x00158d000117b87d) of endpoint 1
6/9/2019, 1:34:39 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002109f86', payload '{"linkquality":44,"duration":2493,"battery":100,"voltage":3002,"click":"long"}'
[... cut ...]
6/9/2019, 1:35:18 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000117b87d', payload '{"state":"ON","power":5.77,"voltage":null,"consumption":7.18,"temperature":36,"linkquality":42}'
6/9/2019, 1:35:18 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genAnalogInput","data":{"presentValue":5.769999980926514}}' of device 'lumi.plug' (0x00158d000117b87d) of endpoint                                                                                                              2
`

[image]

Koenkk, 2019-06-10:
@rickkdotnet  @std10k can you try this firmware (I've now also made a source routing firmware available: 

@all/@nob0dy80 can you try this firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing (`CC2531_SOURCE_ROUTING_20190610.zip`). It also has source routing enabled like the max stability firmware.

rickkdotnet, 2019-06-10:
@Koenkk sure, just flashed this version. First impressions after 20 minutes: similar to 20190608 and 20190425: random 205 errors and availability timeouts... but there are still devices joining, I'll give it another hour.

rickkdotnet, 2019-06-10:
@Koenkk seems fine now

Koenkk, 2019-06-10:
@rickkdotnet great, could you monitor it for a few days to check if any issues arise?

rickkdotnet, 2019-06-10:
@Koenkk Yes I will keep you posted. So far it's fine and even the networkmap looks nice.

Koenkk, 2019-06-10:
@rickkdotnet are you on the dev branch or zigbee2mqtt 1.4?

rickkdotnet, 2019-06-10:
Dev, because of https://github.com/Koenkk/zigbee2mqtt/issues/1566#issuecomment-500145937 .

Oh, I just noticed I cannot switch to log_level 'trace' via MQTT ( PM Could not set log level to 'trace'. Allowed level: 'error,warn,info,debug' )

Koenkk, 2019-06-10:
Ok great, i will fix setting log level via MQTT, for now set it via configuration.yaml

rickkdotnet, 2019-06-11:
> Ok great, i will fix setting log level via MQTT, for now set it via configuration.yaml

Yes, that would be nice. I'm logging to RAM and trace fills up the disk fast. Would be nice if I could switch it on demand, when an issue arises.  

As for the firmware... I've lost a gledopto controller overnight, availability timeouts and error 205. 
I could still control it from the remote bound to the group and still received status changes at z2m. 

https://pastebin.com/WkyyiCJL

z2m restart didn't fix it.  powercycle didn't fix it. Deleted it from the database and tried to reset the controller (didn't work) and it came back somewhere in between. 

Will keep an eye on it.

rickkdotnet, 2019-06-11:
Ok, as for the https://github.com/Koenkk/zigbee2mqtt/issues/1566#issuecomment-500145937, I'm monitoring last_seen and noticed I lost a motion sensor. Incidentically the one right next to coordinator, which never gave any problems until now. 

Per your request I (short) pressed the button and got this in the logs: 

```
6/11/2019,` 7:21:55 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d000313e6e5"' of device 'lumi.sensor_motion.aq2' (0x00158d000313e6e5) of endpoint 1
6/11/2019, 7:21:55 PM - trace: Checking if 0x00158d000313e6e5 is pingable
6/11/2019, 7:21:55 PM - trace: 0x00158d000313e6e5 is pingable (false) not router or battery powered

```
And nothing happens. If I press the button again the log message is *not* repeated. 

If I *long* press the button (i.e: initiate pairing) the message *is* repeated, but nothing happens after that.

After that I zigbee2mqtt/bridge/config/remove 'd the sensor, long pressed the button, the sensor joined again. 

The last messages of the sensor were: 
```
6/11/2019, 11:09:05 AM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3005,"3":29,"4":17320,"5":17,"6":[0,8],"10":53509,"11":38,"100":0}}}' of device 'lumi.sensor_motion.aq2' (0x00158d000313e6e5) of endpoint 1
6/11/2019, 11:09:05 AM - trace: Checking if 0x00158d000313e6e5 is pingable
6/11/2019, 11:09:05 AM - trace: 0x00158d000313e6e5 is pingable (false) not router or battery powered
6/11/2019, 11:09:05 AM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3005,null,29,17320,17,[0,8],null,null,null,53509,38,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}' of device 'lumi.sensor_motion.aq2' (0x00158d000313e6e5) of endpoint 1
6/11/2019, 11:09:06 AM - trace: Checking if 0x00158d000313e6e5 is pingable
6/11/2019, 11:09:06 AM - trace: 0x00158d000313e6e5 is pingable (false) not router or battery powered
6/11/2019, 11:09:06 AM - info: MQTT publish: topic 'zigbee2mqtt/gang-ms', payload '{"illuminance":38,"linkquality":84,"last_seen":"2019-06-11T11:09:05+00:00","occupancy":true,"battery":100,"voltage":3005}'
6/11/2019, 11:09:33 AM - info: MQTT publish: topic 'zigbee2mqtt/gang-ms', payload '{"illuminance":38,"linkquality":84,"last_seen":"2019-06-11T11:09:32+00:00","occupancy":false,"battery":100,"voltage":3005}'
```

Hope this gives some clue as to what's happening.

Koenkk, 2019-06-12:
`endDeviceAnnce` that the device is still in the network but somehow is not sending any data anymore, can you do zigbee sniffing of this device while triggering occupancy?

rickkdotnet, 2019-06-12:
I'll try. I haven't done any sniffing yet, but I just found a GBAN CC2530+RFX2401 stick in the mail so if that works I should be able to set something up I guess.

rickkdotnet, 2019-06-12:
@Koenkk well, the new stick works but I can only find the CC2530_CC2591_20190515 firmware for it, which isn't stable (same behavior as the CC2531 and the stick has also already crashed once). 
I'll see if I can flash it as a sniffer, but if you build new firmware it would be nice if you can build one for this stick too, so I have some options. (Or perhaps I should try this myself... )

Koenkk, 2019-06-13:
@rickkdotnet https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2530_CC2591_SOURCE_ROUTING_20190611.zip added it.

rickkdotnet, 2019-06-18:
The CC2530_CC2591 locked up today on SOURCE_ROUTING_20190611. Didn't have time to analyze it yet. I've also received a second CC2531 which I'm trying to get going as a sniffer, but haven't had much luck yet on MacOS. Can't get whsniff to compile, tried a couple of other options but nothing work so far. Will try again soon.

paulianttila, 2019-06-18:
@rickkdotnet, I have been using https://github.com/andrewdodd/ccsniffpiper with wireshark succesfully on mac.

[19 more replies]

closed 2019-10-10 by stale[bot]