Koenkk/zigbee2mqtt #1568: Z-Stack 3 on CC2538 adventures

antst, 2019-05-25:
Me and @kirovilya were working recently on the CC2538 + Z-Stack 3 solution from hardware/firmware side.

As result, I am now testing coordinator on CC2538+CC2592 (ie 20dBm amplifier) on my test mesh with 22 devices. It was never stable enough with CC2530/2531 and first impression is that CC2538 is much better. But it is first day of testing only.

While it is in test, we will start adding here instructions etc.

lolorc, 2019-05-25:
what kind of board are you working with ?
I have one from aliexpress https://www.aliexpress.com/item/32791437711.html?spm=a2g0s.9042311.0.0.5ca94c4d6q47wx but I'm yet to do anything with it.

kirovilya, 2019-05-26:
@lolorc I have the same https://ru.aliexpress.com/item/32879578178.html

kirovilya, 2019-05-26:
What you should know::
1. How to flash the module (flasher and connection)
2. How to connect the module to the serial port
3. How to prepare ZNP-firmware for flash (cc2592 patch and special directive for connecting to the USB or UART)

lolorc, 2019-05-27:
1,2,3 aren't a big deal, It's just about spare time I have left at the moment. :-)

antst, 2019-05-31:
[image]

Here I attach minimal schematics for 2538. Power regulator (surrounded by dashed line) can be replaced by cheap simple module, there are plenty of them around, just look for "AMS1117-3.3 LDO 800MA". CC2538 module I choose can be purchased on aliexpress [here](https://www.aliexpress.com/item/32880588264.html?spm=a2g0o.productlist.0.0.524e330f3GDjsD&algo_pvid=a6d79351-16aa-4ca5-b4b1-4a6e16e471c4&algo_expid=a6d79351-16aa-4ca5-b4b1-4a6e16e471c4-0&btsid=91c32439-1f9d-4f4d-8f25-a1425eb1bc31&ws_ab_test=searchweb0_0%2Csearchweb201602_10%2Csearchweb201603_52)

If you choose another module, then you need to figure out pinout yourself. Practically, there are two most common modules, the one I used (which has PCB antenna and IPX connecter for external one) and another one which has only IPX. IMPORTANT: Pinout of those modules is different.

As for programming, you will need J-link and I included pinout for that for typical 20-pin interface.
You can use native segger application for flashing or you can flash from IAR.
Alternatively, if you don't have j-link, you can flash via any SBC with GPIO (Raspberry Pi for example) by using OpenOCD. 

I will upload firmware shortly.

antst, 2019-05-31:
Just in case, I run it for a week and very happy. 27 devices, no lags, no lost devices, no crashes, minimal latency (I also added some modification to Z2M to reduce delay between commands from default 250ms to 50ms, now coordinator can handle it), and it works like a charm :)

antst, 2019-05-31:
Firmware is based on Z-Stack 3.0.2.

antst, 2019-06-13:
teaser )

[image]

fluppie, 2019-06-13:
Nice! I want one :).

rstein, 2019-06-14:
Great progress, thank you for your effort!
Can you already recommend a specific external antenna (gain)?

antst, 2019-06-14:
> Great progress, thank you for your effort!
> Can you already recommend a specific external antenna (gain)?

Not really. But keep in mind, there is already amplifier and this module has transmission power of 20dBm.

antst, 2019-06-14:
And antenna on this stick is optional. Module has PCB antenna also, it is configurable (via re-solder of resistor) to use PCB antenna or external one/

matlab22, 2019-06-18:
> teaser )
> 
> [image]

How do you connect the sma? Ipex coax cable? Where is the ldo on your pcb? Back side?

antst, 2019-06-18:
Yep, for use with external antenna it is supposed to be connected by IPX coax, and resolver resistor on module. Didn't try it yet. LDO is on back side, yep. Device is not fully assembled. I am still waiting for some parts.

9600, 2019-06-19:
This sounds pretty great. Would the firmware need modifying for use with:

http://www.ti.com/tool/CC2538-CC2592EMK

?

Are the sources in a repo somewhere? And do you have details of the zigbee2mqtt changes you made?

drbios, 2019-06-25:
heelo
what about this board??

https://es.aliexpress.com/item/CC2538-CC2592-Placa-de-desarrollo-Contiki-6-aprendizaje-bajo/32836343416.html?spm=a2g0o.cart.0.0.4bd03c00mo2Xru
[image]

kirovilya, 2019-06-25:
@drbios I think this is a good solution. pins for jtag and usb are exists.

drbios, 2019-06-25:
@kirovilya but the questions is ...how to upload the firmware?  Jatg (whit an external programer)  or USB (this is a development board  so this maybe the way??)

kirovilya, 2019-06-25:
@drbios I have only been flashing through the jtag, but somewhere I read about the possibility through usb (not tested).

antst, 2019-06-25:
Through usb only if 
1) there is already bootloader
2) exposed usb is not “native” usb which this chip has, but rather usb-uart. As bootloader supports only flashing through uart

arteck, 2019-06-27:
@antst where can i buy the black pcb ??

antst, 2019-06-29:
Nowhere )
I am building prototype on this PCB, still waiting remaining components.
Once it is fully tested and functional and firmware is adjusted, we will release schematics and PCB layout open source. But from there you are on your own. 
There is quite a gap between hobby design and production.
If someone will want to try small-scale production of this design, I will be more than happy.
Otherwise, I guess, it is not bad idea to order PCBs in China for a group of people (you can’t order single PCB anyway, but 5-10 at least) and then solder.
In Russian ZigBee community there are some people (who like to solder) who volunteer to solder multiple PCBs and share with other people for compensation of components costs. Maybe you can find/form such community around.
We also plan to release more designs for end-devices also. There are few already in the design/prototyping stage. So, such community can be handy.

antst, 2019-06-29:
Idea of this design is to be cheap and available for DIY.
Otherwise, there are devices (like one above or CC2538-CC2592EMK from TI) which are freely available, but for very different price tag :)

antst, 2019-06-29:
But it all takes time ) orders of components and PCB prototypes from China have quite a latency. And working on firmware requires some free time ) we all have jobs, fortunately or unfortunately :))))

mslr, 2019-06-30:
You can suggest the correct format of the flashing command for OpenOCD in hex file. Or tell me where to get the firmware in bin or elf format.

antst, 2019-07-01:
BTW, there was bug in schematics, I updated it.

hdo, 2019-07-08:
@antst 

Do you mind describing how to build a firmware for the CC2538 devices? I'm interested in this device with serial interface instead of USB. 

Thanks ;-)

antst, 2019-07-08:
I am planning to make firmware for UART instead of USB, very soon, possibly this week.
There was already one attempt  to use my patch for z-stack and modify it for UART, but it didn't work out, so I have to check. I am planning to release sources with instruction, but at first I need to organise it in the way that it will not require sharing whole Z-stack, which contradicts TI policy.

hdo, 2019-07-09:
@antst 
Hey great to hear. Thanks! I'm looking forward to this ;-)

drbios, 2019-07-09:
@antst  which is the best way to connect ... UART or USB? (In a standard raspberry pi 3b+)

hdo, 2019-07-09:
> @antst which is the best way to connect ... UART or USB? (In a standard raspberry pi 3b+)

In my case i want to connect the CC2538 module to a serial tcp board. If you're using a raspberry pi i would recommend going with usb for simplicity.

arteck, 2019-07-11:
@antst I am building prototype on this PCB, still waiting remaining components.

any news here ???

antst, 2019-07-11:
> @antst I am building prototype on this PCB, still waiting remaining components.
> 
> any news here ???

Almost done :) Found couple of bugs with PCB layout. Will tune firmware for stuff on PCB over next couple of days and release.

antst, 2019-07-12:
I updated firmware. Original I uploaded was wrong one, by mistake added wrong hex. PCB layout coming today. It will have pins for UART. Though not RPi HAT socket yet. HAT will have separate pcb

drbios, 2019-07-12:
what about this board??
https://es.aliexpress.com/item/32563769202.html?spm=a2g0o.cart.0.0.3e8b3c00m1HDLs

drbios, 2019-07-14:
Any news how to flash this chip with a openOCD?

drbios, 2019-07-24:
@antst this may be a starting point?
https://learn.adafruit.com/programming-microcontrollers-using-openocd-on-raspberry-pi/overview

mslr, 2019-07-24:
Working starting point for RPI openocd JTAG.
https://movr0.com/2016/09/02/use-raspberry-pi-23-as-a-jtagswd-adapter/
But there are no flashing commands. And most likely you will need a .bin version of the firmware, which is not.

JLFN, 2019-07-26:
Does anyone have this one ? U2ZB_38Dongle
https://www.aliexpress.com/item/32737589263.html

https://item.taobao.com/item.html?id=537985506200

arteck, 2019-07-29:
@antst https://github.com/antst/CC2538ZNP/issues/1

kirovilya, 2019-07-29:
cc2538 without amplifier but in usb-stick form https://ru.aliexpress.com/item/4000016607626.html

antst, 2019-08-06:
Just in case, I updated firmware.
Increased some limits. etc.
It allowed me to decrease delay in sending commands from Z2M to unbelievable 5ms )
Also was able to turn on instantly 15 lamps.
And, theoretically, limits on network size increased quite a lot.

antst, 2019-08-06:
Updated one more time.

antst, 2019-08-06:
By the way, for those who don't need RF frontend (i.e. increased signal transmission levels), there is an option on [aliexpress](https://www.aliexpress.com/item/4000016607626.html?spm=a2g0o.productlist.0.0.4e8f1f19j0rR5P&algo_pvid=5c4e3435-52d0-47c1-9ef9-58c9cc1ce318&algo_expid=5c4e3435-52d0-47c1-9ef9-58c9cc1ce318-1&btsid=e9353a75-cdb8-43b3-9fb1-40bbc0ba6069&ws_ab_test=searchweb0_0,searchweb201602_10,searchweb201603_52) 

"naked" CC2538 has 7dBm transmitter which is still 2.5 times better than CC2531/CC2530.

drbios, 2019-08-07:
@antst excellent news ...i got a pair of cc2538-cc2592 today (same units you suggest) the only thing than i need now is a guide to know how to flash it with my raspberry :(

drbios, 2019-08-07:
Is this information useful some how?
https://github.com/JelmerT/cc2538-bsl

antst, 2019-08-08:
All, 

you  can find schematics, PCB layout and BOM for version of USB stick with this chip [here](https://modkam.ru/?p=1112#more-1112). Article is in Russian. But BOM and gerber files are international )

drbios, 2019-08-08:
@antst this is awesome ... google translate will help ...
really thanks
i am pretty sure i will have a lot of question later ...but i will star buying the part right now

drbios, 2019-08-10:
is this BOM list  OK??
[LCSC_Exported_20190811_013355.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/3489159/LCSC_Exported_20190811_013355.xlsx)

halfcherry, 2019-08-11:
https://www.aliexpress.com/item/32836343416.html?spm=a2g0o.productlist.0.0.14192026Y0Ka77&algo_pvid=623eeca3-3ba9-4e06-8b03-f0c120c44658&algo_expid=623eeca3-3ba9-4e06-8b03-f0c120c44658-3&btsid=fe0af402-baa3-4f29-9c85-dfd2c1b036ea&ws_ab_test=searchweb0_0,searchweb201602_5,searchweb201603_52

JLFN, 2019-08-11:
@halfcherry there is cc2530 with antenna

halfcherry, 2019-08-11:
@JLFN I'm using 1 cc2531 coordinator and 2 x cc2531 as router all with soldered antenna.
My understanding is that since cc2592 is a 20dbm PA which itself already has much better range than cc2531/cc2530. If I add a powered [amplifier,](https://www.aliexpress.com/item/705912287.html) it can even get better.

axiom00, 2019-08-15:
@antst - in the blog post. It mentioned soldering a resistor when using an external antenna. Do you know what value the resistor should be?

axiom00, 2019-08-15:
> @antst - in the blog post. It mentioned soldering a resistor when using an external antenna. Do you know what value the resistor should be?

@antst - please disregard my question. Studying the images a little more, I can see the resistor is soldered to different trace depending if the application is using onboard or external antenna.

Parts ordered!

drbios, 2019-08-16:
Due my extremely low soldering abilities I bought this :
US $14 | Mini cc2538 floor Cc2538mini base plate
https://s.click.aliexpress.com/e/vo2wJmo

kirovilya, 2019-09-03:
Successfully flashed a cc2538-stick without an amplifier. https://ru.aliexpress.com/item/4000016607626.html

[image]
[image]
[image]
[image]

Firmware ZNP Z-Stack 3.0.2 without PA with USB, thanks to Anton @antst 

Two days works fine with ioBroker.zigbee

antst, 2019-09-03:
So, we have
1) PCB for cc2538PA module with frontend
2) we have ready-to use USB stick 
3) apparently there is a base plate for this(?) module.
4) I am working on yet another version of coordinator, where it will have embedded flasher, so you have device which you can always and remotely reflash, reboot, etc. Which does not require separate J-link.

drbios, 2019-09-04:
@antst the base Plate sadly is not for modules with a PA

dzungpv, 2019-09-04:
@antst When you release the modifield diff?

joro-in, 2019-09-05:
> 4. I am working on yet another version of coordinator, where it will have embedded flasher, so you have device which you can always and remotely reflash, reboot, etc. Which does not require separate J-link.

@antst Is this a hardware or firmware change? Thanks :)

[258 more replies]

closed 2020-12-23 by Koenkk