Koenkk/zigbee2mqtt #1571: Pairing DNCKATSW001 (https://github.com/dzungpv/dnckatsw00x/)

olili, 2019-05-26:
I flashed my CC2530 (accurate: E18-MS1-PCB) successfully with the firmware of https://github.com/dzungpv/dnckatsw00x/

I connected no buttons. Only VDD and GND. 

How can I pair it with zigbee2mqtt / CC2531 ? 
Any hints?

Koenkk, 2019-05-26:
@dzungpv can you help?

dzungpv, 2019-06-01:
> 
> 
> I flashed my CC2530 (accurate: E18-MS1-PCB) successfully with the firmware of https://github.com/dzungpv/dnckatsw00x/
> 
> I connected no buttons. Only VDD and GND.
> 
> How can I pair it with zigbee2mqtt / CC2531 ?
> Any hints?

It is automatically pair, but what do you do with it if there is no IO connect?

olili, 2019-06-01:
thx. 

In between I made some progress by my self. 

Your HEX is not working with the popular E18-MSxx  . These chips have a CC2592 Amp. on board. 
So what I did: installed z-stack, applied z-stack router patch from Koenkk and after that I started with your manual. Et voila - I was able to create at least a 1 Ch device  after modifying the LED access (P10 and P11 are in use for accessing CC2592).  

Nevertheless I have not understood what patches/adaptions to do to get a 4 Btn / 4 Rel. solution. 

Could you provide some more details to this? Maybe a patchfile? 

Thx in adcance!

dzungpv, 2019-06-19:
@olili For other board you must re-config the port for output and button, I have the simple 4 gang a long time ago and now replace with a custom design board, and the old project not saved, so you can see the patch for 1 gang here https://github.com/dzungpv/dnckatsw00x/blob/master/dnckatsw001.patch  and adapt to 4 gang very easy with my guide https://github.com/dzungpv/dnckatsw00x/blob/master/README.md

closed 2019-08-25 by stale[bot]