Koenkk/zigbee2mqtt #1572: Sengled Z01-A60EAE27 support

dobegor, 2019-05-26:
## Debug Info
zigbee2mqtt version: git
CC253X firmware version: Z-Stack_Home_1.2

I've tried to add support for Sengled Z01-A60EAE27. The led bulb supports brightness, color temperature change, energy meter.

I've added a following object in devices.js:
```
{
        zigbeeModel: ['Z01-A60EAE27'],
	model: 'Z01-A60EAE27',
	vendor: 'Sengled',
	supports: generic.light_onoff_brightness.supports,
        fromZigbee: generic.light_onoff_brightness.fromZigbee.concat([
            fz.ignore_metering_change,
            fz.ignore_diagnostic_change,
        ]),
        toZigbee: generic.light_onoff_brightness.toZigbee,
},
```

But it only reports "OFF" state, it doesn't report when it turns on, and I can't find any way to control it via MQTT (sending to `zigbee/<device_id>` topic with payload `{"state": "OFF"}` does nothing).

dobegor, 2019-05-26:
Turning the switch off produces the following output:
```
  serialport/bindings/poller received "readable" +2m
  serialport/binding-abstract read +2m
  serialport/bindings/unixRead Starting read +2m
  serialport/bindings/unixRead Finished read 8 bytes +1ms
  serialport/stream binding.read finished +2m
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'srcRtgInd',
  cc-znp   payload: { dstaddr: 22394, relaycount: 0, relaylist: <Buffer > },
  cc-znp   fcs: 175,
  cc-znp   csum: 175 } +2m
  serialport/stream _read reading +8ms
  serialport/binding-abstract read +10ms
  serialport/bindings/unixRead Starting read +9ms
  cc-znp:AREQ <-- ZDO:srcRtgInd, { dstaddr: 22394, relaycount: 0, relaylist: <Buffer > } +2m
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +2m
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +2ms
  serialport/bindings/poller Polling for "readable" +14ms
  serialport/bindings/poller received "readable" +44ms
  serialport/binding-abstract read +47ms
  serialport/bindings/unixRead Starting read +46ms
  serialport/bindings/unixRead Finished read 32 bytes +0ms
  serialport/stream binding.read finished +49ms
  cc-znp { sof: 254,
  cc-znp   len: 27,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 6,
  cc-znp      srcaddr: 22394,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 47,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 6465354,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 7,
  cc-znp      data: <Buffer 08 7a 0a 00 00 10 00> },
  cc-znp   fcs: 10,
  cc-znp   csum: 10 } +58ms
  serialport/stream _read reading +11ms
  serialport/binding-abstract read +12ms
  serialport/bindings/unixRead Starting read +11ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 22394, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 47, securityuse: 0, timestamp: 6465354, transseqnumber: 0, len: 7, data: <Buffer 08 7a 0a 00 00 10 00> } +59ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":6,"srcaddr":22394,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":47,"securityuse":0,"timestamp":6465354,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,122,10,0,0,16,0]}} +7m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +66ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +10ms
  serialport/bindings/poller Polling for "readable" +22ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":6,"srcaddr":22394,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":47,"securityuse":0,"timestamp":6465354,"transseqnumber":0,"len":7,"data":{"0":8,"1":122,"2":10,"3":0,"4":0,"5":16,"6":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":0,"seqNum":122,"cmdId":"report","payload":[{"attrId":0,"dataType":16,"attrData":0}]}} +9ms
  zigbee2mqtt:debug 5/26/2019, 1:35:34 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'Z01-A60EAE27' (0xb0ce1814030460c9) of endpoint 1
  zigbee2mqtt:info 5/26/2019, 1:35:34 PM MQTT publish: topic 'zigbee2mqtt/0xb0ce1814030460c9', payload '{"state":"OFF","linkquality":47}'
  serialport/bindings/poller received "readable" +38ms
  serialport/binding-abstract read +48ms
  serialport/bindings/unixRead Starting read +39ms
  serialport/bindings/unixRead Finished read 8 bytes +0ms
  serialport/stream binding.read finished +50ms
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'srcRtgInd',
  cc-znp   payload: { dstaddr: 22394, relaycount: 0, relaylist: <Buffer > },
  cc-znp   fcs: 175,
[... 21 lines cut ...]
  cc-znp      srcaddr: 22394,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 44,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 6465721,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 9,
  cc-znp      data: <Buffer 08 7b 0a 00 04 2a 00 00 00> },
  cc-znp   fcs: 193,
  cc-znp   csum: 193 } +58ms
  serialport/stream _read reading +11ms
  serialport/binding-abstract read +12ms
  serialport/bindings/unixRead Starting read +12ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1794, srcaddr: 22394, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 44, securityuse: 0, timestamp: 6465721, transseqnumber: 0, len: 9, data: <Buffer 08 7b 0a 00 04 2a 00 00 00> } +59ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1794,"srcaddr":22394,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":44,"securityuse":0,"timestamp":6465721,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[8,123,10,0,4,42,0,0,0]}} +110ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +68ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +11ms
  serialport/bindings/poller Polling for "readable" +25ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1794,"srcaddr":22394,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":44,"securityuse":0,"timestamp":6465721,"transseqnu [... cut ...]
  zigbee2mqtt:debug 5/26/2019, 1:35:34 PM Received zigbee message of type 'attReport' with data '{"cid":"seMetering","data":{"instantaneousDemand":0}}' of device 'Z01-A60EAE27' (0xb0ce1814030460c9) of endpoint 1
  zigbee2mqtt:warn 5/26/2019, 1:35:34 PM No converter available for 'Z01-A60EAE27' with cid 'seMetering', type 'attReport' and data '{"cid":"seMetering","data":{"instantaneousDemand":0}}'
  zigbee2mqtt:warn 5/26/2019, 1:35:34 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Koenkk, 2019-05-26:
You should send to `zigbee2mqtt/device_id/set`, `{"state": "ON"}`

dobegor, 2019-05-26:
Thanks, I’ll try that in a few hours.
But why there are no messages when the bulb turns on, and even on manual reconnect there’s always “OFF” state, even if it’s on? Should I provide any additional debug info?

Best regards,
George Dobrovolsky

Koenkk, 2019-05-26:
How do you turn the bulb on?

dobegor, 2019-05-26:
Using the usual wall switch. The bulb sends message when I turn it off, though.

Best regards,
George Dobrovolsky

dobegor, 2019-05-26:
@Koenkk yes, posting to `zigbee2mqtt/device_id/set` did the trick, thanks! Somehow it slipped past me in the docs.

However, I still wonder why isn't there a message when I turn on the power to the bulb, but there is one when I turn it off.

Koenkk, 2019-05-27:
@dobegor for this you could use the availability feature: #775

dobegor, 2019-05-27:
@Koenkk I've just tried the availability feature. It does set the device availability to offline after a timeout, but it does not come back to "online" for this bulb unless I make some action via HA.

I've heard there's a `configure` section for some bulbs, but I've got no idea how to use it and I don't own a Sengled hub to sniff anything. Is there some sort of spec / any info how can I proceed with this?

Awesome thanks for help!

Koenkk, 2019-05-28:
@dobegor filling the configure section should be similar as the reporting functionality for this kind of device. Can you try this? (http://www.zigbee2mqtt.io/information/report.html)

dobegor, 2019-05-29:
@Koenkk report feature works - after powering on it takes 2-3 seconds and the bulb successfully reports that it's on. is there a possibility to enable reporting for this device in the `configure` section, rather that global setting?

Koenkk, 2019-05-29:
Good, yes this is possible but not preferred, in this way we keep behavior across bulbs the same.

dobegor, 2019-05-29:
@Koenkk I'd prefer that over an advanced setting so that bulb would work out-of-the-box as intended. Or this setting is going to be default in future releases?

Koenkk, 2019-05-29:
@dobegor It may become default in future releases.

dobegor, 2019-05-29:
OK then 👍 
I'll file a PR for this bulb.
Awesome thanks for help!

Koenkk, 2019-06-10:
Closing this one as waiting for pr.

closed 2019-06-10 by Koenkk