Koenkk/zigbee2mqtt #1574: Unable to pair anything (Tradfri)

MRvH88, 2019-05-26:
I am trying to add new devices to zigbee2mqtt, however it is not working anymore. I am also not getting any logs regarding joining of devices.

I am running zigbee2mqtt version 1.4.0 (commit #unknown) in Docker and coordinator firmware version: '20190425' on a CC2531.

I am trying to add different Tradfri lights to zigbee2mqtt after resetting them (6x On/Off), but am not seeing anything in the logs (Debug level).

Steps I have taken:
* Different USB ports / reset button / reboot
* Reflash CC2531 (on Mac and Windows)
* Different CC2531 sticks
* New docker without old config
* Turn off Tradfri Hub
* Turned Join on in config file and/or through Mqtt

Additional remark: the first time I plug in a freshly flashed CC2531 it blinks red/green and I have to wiggle it or remove and plug it in again to make it green. (Did not have this previously I think, and it happens on both my CC2531).

I have 4 Tradfri motion sensors working when in range, had a GU10 (LED1537R6) that stopped working and tried to re-add it but its not joining. Same goed for other GU10s (LED1537R6) and Bulb E12/E14 (LED1536G5) I am trying to add.

I am really struggling to find what I am missing and why it is not working anymore, hope someone can assist.

Koenkk, 2019-05-27:
Did you also try pairing it very close to the coordinator?

MRvH88, 2019-05-27:
Yes the lights were very close to the coordinator. Tried the same coordinator on a different ROCK64 and that worked right away. I copied all information to my main device and am receiving the following information:

5/27/2019, 4:38:21 PM - error: Zigbee publish to device 'Couch Light', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 183. APS no ack.
5/27/2019, 4:38:21 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 183. APS no ack.","meta":{"entity":{"ID":"0xd0cf5efffe1724d7","type":"device","friendlyName":"Couch Light"},"message":"{\"state\": \"OFF\"}"}}'

My coordinator is directly in my usb slot, can it be that this is causing interference? It did work for some time however now it is not working at al. I did however add an additional HDD.

Koenkk, 2019-05-27:
You could try an usb extension cable.

MRvH88, 2019-05-28:
Ordered one yesterday, everything is correctly again through a spare sbc. 

I guess it is either interferance or bad usb slots. Will inform if it solves the issue.

bochelork, 2019-05-29:
I couldn't pair my Ikea bulbs on 20190425 firmware as well. Went back to 20190315 and everything paired quick and easily. Even from >10meters. Ordered extension cable and second CC2531 to be able to snif data.

andreaslink-de, 2019-06-03:
Interesting, I've easily paired an IKEA remote (E1524) out of the box, but when I also tried to add a lamp (simple 1000lm E27), I only get the messages:
`PM Device with modelID 'undefined' is not supported.`
`PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`
And I have no clue, how to get this solved? Do I need to create a new profile/device here? It's running firmware 1.2.214.

Koenkk, 2019-06-05:
@andreaslink-de something went wrong when pairing the device, please try to re-pair it.

andreaslink-de, 2019-06-05:
Thanks @Koenkk you are right. I repeated re-pairing several times and I "somehow got it paired", but I honestly did not recognize the moment it happend. At one point I could see it in my database file, so I thought, this one is looking ok now and then I was able to control it. Is there any sign on the light itself, when paring was successful or any other significant marker/message to proof it had been successful?

MRvH88, 2019-06-06:
USB extension cable seems to have solved the problem, did not add anything new but the connection seems to be ok. Adding a second HDD seems to have created too much interference. As far as I'm concerned this issue can be closed.

Koenkk, 2019-06-07:
@andreaslink-de this can be seen in the zigbee2mqtt startup log where it reports the model of the device.

@MRvH88 great that your problem has been solved!

closed 2019-06-07 by Koenkk