Koenkk/zigbee2mqtt #1577: Unable to get devices in HASSIO ( Ikea bulb not recoqnized? )

Bart-1992, 2019-05-27:
# Bug Report
## What happened

I've tried to get Zigbee2mqtt up and running via the Hassio intergrated add-on. Looking at the log's everything is running fine however the device i'm trying to control is not recognized --> see log @ line 19/21. Probably the device is not appearing in the Hassio UI because it's not parsed to the MQTT broker? 

[https://hastebin.com/xebibixexa.sql](url)

## What did you expect to happen
It should be recognized since it's in the supported devices list? 

## Debug Info
zigbee2mqtt version:
Running on zigbee2mqtt 1.4. Hassio add-on

Koenkk, 2019-05-28:
On line 23 you see that the device discovery message is send.

- Did you enable discovery in home assistant? (http://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery)
- Is there anything under Home Assistant -> Configuration -> Integrations -> MQTT?

Bart-1992, 2019-05-28:
Thanks, got it working!
Apparently something was wrong within the broker settings....!

closed 2019-05-28 by Bart-1992