Koenkk/zigbee2mqtt #1578: NPM Install Failure - ENOENT: no such file or directoy

elardner, 2019-05-27:
I am having an issue trying to install zigbee2mqtt on my RasbPi 3 B+ that I am not having any luck resolving. After pulling the files into /opt/zigbee2mqtt and changing ownership to "pi", I run "npm install" and am presented with the following error:

```
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/acorn-cd6124d3/dist/acorn.js.map'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/source-map-52448a65/dist/source-map.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/source-map-23af578d/dist/source-map.js'
npm ERR! path /opt/zigbee2mqtt/node_modules/.staging/zcl-id-67ebab72/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open
npm ERR! enoent ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/zcl-id-67ebab72/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-05-27T12_27_08_355Z-debug.log

```
I have tried wiping out /opt/zigbee2mqtt and starting over, moving to the dev version and upgrading npm to 6.9.1 and am still having the issue.

Koenkk, 2019-05-28:
Can you try with `rm -rf node_modules && sudo npm install`?

elardner, 2019-05-28:
Removed node_modules and ran `sudo npm install` and have a different failure:

```
npm ERR! err.code.match is not a function

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-05-28T11_27_14_810Z-debug.log
```

Koenkk, 2019-05-28:
Can you post the whole log?

elardner, 2019-05-28:
Should be all here:
[](https://pastebin.com/RbFipeGV)
[](https://pastebin.com/A893JrBM)

Koenkk, 2019-05-28:
Seems to be an npm issue. #1102 seems to be similar.

elardner, 2019-05-28:
Is there a recommended version of npm to use for this? Tried 6.9.0 fresh install and am seeing the same errors with `npm install` and `sudo npm install`.

Koenkk, 2019-05-28:
I've just tested with npm 6.9.0 which seems to work fine. Are you on nodejs 10? (`node --version`).

elardner, 2019-05-28:
Yeah: 10.15.3

elardner, 2019-05-29:
Reinstalled Hassbian and clean installed the required packages (nodejs upgraded to 10.16.0 and npm is 6.9.0) and the install completed successfully.

Koenkk, 2019-05-29:
Great!

closed 2019-05-29 by Koenkk