Koenkk/zigbee2mqtt #1580: permit_join false not honored - Aqara Router Device LLKZMK11LM

BLiNdGuN, 2019-05-27:
Hello,

My setup is not honoring the permit_join: false
HA plugin 1.4.0,, CC2531 with '20190425' firmware.

I have a bunch of Xiaomi Devices acting like routers.
10 units - ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
1 unit -  JTQJ-BF-01LM/BW - Xiaomi MiJia gas leak detector
1 unit - LLKZMK11LM - Xiaomi Aqara wireless relay controller (Router)

And i think it is related to that..
I've read some closed posts mentioning similar problems.

With:
zigbee2mqtt/bridge/config/#
{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"info","permit_join":false}

having a subscription on:
zigbee2mqtt/bridge/log/#
I manage to see that every new device connects.. and pairs with my authorization.

Anyone with the same kind of problem?
I don't have a second cc2531 to sniff and debug..

Any ideas?
best regards.

Koenkk, 2019-05-28:
Does it only happen to devices that had already been connected to the network?

BLiNdGuN, 2019-05-28:
Nope.. with new devices.
Devices never paired with my CC2531 zigbee network.
I have a bunch of xiaomi wireless switches i don't need and i've been testing with them.

After posting the issue i managed to see that if i turn all the "xiaomi routers" off, no new pairings occur with permit_join:false
As soon as i turn on the xiaomi routers again, unauthorized new devices start to manage to pair again..

Any suggestions?
Best regards.

Koenkk, 2019-05-28:
Could you share your zigbee sniff log when pairing a device with permit join disabled? (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

BLiNdGuN, 2019-05-28:
Only have one CC2531.. the one that i currently use on my zigbee network and that i need to be working to have my home operational..
I have to buy another one..... till then.. no sniffing possible..

Koenkk, 2019-05-29:
Can you try changing `255` on https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L192 and https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L202 to `240` and https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L214 to `return this.shepherd.controller._permitJoinTime > 0;`

This should automatically close the network after 240 seconds when permit join is false.

BLiNdGuN, 2019-06-04:
Back home..
I'll try.
feedback soon..

closed 2019-08-10 by stale[bot]