Koenkk/zigbee2mqtt #1581: Unsupported device: HEIMAN smart remote controller: HS1RC-E

bambouzel, 2019-05-28:
https://nl.aliexpress.com/item/32975159931.html?spm=a2g0s.9042311.0.0.177d4c4dWELfwJ

```
2019-5-28 20:26:12 - info: Starting **zigbee2mqtt version 1.4.0** (commit #927c4db)
2019-5-28 20:26:12 - info: Starting zigbee-shepherd
2019-5-28 20:26:14 - info: zigbee-shepherd started
2019-5-28 20:26:14 - info: Coordinator **firmware version: '20180815'**
```
I enabled full debug, enabled join and got the following:

```
2019-5-28 20:43:27 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'undefined' (0x000d6f00104f246e) of endpoint 1
2019-5-28 20:43:27 - warn: Device with modelID 'undefined' is not supported.
2019-5-28 20:43:27 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

Then I added a section to the devices and restarted

```
    {
        zigbeeModel: ['RC-EM'],
        model: 'HS1RC-E',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        supports: 'arm disarm home sos',
        fromZigbee: [],
        toZigbee: [],
    },
```
In the startup log:
`
2019-5-28 20:58:31 - info: 0x000d6f00104f246e (0x000d6f00104f246e): HS1RC-E - HEIMAN Smart remote controller (EndDevice)
`

Then I clicked on some buttons:

```
2019-5-28 21:04:00 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:04:07 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:04:09 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:04:15 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"manufacturerName":"HEIMAN"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:04:17 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"65533":1,"modelId":"RC-EM","dateCode":"2017.11.1","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:04:17 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-28 21:04:21 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"65533":1,"batteryVoltage":29,"batteryPercentageRemaining":172}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:04:21 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":172}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:04:21 - warn: No converter available for 'HS1RC-E' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":172}}'
2019-5-28 21:04:21 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-5-28 21:04:21 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-28 21:04:25 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
```

For the above messages I can create a converter but I do get other messages that represent the button clicks, however they have another ID

```
2019-5-28 21:03:34 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":0}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-28 21:03:34 - debug: Ignoring message from coordinator

2019-5-28 21:18:02 - debug: Received zigbee message of type 'cmdEmergency' with data '{"cid":"ssIasAce","data":{}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-28 21:18:02 - debug: Ignoring message from coordinator
```

So this represents the arm and the sos button but why is it coming from an undefined device?

Hope you can help me,

Thank you,

--Ivo

tb-killa, 2019-05-28:
Hi, thanks for sharing the Informations.
The ``undefined`` comes because on first pairing its not complete but we actually only update the entry if ``reported`` comes in.
Maybe restart of zigbee2mqtt could help.?

@Koenkk : We should duplicate https://github.com/Koenkk/zigbee-shepherd/commit/6c05654ef31b093386d368a9db8e51691452968e for ``ind:readRsp`` too.

Do you see inside your logfile some sort of entrys like this: ``Message has unsupported cmdID:`` ??

For ``cid 'genPowerCfg'`` you could use actually existings converter to get them include and report.

I see that some of your button actions results on different endpoint (log entry shows ``ep:11``).

bambouzel, 2019-05-28:
Hi, thank you for the fast reply ... made me reboot the raspberry ...

```
2019-5-28 21:57:20 - info: Coordinator firmware version: '20180815'
2019-5-28 21:57:20 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0018e1a819","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1559073440,"joinTimeLeft":0}
2019-5-28 21:57:20 - info: Currently 6 devices are joined:
2019-5-28 21:57:20 - info: 0x00158d0002c66b05 (0x00158d0002c66b05): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2019-5-28 21:57:20 - info: 0x00158d0002b5685f (0x00158d0002b5685f): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
2019-5-28 21:57:20 - info: 0x00158d0002c992ee (0x00158d0002c992ee): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
2019-5-28 21:57:20 - info: 0x00158d0002b5d1d1 (0x00158d0002b5d1d1): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2019-5-28 21:57:20 - info: 0x00158d0002b00d26 (0x00158d0002b00d26): WXKG12LM - Xiaomi Aqara wireless switch (with gyroscope) (EndDevice)
2019-5-28 21:57:20 - info: 0x000d6f00104f246e (0x000d6f00104f246e): HS1RC-E - HEIMAN Smart remote controller (EndDevice)
```
the **undefined device** is mentioned in the second logline: `"ieeeAddr":"0x00124b0018e1a819"
`
after clicking the arm button on the remote:

```
2019-5-28 21:59:11 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-28 21:59:11 - debug: Ignoring message from coordinator
2019-5-28 21:59:19 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-28 21:59:19 - debug: Ignoring message from coordinator
2019-5-28 21:59:26 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-28 21:59:26 - debug: Ignoring message from coordinator
2019-5-28 21:59:37 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-28 21:59:37 - debug: Ignoring message from coordinator
2019-5-28 21:59:37 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:59:43 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:59:45 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:59:51 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"manufacturerName":"HEIMAN"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:59:53 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"65533":1,"modelId":"RC-EM","dateCode":"2017.11.1","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:59:53 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-28 21:59:57 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"65533":1,"batteryVoltage":28,"batteryPercentageRemaining":150}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:59:57 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":28,"batteryPercentageRemaining":150}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 21:59:57 - warn: No converter available for 'HS1RC-E' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":28,"batteryPercentageRemaining":150}}'
2019-5-28 21:59:57 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-5-28 21:59:57 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-28 22:00:01 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 22:00:01 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-28 22:00:07 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":271,"zoneStatus":0,"iasCieAddr":"0x0000000000000000","zoneId":255}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 22:00:09 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"65533":1}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-28 22:00:09 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
[... cut ...]
  serialport/binding-abstract read +9ms
  serialport/bindings/unixRead Starting read +9ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 11 } +2s
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: {"status":0,"endpoint":1,"transid":11} +2s
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 11 +2s
  serialport/bindings/unixRead Finished read 38 bytes +4ms
  serialport/stream binding.read finished +6ms
```

I will get back to you tomorrow.

Thank you,

--Ivo

bambouzel, 2019-05-29:
I have added a generic converter to handle the genPowerCfg message.

When I start zigbee2mqtt I get:

`2019-5-29 20:13:16 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0018e1a819","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1559153596,"joinTimeLeft":0}
`
The remote appears in the list of devices:
`2019-5-29 20:13:16 - info: 0x000d6f00104f246e (0x000d6f00104f246e): HS1RC-E - HEIMAN Smart remote controller (EndDevice)
`
but I don't see anything appearing in HomeAssistant.

When I press the arm button I get:

```
2019-5-29 20:25:35 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 20:25:35 - debug: Ignoring message from coordinator
2019-5-29 20:25:42 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 20:25:42 - debug: Ignoring message from coordinator
2019-5-29 20:25:50 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 20:25:50 - debug: Ignoring message from coordinator
2019-5-29 20:26:01 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 20:26:01 - debug: Ignoring message from coordinator
2019-5-29 20:26:01 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:07 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:09 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:15 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"manufacturerName":"HEIMAN"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:17 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"65533":1,"modelId":"RC-EM","dateCode":"2017.11.1","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:17 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-29 20:26:21 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"65533":1,"batteryVoltage":28,"batteryPercentageRemaining":126}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:21 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":126}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:21 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":126}'
2019-5-29 20:26:21 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-29 20:26:25 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:25 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-29 20:26:31 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":271,"zoneStatus":0,"iasCieAddr":"0x0000000000000000","zoneId":255}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:33 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"65533":1}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:33 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-29 20:26:35 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-29 20:26:35 - debug: Accept device incoming with ieeeAddr '0x000d6f00104f246e' permit join is 'true'
2019-5-29 20:26:35 - debug: Allowing device '0x000d6f00104f246e' to join
2019-5-29 20:26:35 - debug: Received zigbee message of type 'devIncoming' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:35 - info: Device incoming...
2019-5-29 20:26:35 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2019-5-29 20:26:35 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 20:26:36 - info: Succesfully reenabled joining
```

After restarting HomeAssistant I see in the debug log:

```
  zigbee2mqtt:debug 2019-5-29 20:35:50 Received MQTT message on 'hass/status' with data 'online'
[... cut ...]
  zigbee2mqtt:info 2019-5-29 20:36:10 MQTT publish: topic 'zigbee2mqtt/0x00158d0002b5d1d1', payload '{"illuminance":34,"linkquality":31,"occupancy":false,"battery":100,"voltage":3005}'
  zigbee2mqtt:info 2019-5-29 20:36:10 MQTT publish: topic 'zigbee2mqtt/0x00158d0002b00d26', payload '{"battery":100,"voltage":3155,"linkquality":70}'
  zigbee2mqtt:info 2019-5-29 20:36:10 MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":126}'
```

But still I don't see the remote appearing in Home Assistant.

tb-killa, 2019-05-29:
I think its a IAS Device.
Maybe we have to bring them to IAS Zone first to get them work?

Please check if we could use this ?
Extend your device entry with this and restart z2m with debug param and provide log.

```
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 100}, cb),
            ];

            execute(device, actions, callback, 250);
        },
    },
```

bambouzel, 2019-05-29:
I can now see the device in Home Assistant but I'll need to experiment a bit to get the four states in there.

In the zigbee log I have the following:
```
2019-5-29 21:22:57 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
2019-5-29 21:22:57 - warn: This can be ignored if the device is working properly
2019-5-29 21:27:10 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
2019-5-29 21:27:10 - warn: This can be ignored if the device is working properly
2019-5-29 21:27:19 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #3)
2019-5-29 21:27:19 - warn: This can be ignored if the device is working properly
2019-5-29 21:27:33 - warn: No converter available for 'HS1RC-E' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"iasCieAddr":"0x00124b0018e1a819"}}'
2019-5-29 21:27:33 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Clicking on the arm button gives:

```
2019-5-29 21:41:12 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:41:12 - debug: Ignoring message from coordinator
2019-5-29 21:41:14 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:41:14 - debug: Ignoring message from coordinator
2019-5-29 21:41:22 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:41:22 - debug: Ignoring message from coordinator
2019-5-29 21:41:30 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:41:30 - debug: Ignoring message from coordinator
2019-5-29 21:41:40 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:41:40 - debug: Ignoring message from coordinator
2019-5-29 21:41:40 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 21:41:40 - debug: Configuring 0x000d6f00104f246e (0x000d6f00104f246e) ...
2019-5-29 21:41:41 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
2019-5-29 21:41:41 - warn: This can be ignored if the device is working properly
2019-5-29 21:41:46 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 21:41:46 - debug: Configuring 0x000d6f00104f246e (0x000d6f00104f246e) ...
2019-5-29 21:41:46 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 21:41:46 - warn: No converter available for 'HS1RC-E' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}
2019-5-29 21:41:46 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-5-29 21:41:48 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
```

Clicking on the home button gives

```
2019-5-29 21:53:45 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:53:45 - debug: Ignoring message from coordinator
2019-5-29 21:53:53 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:53:53 - debug: Ignoring message from coordinator
2019-5-29 21:54:01 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:54:01 - debug: Ignoring message from coordinator
2019-5-29 21:54:11 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 21:54:11 - debug: Ignoring message from coordinator
2019-5-29 21:54:11 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
```

So armmode 3 seems to be the arm button, armmode 1 is the home button etc...

tb-killa, 2019-05-29:
please check with this config:
``
fromZigbee: [fz.KEF1PA_arm, fz.KEF1PA_panic],
``
you have to extend with your battery converter but this should show you the 4 button states.
please report :)

bambouzel, 2019-05-29:
I took the fz.heiman_smart_controller_armmode instead, seemed a bit more appropiate ...

```
{
   3023         zigbeeModel: ['RC-EM'],
   3024         model: 'HS1RC-E',
   3025         vendor: 'HEIMAN',
   3026         description: 'Smart remote controller',
   3027         supports: 'arm disarm home sos',
   3028         fromZigbee: [fz.heiman_smart_controller_armmode, fz.generic_battery_change],
   3029         toZigbee: [],
   3030         configure: (ieeeAddr, shepherd, coordinator, callback) => {
   3031             const device = shepherd.find(ieeeAddr, 1);
   3032             const actions = [
   3033                 (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
   3034                 (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
   3035             ];
   3036
   3037             execute(device, actions, callback, 1000);
   3038         },
   3039     }
```

I get the following log:

```
2019-5-29 22:29:10 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
2019-5-29 22:29:10 - warn: This can be ignored if the device is working properly
...
2019-5-29 22:30:12 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 22:30:12 - debug: Ignoring message from coordinator
2019-5-29 22:30:20 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 22:30:20 - debug: Ignoring message from coordinator
2019-5-29 22:30:27 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 22:30:27 - debug: Ignoring message from coordinator
...
2019-5-29 22:30:38 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-29 22:30:38 - debug: Ignoring message from coordinator
2019-5-29 22:30:38 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 22:30:38 - debug: Configuring 0x000d6f00104f246e (0x000d6f00104f246e) ...
2019-5-29 22:30:43 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
2019-5-29 22:30:43 - warn: This can be ignored if the device is working properly
2019-5-29 22:30:49 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 22:30:49 - debug: Configuring 0x000d6f00104f246e (0x000d6f00104f246e) ...
2019-5-29 22:30:50 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #3)
2019-5-29 22:30:50 - warn: This can be ignored if the device is working properly
2019-5-29 22:30:50 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-29 22:30:51 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":271,"zoneStatus":0,"iasCieAddr":"0x00124b0018e1a819","zoneId":255}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 22:30:56 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"65533":1,"batteryVoltage":26,"batteryPercentageRemaining":84}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 22:30:56 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":26,"batteryPercentageRemaining":84}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 22:30:56 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":84}'
2019-5-29 22:30:56 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-5-29 22:31:00 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-5-29 22:31:00 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
```

bambouzel, 2019-05-31:
I took a step back and first removed all topics involving `0x000d6f00104f246e` from mosquitto, homeassistant and zigbee2mqtt.
Then I copied and adjusted an existing HS1RC configuration into
```
    {
        zigbeeModel: ['RC-EM'],
        model: 'HS1RC-E',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        supports: 'action',
        fromZigbee: [
            fz.battery_200, fz.ignore_power_change,
            fz.KEF1PA_arm, fz.KEF1PA_panic
        ],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];

            execute(device, actions, callback, 1000);
        },
    },
```
added the ` 'HS1RC-E': [configurations.sensor_action, configurations.sensor_battery],` to the homeassistant.js.

Starting z2m gives the following:

```
2019-5-31 14:56:19 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-31.14-56-19'
2019-5-31 14:56:19 - debug: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-05-30.10-11-27'
2019-5-31 14:56:20 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
2019-5-31 14:56:20 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-5-31 14:56:20 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-5-31 14:56:20 - info: Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
2019-5-31 14:56:20 - info: Starting zigbee-shepherd
2019-5-31 14:56:22 - info: zigbee-shepherd started
2019-5-31 14:56:22 - info: Coordinator firmware version: '20180815'
2019-5-31 14:56:22 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0018e1a819","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1559307382,"joinTimeLeft":0}
2019-5-31 14:56:22 - info: Currently 6 devices are joined:
..
2019-5-31 14:56:22 - info: 0x000d6f00104f246e (0x000d6f00104f246e): HS1RC-E - HEIMAN Smart remote controller (EndDevice)
..
2019-5-31 14:56:22 - debug: Configuring 0x000d6f00104f246e (0x000d6f00104f246e) ...
2019-5-31 14:56:22 - info: zigbee-shepherd ready
2019-5-31 14:56:22 - info: Connected to MQTT server
2019-5-31 14:56:22 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
...
2019-5-31 14:56:22 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":36,"linkquality":49}'
...
2019-5-31 14:56:22 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_action","unique_id":"0x000d6f00104f246e_action_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-5-31 14:56:22 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_battery","unique_id":"0x000d6f00104f246e_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-5-31 14:56:22 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_linkquality","unique_id":"0x000d6f00104f246e_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-5-31 14:56:22 - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"927c4db","coordinator":20180815,"log_level":"debug","permit_join":true}'
2019-5-31 14:56:23 - debug: Mounted the cieApp (epId 11)
2019-5-31 14:56:30 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
2019-5-31 14:56:30 - warn: This can be ignored if the device is working properly
...
2019-5-31 14:59:42 - info: Succesfully reenabled joining
```

Clicking the arm button:

```
[... cut ...]
2019-5-31 15:14:23 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":0}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 15:14:23 - debug: Ignoring message from coordinator
2019-5-31 15:14:23 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
```

So the button clicks react with the `ssIasAce/cmdArm` but on an undefined device of endpoint 11.

bambouzel, 2019-05-31:
The incoming cmdArm in the log:

```
  serialport/bindings/poller received "readable" +4s
  serialport/binding-abstract read +4s
  serialport/bindings/unixRead Starting read +4s
  serialport/bindings/unixRead Finished read 31 bytes +1ms
  serialport/stream binding.read finished +4s
  cc-znp { sof: 254,
  cc-znp   len: 26,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 1281,
  cc-znp      srcaddr: 1458,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 11,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 42,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 11214888,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 6,
  cc-znp      data: <Buffer 01 64 00 00 00 ff> },
  cc-znp   fcs: 217,
  cc-znp   csum: 217 } +4s
  serialport/stream _read reading +16ms
  serialport/binding-abstract read +17ms
  serialport/bindings/unixRead Starting read +17ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1281, srcaddr: 1458, srcendpoint: 1, dstendpoint: 11, wasbroadcast: 0, linkquality: 42, securityuse: 0, timestamp: 11214888, transseqnumber: 0, len: 6, data: <Buffer 01 64 00 00 00 ff> } +4s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1281,"srcaddr":1458,"srcendpoint":1,"dstendpoint":11,"wasbroadcast":0,"linkquality":42,"securityuse":0,"timestamp":11214888,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[1,100,0,0,0,255]}} +4s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4s
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +8ms
  serialport/bindings/poller Polling for "readable" +27ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1281,"srcaddr":1458,"srcendpoint":1,"dstendpoint":11,"wasbroadcast":0,"linkquality":42,"securityuse":0,"timestamp":11214888,"transseqnumber":0,"len":6,"data":{"0":1,"1":100,"2":0,"3":0,"4":0,"5":255},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":0},"manufCode":0,"seqNum":100,"cmdId":"arm","payload":{"armmode":0}}} +8ms
  zigbee2mqtt:debug 2019-5-31 15:13:57 Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":0}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
  zigbee2mqtt:debug 2019-5-31 15:13:57 Ignoring message from coordinator
```

Source endpoint is 1, destination endpoint is 11 but message says:

`  zigbee2mqtt:debug 2019-5-31 15:13:57 Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":0}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
`

tb-killa, 2019-05-31:
You have to remove the configure part for ias, because it's not needing. 

```
configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];

            execute(device, actions, callback, 1000);
        },
    },
```

Please do me a check. 
You have 4 buttons right? 
Please provide for every button push the needing ``armmode `` return code. 
It could be that some sort of button provide ``zonestate`` OR ``zoneType`` Infos too. 
I will help to build the needing converter.

bambouzel, 2019-05-31:
Ok, removed the `configure` part.

Arm button (displaying lock):

```
2019-5-31 16:00:40 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 16:00:40 - debug: Ignoring message from coordinator
```

Arm button (displaying lock with house):

```
2019-5-31 16:01:10 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":1}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
```

Disarm button (displaying unlock):

```
2019-5-31 16:01:53 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":0}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 16:01:53 - debug: Ignoring message from coordinator
```

Panic button (displaying just a circle ;)):

```
2019-5-31 16:02:33 - debug: Received zigbee message of type 'cmdEmergency' with data '{"cid":"ssIasAce","data":{}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 16:02:33 - debug: Ignoring message from coordinator
```

tb-killa, 2019-05-31:
For last ones (Panic Button) you could use ``fz.heiman_smart_controller_emergency``
For the other three states please insert this to ``fromZigbee.js`` and also to your ``devices.js`` part (``fz.HS1RC-E_arm``).
```
HS1RC-E_arm: {
        cid: 'ssIasAce',
        type: 'cmdArm',
        convert: (model, msg, publish, options) => {
            const action = msg.data.data['armmode'];
            delete msg.data.data['armmode'];
            const modeLookup = {
		0: 'unlock',
                1: 'home',
                3: 'lock',
            };
            return {action: modeLookup[action]};
        },
    },
```

bambouzel, 2019-05-31:
First of all, thanks for helping!
The logs still show the same loglines ...
I will now add an automation in homeassistant so I can track what button was pressed.
I will get back to u later.

bambouzel, 2019-05-31:
Hélas, I don't see anything coming in homeassistant from the `0x000d6f00104f246e_action` entity which is configured in homeassistant.js:

`'HS1RC-E': [configurations.sensor_action, configurations.sensor_battery],`

The z2m log contains:

```
2019-5-31 19:33:53 - warn: No converter available for 'HS1RC-E' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":271,"zoneStatus":0,"iasCieAddr":"0x00124b0018e1a819","zoneId":23}}'
2019-5-31 19:33:53 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-5-31 19:33:55 - warn: No converter available for 'HS1RC-E' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"65533":1}}'
2019-5-31 19:33:55 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-5-31 19:33:57 - info: Device incoming...
2019-5-31 19:33:57 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
```

If I click a button I don't see anything being published to mqtt, only these loglines:

```
2019-5-31 19:44:06 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 19:44:06 - debug: Ignoring message from coordinator
2019-5-31 19:44:14 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 19:44:14 - debug: Ignoring message from coordinator
2019-5-31 19:44:22 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 19:44:22 - debug: Ignoring message from coordinator
2019-5-31 19:44:32 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-5-31 19:44:32 - debug: Ignoring message from coordinator
2019-5-31 19:44:32 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
```

bambouzel, 2019-05-31:
I'm now removing z2m configuration/database/state and start from clean state.

bambouzel, 2019-06-02:
Searched a bit an the double lines in the database are normal, so repeated the above:
1. pressed reset on the remote for 5 seconds -> saw the ZDO leaving request passing
2. stopped z2m, cleared devices in configuration, removed database, cleared state
3. started z2m

```
2019-6-2 15:19:56 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-02.15-19-56'
2019-6-2 15:19:56 - debug: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-06-01.20-16-53'
2019-6-2 15:19:56 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
2019-6-2 15:19:57 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-6-2 15:19:57 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-2 15:19:57 - info: Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
2019-6-2 15:19:57 - info: Starting zigbee-shepherd
2019-6-2 15:19:59 - info: zigbee-shepherd started
2019-6-2 15:19:59 - info: Coordinator firmware version: '20180815'
2019-6-2 15:19:59 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0018e1a819","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1559481599,"joinTimeLeft":0}
2019-6-2 15:19:59 - info: Currently 0 devices are joined:
2019-6-2 15:19:59 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-6-2 15:19:59 - warn: Allowing new devices to join.
2019-6-2 15:19:59 - warn: Set `permit_join` to `false` once you joined all devices.
2019-6-2 15:19:59 - info: Zigbee: allowing new devices to join.
2019-6-2 15:19:59 - info: Connecting to MQTT server at mqtt://localhost
2019-6-2 15:19:59 - info: zigbee-shepherd ready
2019-6-2 15:19:59 - info: Connected to MQTT server
2019-6-2 15:19:59 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-6-2 15:19:59 - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"927c4db","coordinator":20180815,"log_level":"debug","permit_join":true}'
2019-6-2 15:19:59 - debug: Mounted the cieApp (epId 11)
2019-6-2 15:23:19 - info: Succesfully reenabled joining
2019-6-2 15:24:57 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-2 15:26:39 - info: Succesfully reenabled joining
2019-6-2 15:29:57 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-2 15:29:59 - info: Succesfully reenabled joining
```

4. pressed reset on the remote for 5 seconds

```
2019-6-2 15:34:05 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'undefined' (0x000d6f00104f246e) of endpoint 1
2019-6-2 15:34:05 - info: New device 'undefined' with address 0x000d6f00104f246e connected!
2019-6-2 15:34:05 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-2 15:34:05 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f00104f246e","meta":{}}'
2019-6-2 15:34:05 - warn: Device with modelID 'undefined' is not supported.
2019-6-2 15:34:05 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-6-2 15:34:07 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-2 15:34:07 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_action","unique_id":"0x000d6f00104f246e_action_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-2 15:34:07 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_battery","unique_id":"0x000d6f00104f246e_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-2 15:34:07 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_linkquality","unique_id":"0x000d6f00104f246e_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbe [... cut ...]
2019-6-2 15:34:33 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
```

What do the following lines mean?

```
2019-6-2 15:34:29 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":271,"zoneStatus":0,"iasCieAddr":"0x0000000000000000","zoneId":255}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-2 15:34:31 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"65533":1}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
```

bambouzel, 2019-06-04:
I cannot seem to get this remote working
Starting over with the simple configuration, no configure part
```
    {
        zigbeeModel: ['RC-EM'],
        model: 'HS1RC-E',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        supports: 'action',
        fromZigbee: [
            fz.battery_200, fz.ignore_power_change,
            fz.HS1RC_E_arm, fz.heiman_smart_controller_emergency,
        ],
        toZigbee: [],
    },
```
```
    HS1RC_E_arm: {
        cid: 'ssIasAce',
        type: 'cmdArm',
        convert: (model, msg, publish, options) => {
            const action = msg.data.data['armmode'];
            delete msg.data.data['armmode'];
            const modeLookup = {
                0: 'unlock',
                1: 'home',
                3: 'lock',
            };
            return {action: modeLookup[action]};
        },
    },
```
Pressing reset for 2 seconds makes the remote trying to join
```
2019-6-4 18:59:28 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'undefined' (0x000d6f00104f246e) of endpoint 1
2019-6-4 18:59:28 - info: New device 'undefined' with address 0x000d6f00104f246e connected!
2019-6-4 18:59:28 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-4 18:59:28 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f00104f246e","meta":{}}'
2019-6-4 18:59:28 - warn: Device with modelID 'undefined' is not supported.
2019-6-4 18:59:28 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-6-4 18:59:30 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-4 18:59:30 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_action","unique_id":"0x000d6f00104f246e_action_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-4 18:59:30 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_battery","unique_id":"0x000d6f00104f246e_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-4 18:59:30 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_linkquality","unique_id":"0x000d6f00104f246e_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-4 18:59:36 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"manufacturerName":"HEIMAN"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-4 18:59:38 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"65533":1,"modelId":"RC-EM","dateCode":"2017.11.1","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-4 18:59:38 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-6-4 18:59:45 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"65533":1,"batteryVoltage":27,"batteryPercentageRemaining":107}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-4 18:59:45 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":53.5,"linkquality":0}'
2019-6-4 18:59:45 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-6-4 18:59:52 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-4 18:59:52 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00104f246e"'
2019-6-4 18:59:59 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":271,"zoneStatus":0,"iasCieAddr":"0x0000000000000000","zoneId":255}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
[... cut ...]
2019-6-4 19:07:22 - debug: Received zigbee message of type 'cmdEmergency' with data '{"cid":"ssIasAce","data":{}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-6-4 19:07:22 - debug: device: 'undefined' 0x00124b0018e1a819
2019-6-4 19:07:22 - debug: Ignoring message from coordinator
```
The message gets ignored because it comes from `0x00124b0018e1a819` which is the coordinator and not the device
I need some help ;)
The closest post with sowewhat the same problem  description was [https://github.com/Koenkk/zigbee2mqtt/issues/801](url)

bambouzel, 2019-06-05:
@tb-killa I'm running out of ideas, do you have some more stuff I could try out?  I don't understand why the battery level comes from the device of endpoint 1 while the button clicks seems to come from the coordinator of endpoint 11.

bambouzel, 2019-06-07:
@Koenkk I'm running out of ideas and getting desperate ;) Hopefully you can have a look and come up with some ideas.
I have an alternative configuration with the configure part which you can look at [https://pastebin.com/GCT6GbR0](url)
I don't understand why the battery and link quality come in ok while the buttons come in via the coordinator on another endpoint
Hopefully you can shed some light
--Ivo

tb-killa, 2019-06-08:
EP 11 are from our ``zapp``.
``debug: Mounted the cieApp (epId 11)``.
This help to get the ``IAS`` Devices more integrated.

modify your ``devices.js`` entry with this part and restart with debug and provide logfile of pairing and button push

```
configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('ssIasZone', coordinator, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];

            execute(device, actions, callback, 250);
        },
```

bambouzel, 2019-06-08:
```
2019-6-8 19:23:06 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-08.19-23-06'
2019-6-8 19:23:06 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
2019-6-8 19:23:07 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-6-8 19:23:07 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-8 19:23:07 - info: Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
2019-6-8 19:23:07 - info: Starting zigbee-shepherd
2019-6-8 19:23:09 - info: zigbee-shepherd started
2019-6-8 19:23:09 - info: Coordinator firmware version: '20180815'
2019-6-8 19:23:09 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0018e1a819","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1560014589,"joinTimeLeft":0}
2019-6-8 19:23:09 - info: Currently 0 devices are joined:
2019-6-8 19:23:09 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-6-8 19:23:09 - warn: Allowing new devices to join.
2019-6-8 19:23:09 - warn: Set `permit_join` to `false` once you joined all devices.
2019-6-8 19:23:09 - info: Zigbee: allowing new devices to join.
2019-6-8 19:23:09 - info: Connecting to MQTT server at mqtt://localhost
2019-6-8 19:23:09 - info: zigbee-shepherd ready
2019-6-8 19:23:09 - info: Connected to MQTT server
2019-6-8 19:23:09 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-6-8 19:23:09 - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"927c4db","coordinator":20180815,"log_level":"debug","permit_join":true}'
2019-6-8 19:23:09 - debug: Mounted the cieApp (epId 11)
2019-6-8 19:23:44 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'undefined' (0x000d6f00104f246e) of endpoint 1
2019-6-8 19:23:44 - info: New device 'undefined' with address 0x000d6f00104f246e connected!
2019-6-8 19:23:44 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-8 19:23:44 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f00104f246e","meta":{}}'
2019-6-8 19:23:44 - warn: Device with modelID 'undefined' is not supported.
2019-6-8 19:23:44 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-6-8 19:23:46 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-8 19:23:46 - debug: Configuring 0x000d6f00104f246e (0x000d6f00104f246e) ...
2019-6-8 19:23:46 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_action","unique_id":"0x000d6f00104f246e_action_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-8 19:23:46 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_battery","unique_id":"0x000d6f00104f246e_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-8 19:23:46 - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f00104f246e/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f00104f246e","json_attributes_topic":"zigbee2mqtt/0x000d6f00104f246e","name":"0x000d6f00104f246e_linkquality","unique_id":"0x000d6f00104f246e_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f00104f246e","name":"0x000d6f00104f246e","sw_version":"Zigbee2mqtt 1.4.0","model":"Smart remote controller (HS1RC-E)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-6-8 19:23:49 - warn: Failed to configure 0x000d6f00104f246e (0x000d6f00104f246e) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
2019-6-8 19:23:49 - warn: This can be ignored if the device is working properly
2019-6-8 19:23:54 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"manufacturerName":"HEIMAN"}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
[... cut ...]
2019-6-8 19:31:20 - debug: Ignoring message from coordinator
2019-6-8 19:31:31 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":0}}' of device 'undefined' (0x00124b0018e1a819) of endpoint 11
2019-6-8 19:31:31 - debug: device: 'undefined' 0x00124b0018e1a819
2019-6-8 19:31:31 - debug: Ignoring message from coordinator
2019-6-8 19:31:31 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6f00104f246e"' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1

```

bambouzel, 2019-06-08:
database contains:

```
{
	"id": 1,
	"type": "Coordinator",
	"ieeeAddr": "0x00124b0018e1a819",
	"nwkAddr": 0,
	"manufId": 0,
	"epList": [
		1,
		2,
		3,
		4,
		5,
		6,
		11
	],
	"status": "online",
	"joinTime": 1560014589,
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 5,
			"inClusterList": [],
			"outClusterList": [],
			"clusters": {}
		},
		"2": {
			"profId": 257,
			"epId": 2,
			"devId": 5,
			"inClusterList": [],
			"outClusterList": [],
			"clusters": {}
		},
		"3": {
			"profId": 261,
			"epId": 3,
			"devId": 5,
			"inClusterList": [],
			"outClusterList": [],
			"clusters": {}
		},
		"4": {
			"profId": 263,
			"epId": 4,
			"devId": 5,
			"inClusterList": [],
			"outClusterList": [],
			"clusters": {}
		},
		"5": {
			"profId": 264,
			"epId": 5,
			"devId": 5,
			"inClusterList": [],
			"outClusterList": [],
			"clusters": {}
		},
		"6": {
			"profId": 265,
			"epId": 6,
			"devId": 5,
			"inClusterList": [],
			"outClusterList": [],
			"clusters": {}
		},
		"11": {
			"profId": 260,
			"epId": 11,
			"devId": 1024,
			"inClusterList": [],
			"outClusterList": [
				1280
			],
			"clusters": {
				"ssIasZone": {
					"dir": {
						"value": 2
					},
					"cmdRsps": {
						"enrollReq": {
							"exec": "_exec_"
						},
						"statusChangeNotification": {
							"exec": "_exec_"
						}
					}
				}
			}
		}
	},
	"_id": "XAFCOtCIIAUDayIB"
}

{
	"id": 2,
	"type": "EndDevice",
	"ieeeAddr": "0x000d6f00104f246e",
	"nwkAddr": 26058,
	"manufId": 4619,
	"manufName": "HEIMAN",
	"powerSource": "Battery",
	"modelId": "RC-EM",
	"epList": [
		1
	],
	"zclVersion": 2,
	"appVersion": 16,
	"stackVersion": 2,
	"hwVersion": 16,
	"status": "online",
	"joinTime": 1560014622,
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 1025,
			"inClusterList": [
				0,
				1,
				3,
				1280
			],
			"outClusterList": [
				3,
				1281
			],
			"clusters": {
				"genBasic": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"zclVersion": 2,
						"appVersion": 16,
						"stackVersion": 2,
						"hwVersion": 16,
						"manufacturerName": "HEIMAN",
						"modelId": "RC-EM",
						"dateCode": "2017.11.1",
						"powerSource": 3
					}
				},
				"genPowerCfg": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"batteryVoltage": 27,
						"batteryPercentageRemaining": 121
					}
				},
				"genIdentify": {
					"dir": {
						"value": 3
					},
					"attrs": {
						"65533": 1,
						"identifyTime": 0
					}
				},
				"ssIasZone": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"zoneState": 0,
						"zoneType": 271,
						"zoneStatus": 0,
						"iasCieAddr": "0x0000000000000000",
						"zoneId": 255
					}
				},
				"ssIasAce": {
					"dir": {
						"value": 2
					},
					"attrs": {}
				}
			}
		}
	},
	"dateCode": "2017.11.1",
	"_id": "SpIehKdg2K8RqfWV"
}
```

Koenkk, 2019-06-09:
The latest firmware might fixed your ep11 problem, newer firmwares forward messages to this endpoint to endpoint 1 (I see you are running `20180815` now).

bambouzel, 2019-06-10:
Update firmware to latest, that did the job

```
2019-6-10 09:52:51 - debug: Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'RC-EM' (0x000d6f00104f246e) of endpoint 1
2019-6-10 09:52:51 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":61.5,"linkquality":39,"action":"lock"}'
2019-6-10 09:52:51 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":61.5,"linkquality":39,"action":""}'
```

Thank you!

closed 2019-06-10 by bambouzel

ccorderod, 2019-06-10:
@bambouzel Would you please let me know which firmware rev you are using? I am really struggling with Zigbee2Mqtt; I have tried all the firmwares I have been able to test with z2m 1.4.0; my setup is big (50+ devices, 15 routers among them), and not reliable (ie, HA shows my front door open as I write, and it is not!). Thank you. have a great day.

bambouzel, 2019-06-10:
@ccorderod I just download the lastest from [https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/CC2531_20190425.zip](url).
What kind of door sensor are you using?  Did you already take a look at the debug logging to see what z2m is getting in?

ccorderod, 2019-06-10:
@bambouzel Hmmm. I am running beta branch 20190608 firmware and beta branch 1.4.0 z2m (I am desperate). I have 12 meazon meters (routers), 2 xbee s2c (just routing), 1 diy router, 10 aqara water leak, 14 aqara/xiaomi door/window sensor, 1 xiaomi gas sensor, 4 aqara vibration, 2 xiaomi cubes, 3 hue remote  dimmers, 2 nyce motion sensor, 1 nyce contact sensor, 1 ikea dimmer, 1 ikea remote as of now.
I did spent countless hours analyzing shepherd and z2m logs ... but I can't find a way to stabilize my setup and make it reliable.

bambouzel, 2019-06-10:
@ccorderod  I also have the Xiaomi door sensors and they worked for me.  The problem described on top was with my Heiman remote.  If I were you I would stop the HA service, stop the zigbee2mqtt service if you have one, start zigbee2mqtt in debug mode, wait forit to settle down and then make your sensor trigger by opening the door and then copy the output here in a new issue.  Then we can have a look and find out what is going wrong.

bambouzel, 2019-06-10:
@tb-killa @Koenkk Also removed the configure part from the device and all still seems ok!

kijuz, 2019-08-08:
hi all. i have the same remote control heiman rc-em.
i use the above code but nothing happens. can you share please the converters and the device files?  many thanks

kijuz, 2019-08-08:
zigbee2mqtt:debug 8/8/2019, 2:04:55 PM Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":114}}' of device 'RC-EM' (0x000d6f00104f50e5) of endpoint 1
  zigbee2mqtt:warn 8/8/2019, 2:04:55 PM No converter available for 'HS1RC-E' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":114}}'
  zigbee2mqtt:warn 8/8/2019, 2:04:55 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 8/8/2019, 2:04:55 PM Received zigbee message of type 'devInterview' with data '"0x000d6f00104f50e5"'
  zigbee2mqtt:debug 8/8/2019, 2:04:59 PM Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'RC-EM' (0x000d6f00104f50e5) of endpoint 1
  zigbee2mqtt:debug 8/8/2019, 2:04:59 PM Received zigbee message of type 'devInterview' with data '"0x000d6f00104f50e5"'
  zigbee2mqtt:debug 8/8/2019, 2:05:05 PM Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":271,"zoneStatus":0,"iasCieAddr":"0x0000000000000000","zoneId":255}}' of device 'RC-EM' (0x000d6f00104f50e5) of endpoint 1
  zigbee2mqtt:debug 8/8/2019, 2:05:07 PM Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"65533":1}}' of device 'RC-EM' (0x000d6f00104f50e5) of endpoint 1
  zigbee2mqtt:debug 8/8/2019, 2:05:07 PM Received zigbee message of type 'devInterview' with data '"0x000d6f00104f50e5"'
  zigbee2mqtt:debug 8/8/2019, 2:05:09 PM Received zigbee message of type 'devInterview' with data '"0x000d6f00104f50e5"'
  zigbee2mqtt:debug 8/8/2019, 2:05:09 PM Accept device incoming with ieeeAddr '0x000d6f00104f50e5' permit join is 'true'
  zigbee2mqtt:debug 8/8/2019, 2:05:09 PM Allowing device '0x000d6f00104f50e5' to join
  zigbee2mqtt:debug 8/8/2019, 2:05:09 PM Received zigbee message of type 'devIncoming' with data '"0x000d6f00104f50e5"' of device 'RC-EM' (0x000d6f00104f50e5) of endpoint 1
  zigbee2mqtt:info 8/8/2019, 2:05:09 PM Device incoming...
  zigbee2mqtt:info 8/8/2019, 2:05:09 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 8/8/2019, 2:05:09 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'RC-EM' (0x000d6f00104f50e5) of endpoint 1

kijuz, 2019-08-08:
please help

kijuz, 2019-08-08:
if i insert also the fromzigbee.js file part:

zigbee2mqtt:debug 8/8/2019, 2:30:34 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"RC-EM","powerSource":3}}' of device 'RC-EM' (0x000d6f00104f50e5) of endpoint 1
/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^
TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/zigbee2mqtt-1.5.1/lib/controller.js:143:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/controller.js:143:14)
    at Zigbee.onMessage (/zigbee2mqtt-1.5.1/lib/zigbee.js:275:18)
    at ZShepherd.emit (events.js:198:13)
    at ZShepherd.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/shepherd.js:148:14)
    at ZShepherd.emit (events.js:198:13)

bambouzel, 2019-08-08:
Hi kijuz,

This is what currently is running at my HomeAssistant:

Make sure you are running the correct firmware on your USB receiver.  I'm still running: [CC2531_20190425.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/CC2531_20190425.zip).

**/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js**
   1214     **HS1RC_E_arm**: {
   1215         cid: 'ssIasAce',
   1216         type: 'cmdArm',
   1217         convert: (model, msg, publish, options) => {
   1218             const action = msg.data.data['armmode'];
   1219             delete msg.data.data['armmode'];
   1220             const modeLookup = {
   1221                 0: 'unlock',
   1222                 1: 'home',
   1223                 3: 'lock',
   1224             };
   1225             return {action: modeLookup[action]};
   1226         },
   1227     },

**/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/devices.js**
   3182     {
   3183         zigbeeModel: ['RC-EM'],
   3184         model: '**HS1RC-E**',
   3185         vendor: 'HEIMAN',
   3186         description: 'Smart remote controller',
   3187         supports: 'action',
   3188         fromZigbee: [
   3189             fz.battery_200, fz.ignore_power_change, fz.ignore_iaszone_change,
   3190             fz.**HS1RC_E_arm**, fz.heiman_smart_controller_emergency,
   3191         ],
   3192         toZigbee: [],
   3193     },

**/opt/zigbee2mqtt/lib/extension/homeassistant.js**
    663     '**HS1RC-E**': [configurations.sensor_action, configurations.sensor_battery],

kijuz, 2019-08-08:
hi bambouzel.
thanks for you help. 
i'm running the latest firmware  coordinator firmware version: '20190608'.
i'm running hassio and zigbee2mqtt sw 1.5.1.2.
i put devices.js under share/zigbee2mqtt and fromZigbee.js to share/zigbee2mqtt/converters/
i don't know where to put the home assistant.js

bambouzel, 2019-08-08:
Here are some links for adding new devices to hassio:

[https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html)
[https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)

> 4. (Optional) Add home assistant configuration for your device
> In order to automatically discover this device in home assistant your device needs to be added to > mapping in lib/extension/homeassistant.js.

kijuz, 2019-08-08:
thanks.
i followed the guide for hassio. i had modified the file as you indicated above but the problem persists. zigbee2mqtt crashs with the same error:

zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:109
if (c.cid === cid) {
^
TypeError: Cannot read property 'cid' of undefined

One question: you make a pull request? if yes how much time pass before we can use your configuration?

thanks

bambouzel, 2019-08-09:
kijuz,

Unfortunately I do not use Hass.io.  I'm using RPi3 with Stretch on it and running Homeassistant in a virtual python environment.  I couldn't use Hss.io because I had to make changes to the serial ports.

So I don't use the Z2M plugin as you do.
The place for issues with the plugin really is [https://github.com/danielwelch/hassio-zigbee2mqtt](https://github.com/danielwelch/hassio-zigbee2mqtt).

I hope you find your answer there.

VenkateshGSRao, 2019-11-24:
@bambouzel / @tb-killa / @Koenkk ,
I am also facing same kind of issue with this remote i bought recently.
Since last 2 days I tried everything described by you. nothing seems to be working.
I am not using home assistant but instead i am building my own dashboard/mobile app to view/control devices.

I have raised new issue:
https://github.com/Koenkk/zigbee2mqtt/issues/2409

It will be great if you can help me resolve the issue.
Thanks in advance.