Koenkk/zigbee2mqtt #1582: Continuous reporting of the device list

orek78, 2019-05-28:
Hi guys,
Is it possible to disable continuous reporting available devices list from bridge?
Logs are very cluttered.

`2019-05-29 00:18:01.970 (Zignee2MQTT) Received available devices list from bridge
2019-05-29 00:18:01.970 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000237133f (Temp.gabinet)
2019-05-29 00:18:01.971 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fe27 (Temp.Pokoj Zuzi)
2019-05-29 00:18:01.972 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327708 (Temp.Sypialnia)
2019-05-29 00:18:01.973 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d00027725cd (Temp.Lazienka mala)
2019-05-29 00:18:01.974 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fede (Temp.Holl)
2019-05-29 00:18:01.975 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233f903 (Temp.Wiatrolap)
2019-05-29 00:18:01.977 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327582 (Temp.Salon)
2019-05-29 00:18:01.978 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002725890 (Temp.Lazienka Duza)
2019-05-29 00:18:01.979 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e890b4 (Przycisk kuchnia)
2019-05-29 00:18:01.980 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002775df3 (Czujnik zewnetrzny)
2019-05-29 00:18:01.982 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e793a0 (Przycisk wiatrolap)
2019-05-29 00:18:01.983 (Zignee2MQTT) Device CC2530.ROUTER 0x00124b001c3244c8 (Router1)
2019-05-29 00:18:05.994 (Zignee2MQTT) Received available devices list from bridge
2019-05-29 00:18:05.995 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000237133f (Temp.gabinet)
2019-05-29 00:18:05.996 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fe27 (Temp.Pokoj Zuzi)
2019-05-29 00:18:05.997 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327708 (Temp.Sypialnia)
2019-05-29 00:18:05.998 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d00027725cd (Temp.Lazienka mala)
2019-05-29 00:18:06.000 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fede (Temp.Holl)
2019-05-29 00:18:06.001 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233f903 (Temp.Wiatrolap)
2019-05-29 00:18:06.003 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327582 (Temp.Salon)
2019-05-29 00:18:06.004 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002725890 (Temp.Lazienka Duza)
2019-05-29 00:18:06.006 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e890b4 (Przycisk kuchnia)
2019-05-29 00:18:06.007 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002775df3 (Czujnik zewnetrzny)
2019-05-29 00:18:06.009 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e793a0 (Przycisk wiatrolap)
2019-05-29 00:18:06.011 (Zignee2MQTT) Device CC2530.ROUTER 0x00124b001c3244c8 (Router1)
2019-05-29 00:18:09.972 (Zignee2MQTT) Received available devices list from bridge
2019-05-29 00:18:09.972 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000237133f (Temp.gabinet)
2019-05-29 00:18:09.973 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fe27 (Temp.Pokoj Zuzi)
2019-05-29 00:18:09.974 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327708 (Temp.Sypialnia)
2019-05-29 00:18:09.975 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d00027725cd (Temp.Lazienka mala)
2019-05-29 00:18:09.977 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fede (Temp.Holl)
2019-05-29 00:18:09.978 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233f903 (Temp.Wiatrolap)
2019-05-29 00:18:09.979 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327582 (Temp.Salon)
2019-05-29 00:18:09.980 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002725890 (Temp.Lazienka Duza)
2019-05-29 00:18:09.982 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e890b4 (Przycisk kuchnia)
2019-05-29 00:18:09.983 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002775df3 (Czujnik zewnetrzny)
2019-05-29 00:18:09.985 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e793a0 (Przycisk wiatrolap)
2019-05-29 00:18:09.985 (Zignee2MQTT) Device CC2530.ROUTER 0x00124b001c3244c8 (Router1)
2019-05-29 00:18:14.003 (Zignee2MQTT) Received available devices list from bridge
2019-05-29 00:18:14.004 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000237133f (Temp.gabinet)
2019-05-29 00:18:14.005 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fe27 (Temp.Pokoj Zuzi)
2019-05-29 00:18:14.006 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327708 (Temp.Sypialnia)
2019-05-29 00:18:14.007 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d00027725cd (Temp.Lazienka mala)
2019-05-29 00:18:14.008 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fede (Temp.Holl)
2019-05-29 00:18:14.009 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233f903 (Temp.Wiatrolap)
2019-05-29 00:18:14.011 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327582 (Temp.Salon)
2019-05-29 00:18:14.012 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002725890 (Temp.Lazienka Duza)
2019-05-29 00:18:14.014 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e890b4 (Przycisk kuchnia)
2019-05-29 00:18:14.014 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002775df3 (Czujnik zewnetrzny)
2019-05-29 00:18:14.016 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e793a0 (Przycisk wiatrolap)
2019-05-29 00:18:14.017 (Zignee2MQTT) Device CC2530.ROUTER 0x00124b001c3244c8 (Router1)
2019-05-29 00:18:17.999 (Zignee2MQTT) Received available devices list from bridge
2019-05-29 00:18:18.000 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000237133f (Temp.gabinet)
2019-05-29 00:18:18.001 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fe27 (Temp.Pokoj Zuzi)
2019-05-29 00:18:18.002 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327708 (Temp.Sypialnia)
2019-05-29 00:18:18.003 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d00027725cd (Temp.Lazienka mala)
2019-05-29 00:18:18.004 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fede (Temp.Holl)
2019-05-29 00:18:18.005 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233f903 (Temp.Wiatrolap)
2019-05-29 00:18:18.007 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327582 (Temp.Salon)
2019-05-29 00:18:18.008 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002725890 (Temp.Lazienka Duza)
2019-05-29 00:18:18.010 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e890b4 (Przycisk kuchnia)
2019-05-29 00:18:18.011 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002775df3 (Czujnik zewnetrzny)
2019-05-29 00:18:18.013 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e793a0 (Przycisk wiatrolap)
2019-05-29 00:18:18.014 (Zignee2MQTT) Device CC2530.ROUTER 0x00124b001c3244c8 (Router1)
2019-05-29 00:18:22.024 (Zignee2MQTT) Received available devices list from bridge
[... 195 lines cut ...]
2019-05-29 00:19:22.074 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327708 (Temp.Sypialnia)
2019-05-29 00:19:22.075 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d00027725cd (Temp.Lazienka mala)
2019-05-29 00:19:22.076 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fede (Temp.Holl)
2019-05-29 00:19:22.077 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233f903 (Temp.Wiatrolap)
2019-05-29 00:19:22.079 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327582 (Temp.Salon)
2019-05-29 00:19:22.080 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002725890 (Temp.Lazienka Duza)
2019-05-29 00:19:22.081 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e890b4 (Przycisk kuchnia)
2019-05-29 00:19:22.082 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002775df3 (Czujnik zewnetrzny)
2019-05-29 00:19:22.084 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e793a0 (Przycisk wiatrolap)
2019-05-29 00:19:22.085 (Zignee2MQTT) Device CC2530.ROUTER 0x00124b001c3244c8 (Router1)
2019-05-29 00:19:26.098 (Zignee2MQTT) Received available devices list from bridge
2019-05-29 00:19:26.100 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000237133f (Temp.gabinet)
2019-05-29 00:19:26.103 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fe27 (Temp.Pokoj Zuzi)
2019-05-29 00:19:26.106 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327708 (Temp.Sypialnia)
2019-05-29 00:19:26.109 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d00027725cd (Temp.Lazienka mala)
2019-05-29 00:19:26.112 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233fede (Temp.Holl)
2019-05-29 00:19:26.115 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d000233f903 (Temp.Wiatrolap)
2019-05-29 00:19:26.118 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002327582 (Temp.Salon)
2019-05-29 00:19:26.121 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002725890 (Temp.Lazienka Duza)
2019-05-29 00:19:26.125 (Zignee2MQTT) Device WXKG11LM 0x00158d0001e890b4 (Przycisk kuchnia)
2019-05-29 00:19:26.127 (Zignee2MQTT) Device WSDCGQ11LM 0x00158d0002775df3 (Czujnik zewnetrzny)
`
## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2019-05-29:
Can you post the zigbee2mqtt log?

roygbiv856, 2019-05-29:
I'm having the same problem. It does a "network scan" about every 30 seconds. I thought it might be from the recently installed the Node Red Zigbee2MQTT dashboard

orek78, 2019-05-29:
This is it! A few days ago I installed the Node Red Zigbee2MQTT dashboard. I checked the logs yesterday and did not associate this with Node Red. I switched off the flow and everything went back to normal. Thanks!

closed 2019-05-29 by Koenkk