Koenkk/zigbee2mqtt #1585: NOT Issue - but strange - "info: succesfully reenabled joining

vexter0944, 2019-05-30:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

davekr8, 2019-05-31:
I was wondering about this messages, too. They seems to appear in less den 3-4 minutes interval.

SebiGo, 2019-06-01:
I think, this comment from zigbee.js:184 explains why:
`        // In zigbee 3.0 a network automatically closes after 254 seconds.
        // As a workaround, we enable joining again.
`
To disable the message, see the [configuration documentation](https://www.zigbee2mqtt.io/configuration/configuration.html):

> # Required: allow new devices to join.
> # WARNING: Disable this after all devices have been paired!
> permit_join: true

You can also disable joining through MQTT, see the [message structure description](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html) in chapter "zigbee2mqtt/bridge/config/permit_join".

It's generally a good idea to disable joining, unless you want to join your devices. Otherwise you might end up with your neighbors devices or someone might mess with your zigbee network :)

Koenkk, 2019-06-03:
@SebiGo that is indeed the reason, closing this as this is expected behaviour.

closed 2019-06-03 by Koenkk