Koenkk/zigbee2mqtt #1586: RTCGQ11LM unknown

sonuyos, 2019-05-30:
# Bug Report
## What happened
I have 2 RTCGQ11LM, up until today morning, both were working correct, but then i added a USB extender and both stopped working, after a lot of R&D, both started working, but now only 1 works and 1 doesnt.

## What did you expect to happen
Nothing, just usual process.

## How to reproduce it (minimal and precise)
I am not really sure, after adding the extender, i put it back into the PC to test, it still doesnt work.

## Debug Info
zigbee2mqtt version:1.4.0
CC2531 firmware version:24th April.

The log shows that both have connected, but only 1 sends signal.

My Log - https://pastebin.com/imd18BND

0x00158d000302c5f4 - Doesnt work
0x00158d0003040926  - works.

I did a full reinstall too of the plugin. I am using Home Assitant.

Home Assistant only shows this - http://prntscr.com/nvjat5

Or Unknown.

sonuyos, 2019-05-30:
Now after reinstalling Z2M, both are not connecting.

sonuyos, 2019-05-31:
I left it overnight and slept, got up and both successfully connected, but neither are sending or receiving data.

sonuyos, 2019-06-01:
Now, it worked somehow for past 2 days, and now 1 stopped connecting.

This is my database.

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1af07","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1559415328,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"ii2mdXv9mYv3l1AB"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0003040926","nwkAddr":34884,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"zclVersion":1,"appVersion":5,"stackVersion":2,"hwVersion":1,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"20170627","_id":"jhcGh9YySUTEs9Fl"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000302c5f4","nwkAddr":27269,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"zclVersion":113,"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"zGvqioYOd5hTryDx"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1af07","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1559415337,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"ii2mdXv9mYv3l1AB"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0003040926","nwkAddr":34884,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"zclVersion":1,"appVersion":5,"stackVersion":2,"hwVersion":1,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"20170627","_id":"jhcGh9YySUTEs9Fl"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000302c5f4","nwkAddr":27269,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"zclVersion":113,"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"zGvqioYOd5hTryDx"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1af07","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1559415337,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"d [... cut ...] Id":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"ii2mdXv9mYv3l1AB"}
```

Koenkk, 2019-06-03:
Do you see a message with `log_level: debug` (http://www.zigbee2mqtt.io/configuration/configuration.html) when pressing the reset button on the RTCGQ11LM once?

sonuyos, 2019-06-03:
@Koenkk i have some how sorted it, but the connection is really weak, it works it works, if it doesnt, it doesnt, and its for strange reason, no movement of coordinator or enddevice, it just lose connection, sometime its 32-45% singal, something its 0 or 2%.

Its very fickly, my zigbee is on channel 24 and wifi on channel 1.

closed 2019-08-09 by stale[bot]