Koenkk/zigbee2mqtt #1592: GBAN GB2530 "AF data request fails"

b4dpxl, 2019-06-02:
zigbee2mqtt version: 1.4.0 (commit #6b75465)
CC253X firmware version: 20190515

I recently upgraded my CC2531 (FW: 20190425) to a CC2530 GBAN GB2530-S. At first everything was working great, and link quality seemed much improved. But today some (but not all) of my Xiaomi switches stopped working. In the logs I was seeing:
`Error: AF data request fails, status code: 183. APS no ack.`

Restarting z2m made no difference, but swapping back to the CC2531 made things work again. I didn't repair after swapping between the devices, as everything was working perfectly. 

Could this be a firmware version issue, or something else? Is there another reason why the device with better signal drops out? I've only got 6 devices, all Xiaomi WXKG0xLM or QBKG12LM.

Thanks.

pixeldoc2000, 2019-07-16:
zigbee2mqtt version: 1.4.0 (commit #6b75465)
CC253X firmware version: 20190608

My GBAN GB2530-S is working great. Using it since 1/2 year.

Maybe reflash the device with the current fw?
Maybe hw problem?

b4dpxl, 2019-07-17:
My bad, I should have closed it. After a bit of a flaky start and a few re-pairs, everything settled down and it's working nicely now. Tho I did downgrade the firmware 1 version on the dongle.

closed 2019-07-17 by b4dpxl