Koenkk/zigbee2mqtt #1593: Ikea Tradfri remote control E1524 fires empty action attribute when pressed

wilfredsmit, 2019-06-03:
Ikea Tradfri remote control E1524 fires empty action attribute when pressed, instead of for example 'toggle'.

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190223

`zigbee2mqtt:info 6/3/2019, 8:36:32 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe38f4ad', payload '{"battery":87,"linkquality":44,"action":"","device":{"ieeeAddr":"0x000b57fffe38f4ad","friendlyName":"0x000b57fffe38f4ad","type":"EndDevice","nwkAddr":47943,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","hwVersion":1,"swBuildId":"1.2.214","dateCode":"20170302","status":"offline"}}`

Koenkk, 2019-06-03:
Probably the same as https://github.com/Koenkk/zigbee2mqtt/issues/1583#issuecomment-496800585

Koenkk, 2019-06-10:
I will close this one as it is expected behaviour when homeassistant integration is enabled.

closed 2019-06-10 by Koenkk