Koenkk/zigbee2mqtt #1594: Xiaomi Wall Switches (QBKG03LM, QBKG04LM) are detected as routers which they are not

VivantSenior, 2019-06-03:
Hi,

I have noticed that QBKG03LM and QBKG04LM are not routing the network. I have found explanation [here](https://en.miui.com/forum.php?mod=viewthread&tid=2058422&page=1#pid23859766).

**The Xiaomi Switches without neutral line are not working as routers which I can confirm based on my network map.**

@Koenkk could you please add this information to firmware?
I have created two PRs to the wiki to add this information.

Related to: [#404](https://github.com/Koenkk/zigbee2mqtt/issues/404).

Thanks,
Viv

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190425 CC2531

Koenkk, 2019-06-04:
Fixed in the dev branch, can you verify?

joshdinsdale, 2019-06-07:
Has made my network much more stable and explains why I've been having so many issues! One thing to note, the NodeRed dashboard still lists them as routers for some reason.

Nephiel, 2019-06-08:
I too was wondering if it was right to identify them as routers - see https://github.com/Koenkk/zigbee2mqtt/issues/943#issuecomment-491527487.
Thank you for addressing this.

closed 2019-08-15 by stale[bot]