Koenkk/zigbee2mqtt #1595: Osram AA69697 RGBW bulbs not recognized

mk1blacklimited, 2019-06-03:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

mk1blacklimited, 2019-06-03:
Trying to add 3 identical Osram RGBW AA69697 bulbs on a new zigbee2mqtt setup on an RPi. They are not recognized even if the AA69697 bulb is defined in devices.js

I get

`zigbee2mqtt:warn 6/3/2019, 2:15:17 PM Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn 6/3/2019, 2:15:17 PM Device address '0x8418260000c9fced'`

...when they are added. database.db shows (for one bulb):

`{"id":8,"type":"Router","ieeeAddr":"0x8418260000c9fced","nwkAddr":38936,"manufId":48042,"epList":[3],"status":"online","joinTime":155956$
{"id":8,"type":"Router","ieeeAddr":"0x8418260000c9fced","nwkAddr":38936,"manufId":48042,"epList":[3],"zclVersion":1,"appVersion":3,"stac$
`

I've added a couple of other non-RGBW Osram bulbs and and a Xiaomi motion sensor with no problems at all.

Any pointers?

JackGruber, 2019-06-04:
I had the last time the same behaviour with a Osram light stripe. 
It was first recognized as undefined and after after approx. 10 minutes the deviceID was correctly recognised.

Martin9239, 2019-07-19:
Pretty similar problem two osram bulbs (same typ) and yet the bulbs are being recognized differently:

{"state":"OFF","linkquality":0,"brightness":0,"device":{"ieeeAddr":"0xf0d1b8000010caaf","friendlyName":"SZ-Nachtlampe-L","type":"Router","nwkAddr":30091,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"CLA60 RGBW OSRAM","hwVersion":"unknown","swBuildId":"unknown","dateCode":"unknown","status":"offline"}}

{"state":"OFF","linkquality":0,"color":{"x":0.307,"y":0.305,"saturation":100},"color_temp":370,"color_mode":1,"brightness":0,"device":{"ieeeAddr":"0xf0d1b8000010d094","friendlyName":"SZ-Nachtlampe-R","type":"Router","nwkAddr":15546,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"CLA60 RGBW OSRAM","hwVersion":"unknown","swBuildId":"unknown","dateCode":"unknown","status":"offline"}}

What can i do to fix that?

closed 2019-09-24 by stale[bot]