Koenkk/zigbee2mqtt #1596: Xiaomi Aqara sensors not responding to 'get' command

tenspot, 2019-06-03:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

tenspot, 2019-06-03:
Hi,
I have recently installed a zigbee2mqtt setup and have it working with three devices:
Xiaomi Aqara temperature, humidity and pressure sensor (temperature, humidity and pressure). Model: WSDCGQ11LM
Xiaomi Aqara door & window contact sensor.  Model: MCCGQ11LM
Ikea TRADFRI control outlet. Model: E1603

However, when I try and read the status of either of the Aqara sensors using the 'get' command in MQTT Explorer, I get no response. When I send a 'get' command to the Ikea outlet, I get the device status back.
The commands I have tried are:
    for the door & window contact:
    zigbee2mqtt/deviceID/get/status   payload: empty
    zigbee2mqtt/deviceID/get   payload: {"status": ""}

   temperature, humidity and pressure sensor:
    zigbee2mqtt/deviceID/get/temperature   payload: empty
    zigbee2mqtt/deviceID/get   payload: {"temperature": ""}

CypherMK, 2019-06-03:
I guess it's because these xiaomi sensors are battery powered and only report their status on their own.

Koenkk, 2019-06-04:
@CypherMK is right, the device is probably sleeping when you do the request.

tenspot, 2019-06-04:
Ah yes, never thought of that. I'm new to low power battery tech. 
Many thanks for pointing this out.

closed 2019-06-04 by tenspot