Koenkk/zigbee2mqtt #1597: CC2531 physical buttons

mclaudiopt, 2019-06-04:
Hello
what is the function of the 2 physical buttons present on the cc2531 stick?
thanks

Koenkk, 2019-06-07:
[image]

- S2 is reset button
- S1 I'm not sure about, @ptvoinfo do you know where it is used for by default?

ptvoinfo, 2019-06-08:
The coordinator firmware does not use S1. 

My router  firmware uses these buttons:

S1 – Changes a virtual state of the router (the router sends it in a diagnostic report) and the state of the green light.

S2:
    Short single press – to re-send a binding request to a coordinator (it allows you to test a connection).
    Long single press (about 5 seconds) – to reset the router and initiate a joining process.

Koenkk, 2019-06-10:
Closing this one as the question is answered.

closed 2019-06-10 by Koenkk