Koenkk/zigbee2mqtt #1598: All devices offline

abmantis, 2019-06-04:
# Bug Report
## What happened
Suddenly all devices are disconnected/offline. Restarting Z2M doesn't fix the issue. A few seconds after rebooting, every device fails with
`zigbee2mqtt:error 6/4/2019, 8:58:07 PM Failed to bind for reporting for 'Living Room R' - genOnOff - 1 - (Error: Timed out after 10000 ms)`

I have no idea why this happened. I wasn't even at home when it happened. It is not a signal issue, since the Tradfri Repeater is right next to it and is also lost. Other devices are also close.

My network map looks like this after the issue:

[image]

## Debug Info
zigbee2mqtt version: 1.4.0, dev branch (commit 4497214588b2f5e51e2d18cf0420848c0581d7d5)
CC253X firmware version: 20190223

2114L3, 2019-08-08:
seem to have the same issue. with;
- ubuntu 18.04, 
- Docker 18.06.3-ce on linux, amd64
- zigbee2mqtt version 1.5.1 sha256:b07d4c4bef3dfcd18f4a12c38f50768f26fbf3f979fe20a06deb60daf1e50970
- c2231 usb on CC2531_DEFAULT_20190608
- advanced pinging enabled

works fine for one to a few days then suddenly everything is offline.
restart z2m does nothing, reset button + restart z2m does nothing

```
8/7/2019, 8:10:06 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000001402/availability', payload 'online'
8/7/2019, 8:10:07 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000008832/availability', payload 'online'
8/7/2019, 8:10:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000006cf5/availability', payload 'online'
8/7/2019, 8:10:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000002464/availability', payload 'online'
8/7/2019, 8:10:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000007600/availability', payload 'online'
8/7/2019, 8:10:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000cfb27/availability', payload 'online'
8/7/2019, 8:10:26 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000cb7f/availability', payload 'online'
8/7/2019, 8:10:26 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000ba79/availability', payload 'online'
8/7/2019, 8:10:58 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000005713/availability', payload 'offline'
8/7/2019, 8:11:00 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000005d31/availability', payload 'offline'
8/7/2019, 8:11:13 PM - error: Failed to ping '0x0000000000001402'
8/7/2019, 8:11:13 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000001402/availability', payload 'offline'
8/7/2019, 8:11:17 PM - error: Failed to ping '0x0000000000008832'
8/7/2019, 8:11:17 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000008832/availability', payload 'offline'
8/7/2019, 8:11:23 PM - error: Failed to reenable joining
8/7/2019, 8:11:31 PM - error: Failed to ping '0x0000000000006cf5'
8/7/2019, 8:11:31 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000006cf5/availability', payload 'offline'
8/7/2019, 8:11:35 PM - error: Failed to ping '0x0000000000002464'
8/7/2019, 8:11:35 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000002464/availability', payload 'offline'
8/7/2019, 8:11:35 PM - error: Failed to ping '0x0000000000007600'
8/7/2019, 8:11:35 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000007600/availability', payload 'offline'
8/7/2019, 8:11:36 PM - error: Failed to ping '0x000000000000fb27'
8/7/2019, 8:11:36 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000cfb27/availability', payload 'offline'
8/7/2019, 8:11:36 PM - error: Failed to ping '0x000000000000cb7f'
8/7/2019, 8:11:36 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000cb7f/availability', payload 'offline'
8/7/2019, 8:11:41 PM - error: Failed to ping '0x000000000000ba79'
8/7/2019, 8:11:41 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000ba79/availability', payload 'offline'
8/7/2019, 8:12:08 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000005713/availability', payload 'offline'
8/7/2019, 8:12:10 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000005d31/availability', payload 'offline'
8/7/2019, 8:12:19 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000001402/availability', payload 'offline'
8/7/2019, 8:12:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000008832/availability', payload 'offline'
8/7/2019, 8:12:37 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000006cf5/availability', payload 'offline'
8/7/2019, 8:12:43 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000002464/availability', payload 'offline'
8/7/2019, 8:12:46 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000007600/availability', payload 'offline'
8/7/2019, 8:12:46 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000fb27/availability', payload 'offline'
8/7/2019, 8:12:46 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000cb7f/availability', payload 'offline'
8/7/2019, 8:12:51 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000000000000ba79/availability', payload 'offline'
8/7/2019, 8:13:15 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0000000000005713/availability', payload 'offline'
```

first error on 0x0000000000001402 is a Philips hue 9290011370 https://www.zigbee2mqtt.io/devices/9290011370.html
i have track
i have a mix of xiaomi, philips, tradfri, sengled devices
*addresses masked

resolution is remove and insert USB, and start z2m container (as pulling usb seems to drop the main thread)

previous incident was on the same bulb, which i believe may have been accidentally switched off at the wall.
perhaps there is an issued on the USB firmware when there is a failure to ping at particular points?

2114L3, 2019-08-09:
third event today, same light provoking failure, same method to recover
noted that everything was still working ok while the unit was left off and for an extended peroid. it was on powering **on** the light at the wall switch that z2m failed and all units stated to list as offline
have not tested other bulbs or devices for same effect yet.

closed 2019-10-15 by stale[bot]