Koenkk/zigbee2mqtt #1599: Warnings in HomeAssistant

Edzilla2000, 2019-06-05:
Homeassistant is complaining that I'm using unsupported keys in an mqtt sensor.
It seems it's a configuration that was imported from zigbee2mqtt by the ecozy sensor.
Here is the error I get: 
https://paste.edzilla.info/?001da94e7ae5f2fa#Z16pdhaMUv9pG0j6iJ9h8FoaaUFSSSzNgRizdfRQvuE=

Koenkk, 2019-06-07:
I don't understand this error message, according to https://www.home-assistant.io/components/climate.mqtt/ all keys should be supported, what keys are unsupported?

Edzilla2000, 2019-06-07:
According to the error it's "state_topic" that is no longer supported.
On the previous version, I only had the warning that I had two "state_topic" in my configuration and that it was no longer supported.
This version added the context that I linked above

Koenkk, 2019-06-07:
Missed that, should be fixed in the dev branch, can you confirm?

Edzilla2000, 2019-06-08:
Yes, it works fine now, thanks!

Edzilla2000, 2019-06-08:
I spoke too soon.
I no longer get warnings in the web interface but homeassistant.log is still getting this every 30 secondes;
https://paste.edzilla.info/?cc29f2790dce1a6c#2YUwLOtHaa9l/yb4874yJVaU9o6nWL5NyXE1t6EV9nM=

Koenkk, 2019-06-10:
Is this set somewhere in an automation? The logging provides very little information.

Edzilla2000, 2019-06-16:
No, it's not used in any automation, only the regular discover in HA

clockbrain, 2019-06-17:
I'm seeing similar HA log warnings and in my case have traced it to an empty mqtt cached state message sent on restart of z2m for a device that has lost its connection. The mqtt message sent to state_topic: zigbee2mqtt/0x00158d0002c9865a is simply
```
{}
```
but the equivalent sent at the same time for other working devices is e.g.
```
{"power":0,"current":0,"voltage":236,"power_factor":0,"linkquality":68,"state":"OFF"}
```
Not sure why the cached state message is empty, the database.db entry for the device is ok. 

Fix would somehow be to detect and avoid sending those empty mqtt cached state messages.

Koenkk, 2019-06-19:
@clockbrain fix in the latest dev branch.

closed 2019-08-18 by Koenkk