Koenkk/zigbee2mqtt #1601: Failed to reenable joining

Kevlordofcode, 2019-06-05:
# Bug Report
## What happened

My network (mainly Aqara sensors) suddenly stopped working after several weeks without any problem. Nothing was posted on mqtt anymore.
To fix it I have tried a lots of things, including reseting the cc2531, reflashing it several times, deleting database.db etc.
It is still not working. Zigbee2mqtt runs fine but I cannot pair any device. I have noticed numerous strange entries in the logs:
`error: Failed to reenable joining`

I could not find any other issues mentioning this error so I figured out I would open a new one.
I thank you all in advance for your kind assistance!

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190425
Debug log: https://pastebin.com/0WzhS0qZ

Koenkk, 2019-06-06:
Could you post the log when running with `DEBUG=* npm start`?

Kevlordofcode, 2019-06-06:
Thank you for your quick answer. Here is the requested log: https://pastebin.com/U1qPVPd0

Koenkk, 2019-06-07:
Somehow the CC2531 doesn't respond, can you try with https://github.com/Koenkk/Z-Stack-firmware/raw/e98e0e8c2b5826ad4d29a1046bfeaf78c08b4c53/coordinator/max_stability/CC2531/CC2531_MAX_STABILITY_20190315.zip

Kevlordofcode, 2019-06-07:
Done. Here is the debug log: https://pastebin.com/P4aksVng

Unfortunately it did not solve the problem.

It's like if the stick had become somehow corrupted and it cannot be completely reseted even by reflashing it, is it possible?

Please note that it sometimes manages to reenable the joining but it's only temporary:

```
2019-6-6 15:47:18 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-6 15:47:24 - error: Failed to reenable joining
2019-6-6 15:50:41 - info: Succesfully reenabled joining
2019-6-6 15:52:18 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-6 15:54:01 - info: Succesfully reenabled joining
2019-6-6 15:57:18 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-6 15:57:24 - error: Failed to reenable joining
2019-6-6 16:00:44 - error: Failed to reenable joining
```

Koenkk, 2019-06-07:
It's strange indeed, how do you flash the stick? (this should also provide an option to clear it).

Kevlordofcode, 2019-06-07:
I use the CC debugger with cc-tools on MacOS, with the exact command I found in you documentation:
`sudo ./cc-tool -e -w CC2531ZNP-Prod.hex`

Koenkk, 2019-06-07:
`-e` should erase the chip. To be sure, you could try on windows with flash programmer, if that doesn't help I think there is something wrong with the CC2531 stick.

Kevlordofcode, 2019-06-07:
I'll try on Windows next week (if I find a suitable PC), otherwise I guess I will just order a new CC2531. Thanks anyway for the help!

closed 2019-06-07 by Kevlordofcode

vrolland78, 2019-07-07:
Same issue "error: Failed to reenable joining" in the log and no more update.
Only 1 aqara temp sensor for test actually.
Only solution is to restart the service :(

Kevlordofcode, 2019-07-08:
On my side I figured it out eventually. This was due to Homeassistant trying to access a zwave coordinator stick that was no longer there. I have disabled openzwave in Homeassistant and now it's working perfectly. I would suggest you check if something can somehow interfere with your usb port and disable it, it might help!

rippi46, 2019-07-24:
Hello,
since the update from zigbee2mqtt to version 1.5.1 I have the same problem with my CC2530 via ESP8266. After a restart of the service zigbee2mqtt everything works fine, but after some time I get the following message:

`Jul 23 23:29:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:29:29 PM Failed to reenable joining
Jul 23 23:32:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:32:09 PM Failed to reenable joining
Jul 23 23:34:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:34:49 PM Failed to reenable joining
Jul 23 23:37:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:37:29 PM Failed to reenable joining
Jul 23 23:40:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:40:09 PM Failed to reenable joining
Jul 23 23:42:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:42:49 PM Failed to reenable joining
Jul 23 23:45:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:45:29 PM Failed to reenable joining
Jul 23 23:48:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:48:09 PM Failed to reenable joining
Jul 23 23:50:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:50:49 PM Failed to reenable joining
Jul 23 23:53:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:53:29 PM Failed to reenable joining
Jul 23 23:56:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:56:09 PM Failed to reenable joining
Jul 23 23:58:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-23 11:58:49 PM Failed to reenable joining
Jul 24 00:01:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:01:29 AM Failed to reenable joining
Jul 24 00:04:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:04:09 AM Failed to reenable joining
Jul 24 00:06:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:06:49 AM Failed to reenable joining
Jul 24 00:09:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:09:29 AM Failed to reenable joining
Jul 24 00:12:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:12:09 AM Failed to reenable joining
Jul 24 00:14:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:14:49 AM Failed to reenable joining
Jul 24 00:17:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:17:29 AM Failed to reenable joining
Jul 24 00:20:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:20:09 AM Failed to reenable joining
Jul 24 00:22:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:22:49 AM Failed to reenable joining
Jul 24 00:25:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:25:29 AM Failed to reenable joining
Jul 24 00:28:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:28:09 AM Failed to reenable joining
Jul 24 00:30:49 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:30:49 AM Failed to reenable joining
Jul 24 00:33:29 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:33:29 AM Failed to reenable joining
Jul 24 00:36:09 fhem-pi npm[31386]:   zigbee2mqtt:error 2019-7-24 12:36:09 AM Failed to reenable joining
`
With a second stick, which is connected via USB I do not have this problem, although the same firmware is on both sticks.

This is how the output looks after starting the service.

`Jul 24 09:05:11 fhem-pi systemd[1]: Started zigbee2mqtt2.
Jul 24 09:05:13 fhem-pi npm[30892]: > zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt2
Jul 24 09:05:13 fhem-pi npm[30892]: > node index.js
Jul 24 09:05:16 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:16 AM Logging to directory: '/opt/zigbee2mqtt2/data/log/2019-07-24.09-05-15'
Jul 24 09:05:17 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:17 AM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
Jul 24 09:05:17 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:17 AM Starting zigbee-shepherd
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM zigbee-shepherd started
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM Coordinator firmware version: '20190223'
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM Currently 22 devices are joined:
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM 0x00158d0002f6a6ab (0x00158d0002f6a6ab): MLI-404011 - Müller Licht Tint remote control (EndDevice)
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM 0x90fd9ffffedd7c9e (0x90fd9ffffedd7c9e): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Router)
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM 0x00158d0001ad45dd (0x00158d0001ad45dd): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM 0x00158d0001b980b8 (0x00158d0001b980b8): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM 0x000d6ffffe1d3ace (0x000d6ffffe1d3ace): E1524 - IKEA TRADFRI remote control (EndDevice)
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM 0x00158d00031d719e (0x00158d00031d719e): 404006/404008/404004 - Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, opal white (Router)
Jul 24 09:05:21 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:05:21 AM 0xd0cf5efffe1aea7a (0xd0cf5efffe1aea7a): E1524 - IKEA TRADFRI remote control (EndDevice)
[... cut ...]
Jul 24 09:34:42 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:34:42 AM Successfully reenabled joining
Jul 24 09:35:53 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:35:53 AM MQTT publish: topic 'zigbee2mqtt2/0x00158d0002b3188c', payload '{"linkquality":81,"angle":5,"angle_x":3,"angle_y":-84,"angle_z":5,"angle_x_absolute":87,"angle_y_absolute":174,"battery":100,"voltage":3025}'
Jul 24 09:37:22 fhem-pi npm[30892]:   zigbee2mqtt:info 2019-7-24 9:37:22 AM Successfully reenabled joining
`

Any idea?

Greeting rippi46

petjek, 2019-08-20:
Same here. Everything worked fine for a while. I already added 6 devices (Xiaomi and Ikea) While adding the sixth device the "Failed to reenable joining"-message came up. Service crashed and I tried to resart. It took a while after "sudo systemctl stop zigbee2mqtt.service" until I got back to the prompt and from there on it is not even possible to start the service again.
When starting manual I see this:
`
arne@FHEM:/opt/zigbee2mqtt$  npm start 
> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js
  zigbee2mqtt:info 8/20/2019, 8:27:29 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-08-20.20-27-28'
  zigbee2mqtt:info 8/20/2019, 8:27:29 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 8/20/2019, 8:27:29 PM Starting zigbee-shepherd
  zigbee2mqtt:info 8/20/2019, 8:27:41 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 8/20/2019, 8:28:41 PM Starting zigbee-shepherd
  zigbee2mqtt:error 8/20/2019, 8:29:20 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 8/20/2019, 8:29:20 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 8/20/2019, 8:29:20 PM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 8/20/2019, 8:29:20 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/arne/.npm/_logs/2019-08-20T18_29_57_094Z-debug.log
arne@FHEM:/opt/zigbee2mqtt$
 `

With sudo systemctl start zigbee2mqtt.service and sudo journalctl -u zigbee2mqtt.service -f I get the following:

`
arne@FHEM:/opt/zigbee2mqtt$ sudo systemctl start zigbee2mqtt.service
arne@FHEM:/opt/zigbee2mqtt$ sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at Mon 2019-06-10 08:51:26 CEST. --
Aug 20 20:33:07 FHEM systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 20.
Aug 20 20:33:07 FHEM systemd[1]: Stopped zigbee2mqtt.
Aug 20 20:33:07 FHEM systemd[1]: zigbee2mqtt.service: Failed to reset devices.list: Operation not permitted
Aug 20 20:33:07 FHEM systemd[1]: Started zigbee2mqtt.
Aug 20 20:33:08 FHEM npm[3702]: > zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
Aug 20 20:33:08 FHEM npm[3702]: > node index.js
Aug 20 20:33:08 FHEM npm[3702]:   zigbee2mqtt:info 8/20/2019, 8:33:08 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-08-20.20-33-08'
Aug 20 20:33:08 FHEM npm[3702]:   zigbee2mqtt:info 8/20/2019, 8:33:08 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
Aug 20 20:33:08 FHEM npm[3702]:   zigbee2mqtt:info 8/20/2019, 8:33:08 PM Starting zigbee-shepherd
Aug 20 20:33:20 FHEM npm[3702]:   zigbee2mqtt:info 8/20/2019, 8:33:20 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
`

Had this problem two times now and the only way to solve it the first time was to completely remove the installation and start from scratch.

Koenkk, 2019-08-22:
It looks like the stick crashed, do you have the logs when the first Failed to reeanble joining starts to appear?