Koenkk/zigbee2mqtt #1602: 2430-100 (Gira ZigBee Light Link wall transmitter) loosing connection

int2001, 2019-06-06:
## Gira ZigBee Light Link wall transmitter looses connection to coordinator

The Gira LightLink (3Gang) ist loosing it's connection to the coordinator every morning. 

To reproduce: After roundabout 10h of no action the device is not longer connected to zigbee2mqtt. No debug-output or anything else will be produced. Simply disconnected. 

One have to repair the device manually. (Full-Reset and rePairing of device is required).

## Debug Info
zigbee2mqtt version: Latest Dev: 8ad6aee43a79a6fe59e687836928c5d942022d18
CC253X firmware version: Latest Stable

Koenkk, 2019-06-07:
Can you check using zigbee sniffing that somehow the device is being kicked out the network? (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html). You need to capture the moment when it leaves.

closed 2019-08-14 by stale[bot]

rendgeor, 2019-08-23:
see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1571
Gira light link has problem when other zigbee 3.0 device is in same network!

felixstorm, 2019-11-12:
Just for reference: According to both to [the EDITs here](https://translate.google.com/translate?sl=de&tl=en&u=https%3A%2F%2Fforum.fhem.de%2Findex.php%2Ftopic%2C100772.msg976520.html%23msg976520) and also to my own experience the Jung wall transmitters seem to have the same problem (both Gira and Jung have some of their electronics developed by https://www.insta.de, so both devices are very similar internally).

But at least according to the Jung technical hotline Insta does indeed plan to provide a firmware update to solve this issue (it seems to be well-known also to the Jung), but as of 2019-11-12 there was no estimate yet on when this might be delivered.

felixstorm, 2020-03-13:
Update for Jung devices available (see https://github.com/Koenkk/zigbee2mqtt/issues/2417#issuecomment-598650782), but apparently Gira does really not intend to release an update for their devices.