Koenkk/zigbee2mqtt #1603: Sniffer firmware for CC2531 flash frozen (Alternative flashing methods)

antiogenesneto, 2019-06-06:
Hi,

I tried to flash my sniffer zigbee cc2531 using a raspberry pi 3b and the procedure was frozen in step "writing page 1/128".

[image]
[image]

rickkdotnet, 2019-06-06:
7fff looks like it can't read the flash. I'd doublecheck the wiring.

closed 2019-08-12 by stale[bot]

simfun, 2019-10-07:
It must be a power issue. I had this error too and after i connected a 2A power supply on RPI Zero it flashed the stick steadily.