Koenkk/zigbee2mqtt #1604: Mesh suddenly down

brettyj, 2019-06-06:
# Bug Report
## What happened

Tried controlling devices and then they weren't responding. Had a look at the zigbee2mqtt logs and see this:
````
zigbee2mqtt:error 6/6/2019, 12:35:15 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - on - {}                                                      - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 23                                                     3. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:35:45 PM Zigbee publish to device '0x0017880103e310b4', lightingColorCtrl                                                      - moveToColorTemp - {"colortemp":250,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with                                                      error Error: Timed out after 30000 ms
  zigbee2mqtt:error 6/6/2019, 12:35:52 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - on - {}                                                      - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 23                                                     3. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:35:59 PM Zigbee publish to device '0x0017880103e310b4', lightingColorCtrl                                                      - moveToColorTemp - {"colortemp":250,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with                                                      error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:36:06 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - on - {}                                                      - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 23                                                     3. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:36:13 PM Zigbee publish to device '0x0017880103e310b4', lightingColorCtrl                                                      - moveToColor - {"transtime":0,"colorx":8913,"colory":2621} - {"manufSpec":0,"disDefaultRsp":0} - null fail                                                     ed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:36:20 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - on - {}                                                      - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 23                                                     3. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:36:35 PM Zigbee publish to device '0x0017880100deacca', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     05. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 6/6/2019, 12:36:38 PM Zigbee publish to device '0x0017880103e310b4', lightingColorCtrl                                                      - moveToColor - {"transtime":0,"colorx":8913,"colory":2621} - {"manufSpec":0,"disDefaultRsp":0} - null fail                                                     ed with error Error: AF data request fails, status code: 183. APS no ack.
  zigbee2mqtt:error 6/6/2019, 12:36:42 PM Zigbee publish to device '0x0017880100df4001', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     05. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 6/6/2019, 12:36:58 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - on - {}                                                      - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 18                                                     3. APS no ack.
  zigbee2mqtt:error 6/6/2019, 12:36:58 PM Zigbee publish to device '0x0017880103e310b4', lightingColorCtrl                                                      - moveToColor - {"transtime":0,"colorx":8913,"colory":2621} - {"manufSpec":0,"disDefaultRsp":0} - null fail                                                     ed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:37:05 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - on - {}                                                      - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 23                                                     3. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:37:35 PM Zigbee publish to device '0x0017880103e310b4', lightingColorCtrl                                                      - moveToColor - {"transtime":0,"colorx":45940,"colory":19595} - {"manufSpec":0,"disDefaultRsp":0} - null fa                                                     iled with error Error: Timed out after 30000 ms
  zigbee2mqtt:error 6/6/2019, 12:37:55 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 1                                                     83. APS no ack.
  zigbee2mqtt:error 6/6/2019, 12:38:08 PM Zigbee publish to device '0x0017880103e310b4', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     33. MAC no ack.
  zigbee2mqtt:error 6/6/2019, 12:38:41 PM Zigbee publish to device '0x00178801023122e1', genLevelCtrl - rea                                                     d - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 3000                                                     0 ms
  zigbee2mqtt:error 6/6/2019, 12:39:48 PM Zigbee publish to device '0x0017880100df4001', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     05. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 6/6/2019, 12:39:54 PM Zigbee publish to device '0x0017880100deacca', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     05. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 6/6/2019, 12:40:01 PM Zigbee publish to device '0x0017880100df4001', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     05. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 6/6/2019, 12:42:01 PM Zigbee publish to device '0x0017880100deacca', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     05. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 6/6/2019, 12:42:05 PM Zigbee publish to device '0x0017880100df4001', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 2                                                     05. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 6/6/2019, 12:42:11 PM Zigbee publish to device '0x0017880100df4001', genOnOff - off - {                                                     } - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fai [... cut ...]  Timed out after 3000                                                     0 ms
````

The mesh looks like 
[image]

Normally it looks pretty healthy. I'm a bit nervous as I have door sensors and the like on order for a basic security system and so I need the mesh to be reliable.

The mesh is running on zigbee channel 25, with my wifi and sonos using 1 and 6.

## What did you expect to happen
The lights to work

## How to reproduce it (minimal and precise)

## Debug Info
On latest versions of both

brettyj, 2019-06-06:
Rebooted the Pi and can control some lights but not others and my mesh shows devices completely unconnected now:

[image]

Now what do I do?

brettyj, 2019-06-06:
After about an hour or so from unplugging and replugging the pi, the mesh suddenly reformed and everything is working again - any ideas on what could have caused this and how to prevent this from happening again? I have hue lights all through the house, but could add some Ikea Sockets if that will make the mesh stronger?

Koenkk, 2019-06-07:
What is the exact firmware version you are using?

brettyj, 2019-06-07:
Hey,

Firmware 20190425 CC2531

Koenkk, 2019-06-07:
I see, there may be some stability issue with it, see https://github.com/Koenkk/Z-Stack-firmware/issues/86. Please try my suggestion from https://github.com/Koenkk/Z-Stack-firmware/issues/86#issuecomment-500056683. If you continue to have issue, please report there (in that we we keep things concentrated).

closed 2019-06-07 by Koenkk

brettyj, 2019-06-07:
Hey,

Will do as soon as my new debugger comes through, I'll update this ticket then.

Thanks