Koenkk/zigbee2mqtt #1606: Support Hive DWS003 contact/light/temp - enrollReq

mattaw, 2019-06-07:
Creating entries for Hive DWS003 contact/temperature/light sensor. I was able to get the temperature, light and power reading just fine but I was never able to enroll with IAS successfully and the device would go to sleep.

## Host Info
zigbee2mqtt version: hass.io 1.4.0 and edge
CC253X firmware version: Coordinator firmware version: '20190425'

## To reproduce
1. Stop zigbee2mqtt
2. Delete line from database.db referring to sensor
3. Hold down reset button until red light comes on and turns off
4. Start zigbee2mqtt
5. Put in battery
6. Observe logs

Thoughts: zigbee2mqtt isn't replying to the enrollReq (8.2.2.4.2 Zigbee Zone Enroll Request Command), so the ias bit of the device goes away, but the other bits keep on working as they don't require zoneState = 1 to work.

## Relevant log messages
```
zigbee2mqtt:warn 6/7/2019, 9:58:09 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 6/7/2019, 9:58:09 AM Received zigbee message of type 'devInterview' with data '"0x00124b0015fb8774"'
  zigbee2mqtt:debug 6/7/2019, 9:58:09 AM Accept device incoming with ieeeAddr '0x00124b0015fb8774' permit join is 'true'
  zigbee2mqtt:debug 6/7/2019, 9:58:09 AM Allowing device '0x00124b0015fb8774' to join
2019-06-07T13:58:09.525Z zigbee-shepherd Device: 0x00124b0015fb8774 join the network.
  zigbee2mqtt:debug 6/7/2019, 9:58:09 AM Received zigbee message of type 'devIncoming' with data '"0x00124b0015fb8774"' of device 'DWS003' (0x00124b0015fb8774) of endpoint 6
  zigbee2mqtt:info 6/7/2019, 9:58:09 AM Device incoming...
  zigbee2mqtt:info 6/7/2019, 9:58:09 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 6/7/2019, 9:58:09 AM Device with modelID 'DWS003' is not supported.
  zigbee2mqtt:warn 6/7/2019, 9:58:09 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 6/7/2019, 9:58:09 AM Device with modelID 'DWS003' is not supported.
  zigbee2mqtt:warn 6/7/2019, 9:58:09 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 6/7/2019, 9:58:09 AM Received zigbee message of type 'devStatus' with data '"online"' of device 'DWS003' (0x00124b0015fb8774) of endpoint 6
2019-06-07T13:58:24.697Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":11225900,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[25,2,1,21,0,57,16]}}
2019-06-07T13:58:24.706Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-06-07T13:58:24.707Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":11225900,"transseqnumber":0,"len":7,"data":{"0":25,"1":2,"2":1,"3":21,"4":0,"5":57,"6":16},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":2,"cmdId":"enrollReq","payload":{"zonetype":21,"manucode":4153}}}
2019-06-07T13:58:24.708Z zigbee-shepherd:af Message has unsupported cmdID: enrollReq
2019-06-07T13:58:34.465Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":118,"securityuse":0,"timestamp":11256436,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[25,3,1,21,0,57,16]}}
2019-06-07T13:58:34.470Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-06-07T13:58:34.471Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":118,"securityuse":0,"timestamp":11256436,"transseqnumber":0,"len":7,"data":{"0":25,"1":3,"2":1,"3":21,"4":0,"5":57,"6":16},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":3,"cmdId":"enrollReq","payload":{"zonetype":21,"manucode":4153}}}
2019-06-07T13:58:34.471Z zigbee-shepherd:af Message has unsupported cmdID: enrollReq
2019-06-07T13:58:44.608Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11288146,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[25,4,1,21,0,57,16]}}
2019-06-07T13:58:44.612Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-06-07T13:58:44.613Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11288146,"transseqnumber":0,"len":7,"data":{"0":25,"1":4,"2":1,"3":21,"4":0,"5":57,"6":16},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":4,"cmdId":"enrollReq","payload":{"zonetype":21,"manucode":4153}}}
2019-06-07T13:58:44.614Z zigbee-shepherd:af Message has unsupported cmdID: enrollReq
2019-06-07T13:58:54.393Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11318722,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[25,5,1,21,0,57,16]}}
2019-06-07T13:58:54.396Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-06-07T13:58:54.398Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11318722,"transseqnumber":0,"len":7,"data":{"0":25,"1":5,"2":1,"3":21,"4":0,"5":57,"6":16},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":5,"cmdId":"enrollReq","payload":{"zonetype":21,"manucode":4153}}}
2019-06-07T13:58:54.398Z zigbee-shepherd:af Message has unsupported cmdID: enrollReq
2019-06-07T13:59:04.192Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11349315,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[25,6,1,21,0,57,16]}}
2019-06-07T13:59:04.196Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-06-07T13:59:04.197Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11349315,"transseqnumber":0,"len":7,"data":{"0":25,"1":6,"2":1,"3":21,"4":0,"5":57,"6":16},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":6,"cmdId":"enrollReq","payload":{"zonetype":21,"manucode":4153}}}
2019-06-07T13:59:04.197Z zigbee-shepherd:af Message has unsupported cmdID: enrollReq
2019-06-07T13:59:13.961Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11379850,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[25,7,1,21,0,57,16]}}
2019-06-07T13:59:13.965Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-06-07T13:59:13.966Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11379850,"transseqnumber":0,"len":7,"data":{"0":25,"1":7,"2":1,"3":21,"4":0,"5":57,"6":16},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":7,"cmdId":"enrollReq","payload":{"zonetype":21,"manucode":4153}}}
2019-06-07T13:59:13.967Z zigbee-shepherd:af Message has unsupported cmdID: enrollReq
2019-06-07T13:59:24.715Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":63143,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":11413473,"transseqnumber":0,"len":7,"data":{"t [... cut ...]
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 255,}, cb),
            ];
            execute(device, actions, callback);
        },
    },
```

[example log of configure fail but can still read temp illum batt.txt](https://github.com/Koenkk/zigbee2mqtt/files/3266243/example.log.of.configure.fail.but.can.still.read.temp.illum.batt.txt)

Koenkk, 2019-06-08:
What's your reason for trying to iAs enroll the device? In you post you mention that everything is already working (contact/temperature/light)

mattaw, 2019-06-08:
Sadly the contact doesn't work - it works occasionally, quickly while pairing, but then stops responding ( the device goes to sleep? ) and you just get temperature / light / battery reported on schedule.

I guessed it was the lack of ias enrollment (zoneState never goes 1)

Koenkk, 2019-06-08:
Can you share your `database.db`?

mattaw, 2019-06-08:
```
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0015fb8774","nwkAddr":63143,"manufId":4153,"manufName":"HiveHome.com","powerSource":"Battery","modelId":"DWS003","epList":[6],"zclVersion":1,"appVersion":1,"stackVersion":38,"hwVersion":0,"status":"offline","joinTime":null,"endpoints":{"6":{"profId":260,"epId":6,"devId":1026,"inClusterList":[0,1,3,32,1024,1026,1280],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":38,"hwVersion":0,"manufacturerName":"HiveHome.com","modelId":"DWS003","dateCode":"20161111","powerSource":3,"appProfileVersion":16908544,"locationDesc":"","physicalEnv":0,"deviceEnabled":1}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryVoltage":31,"batteryPercentageRemaining":15,"batteryManufacturer":"CR123A","batterySize":2,"batteryAHrRating":155,"batteryQuantity":1,"batteryRatedVoltage":30,"batteryVoltMinThres":25}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0,"identifyCommissionState":3}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"20161111","_id":"TClDyX8xE04clDNb"}
```

I am so sorry for not including it.

mattaw, 2019-06-08:
I have tried various combinations of read of ssIasZone (got the zoneid set, but zoneState never went to 1), and reporting ssIasZone never worked, just said reportable device.

Koenkk, 2019-06-10:
After adding the `configure` section in `devices.js`, did you re-pair the device? Also, when doing this do you see a `Succesfully configured ....` for this device (search for ieeeAddr)?

mattaw, 2019-06-10:
To answer your question, yes, stopped zigbee2mqtt, deleted db lines, reset sensor, restarted zigbee2mqtt, repair-ed. 

However I decided to try pairing them with a hive hub to see if they got a firmware update (there is muttering on their forums about it). They don't pair to that well at all either! So I am going to work with Hive support first, and then come back to zigbee2mqtt. I don't want to use your time until I am happy they work with the original hub first.

closed 2019-08-16 by stale[bot]

ovid-io, 2021-11-02:
Hey @mattaw, did you ever resolve this issue? I have two of these with same/similar issue. Funnily enough I have another 3 of MOT003 sensors that do it too - they never completely pair, but function fine. Though whenever I enable join all 5 of hive sensors are always starting interview and always failing...

mattaw, 2021-11-02:
Sadly no, life got too busy and I simply bought some aquara units, and been
very happy with them.

sourbrambles, 2022-08-19:
@ovid-io did you ever find a resolution?

ovid-io, 2022-08-19:
@sourbrambles I haven't but sensors appear to work mostly fine otherwise