Koenkk/zigbee2mqtt #1607: Network suddenly down and zigbee2mqtt fails to start with Coordinator initialize timeout

odx, 2019-06-07:
After running a network with 36 devices (including multiple router devices) quite stable for a while. I added 6 Heiman HS3SA smoke detectors. About one day later the network suddenly went down. I am no longer able to startup zigbee2mqtt. I updated to 1.4.0 but it does not help. 
Here is the log: https://hastebin.com/tabiboyaru.pl

After reboot of the pi the coordinator USB dongle has the green LED on which turns off during z2m initialization. When pushing the reset button on the stick it does not go back on. I have to replug it to get the light back on. The coordinator is currently running CC2531ZNP-Prod_20190223.

Could it be that the firmware is corrupted?

I didn't dare to reset the z2m data yet. What files should be preserved in order to keep the devices paired? As far as I know it should be safe to update the sticks firmware at that version without having to repair all the devices, is that correct?

Koenkk, 2019-06-07:
That firmware could indeed cause stability issue, please flash the latest firmware and try again (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin). Does not require re-pairing.

odx, 2019-06-16:
Thanks April's firmware works again.

closed 2019-06-16 by odx