Koenkk/zigbee2mqtt #1608: Single (fully supported) device not adding to HomeAssistant

CamHaug, 2019-06-08:
I am not sure if this is actually a zigbee2mqtt issue, but here goes.

I have added a few more Nue switches to my setup, specifically these: 'FNB56-ZSW02LX2.0' the first 2 i added were fine and add to HA no worries. the 3rd adds to the zigbee network, shows up in the map but does not add to HA. I have since added other new devices to the network with no issues getting them to showup in HA.

The device is also reporting state change etc via zigbee2mqtt with no issues.

I have tried removing from the DB manually and rejoining with no luck.

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: '20190315'

Koenkk, 2019-06-08:
Isn't the device also on the 'States' page in home assistant? You should also try to search for the `ieeeAddr` of the device there (logged on zigbee2mqtt startup)

CamHaug, 2019-06-09:
No has not been picked up by HA anywhere - can only find the address in the zigbee2mqtt logs and on the map.

sooo weird

Koenkk, 2019-06-10:
Is this discovery payload also logged on startup? If so, this seems to be a  home assistant issue.

CamHaug, 2019-06-10:
Comes up with this:
6/10/2019, 5:57:10 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002b7a667', payload '{"state_top":"ON","linkquality":13,"state_bottom":"ON"}'
D
Detects all state changes etc. as i said, super weird as all other devices are picked up by HA with no issues (including other switches of the same version).

CamHaug, 2019-06-24:
@Koenkk 
I think I am getting hung up on the fact that i have 2 of these switches and only one doesn't work. No Idea why that is.
However. I just readded and it occurred to me that its not supported... as i got messages about missing converters.

I have added this to devices.js and the switch states all detect etc:
    {
        zigbeeModel: ['FNB56-ZSW02LX2.0'],
        model: 'HGZB-42',
        vendor: 'Nue / 3A',
        description: 'Smart light switch - 2 gang v2.0',
        supports: 'on/off',
        fromZigbee: [fz.generic_state_multi_ep, fz.ignore_onoff_change],
        toZigbee: [tz.on_off],
        ep: (device) => {
            return {'top': 11, 'bottom': 12};
        },
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep11 = shepherd.find(ieeeAddr, 11);
            execute(ep11, [(cb) => ep11.bind('genOnOff', coordinator, cb)], () => {
                const ep12 = shepherd.find(ieeeAddr, 12);
                execute(ep12, [(cb) => ep12.bind('genOnOff', coordinator, cb)], () => {
                });
            });
        },
    },

May just need you to write support for HA?
Converter Warnings i got were:

/24/2019, 8:33:26 PM - warn: No converter available for 'HGZB-42' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"FeiBit","modelId":"FNB56-ZSW02LX2.0","powerSource":1}}'
6/24/2019, 8:33:26 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
6/24/2019, 8:33:26 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"201808.08","swBuildId":"1000-0001"}}' of device 'FNB56-ZSW02LX2.0' (0x00158d0002b7a667) of endpoint 11
6/24/2019, 8:33:26 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"201808.08","swBuildId":"1000-0001"}}' of device 'FNB56-ZSW02LX2.0' (0x00158d0002b7a667) of endpoint 11
6/24/2019, 8:33:26 PM - warn: No converter available for 'HGZB-42' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"201808.08","swBuildId":"1000-0001"}}'
6/24/2019, 8:33:26 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Cheers -Cam

Koenkk, 2019-06-24:
Converter warning don't matter, these messages are just not parsed by zigbee2mqtt.

To verify if it is a zigbee2mqtt/homeassistant issue, verify that a home assistant discovery message is published on zigbee2mqtt start (search by the device ieeeAddr).

CamHaug, 2019-06-24:
Im not sure I follow? I have added another new device after this one and it discovered just fine.....here is the startup log from though. you can see the device ending in 667

6/24/2019, 8:22:19 PM - info: Logging to directory: '/app/data/log/2019-06-24.20-22-19'
6/24/2019, 8:22:19 PM - debug: Removing old log directory '/app/data/log/2019-06-02.10-57-56'
6/24/2019, 8:22:20 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[26],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
6/24/2019, 8:22:20 PM - debug: Loaded state from file /app/data/state.json
6/24/2019, 8:22:20 PM - debug: Saving state to file /app/data/state.json
6/24/2019, 8:22:20 PM - info: Starting zigbee2mqtt version 1.4.0 (commit #unknown)
6/24/2019, 8:22:20 PM - info: Starting zigbee-shepherd
6/24/2019, 8:22:23 PM - info: zigbee-shepherd started
6/24/2019, 8:22:24 PM - info: Coordinator firmware version: '20190315'
6/24/2019, 8:22:24 PM - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"26","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0018e1aaaf","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190315},"startTime":1561378943,"joinTimeLeft":0}
6/24/2019, 8:22:24 PM - info: Currently 26 devices are joined:
6/24/2019, 8:22:24 PM - info: 0x00124b0014ce4833 (0x00124b0014ce4833): FB56-THM12HM1.4 - undefined Smart Temperature and Humidity Detector (EndDevice)
6/24/2019, 8:22:24 PM - info: 0x00124b0019e12bd1 (0x00124b0019e12bd1): HGZB-042 - Nue / 3A Smart light switch - 2 gang (Router)
6/24/2019, 8:22:24 PM - info: unknown (0x00158d0001dcf845): FNB56-ZIR04FB1.2 - Nue / 3A Zigbee IR Blaster (EndDevice)
6/24/2019, 8:22:24 PM - info: 0x00124b0019e139f3 (0x00124b0019e139f3): HGZB-042 - Nue / 3A Smart light switch - 2 gang (Router)
6/24/2019, 8:22:24 PM - info: 0x00124b0019e12b98 (0x00124b0019e12b98): HGZB-042 - Nue / 3A Smart light switch - 2 gang (Router)
6/24/2019, 8:22:24 PM - info: 0x00158d000216604b (0x00158d000216604b): MG-AUWS01 - Nue / 3A Smart Double GPO (Router)
6/24/2019, 8:22:24 PM - info: 0x00124b0018efea37 (0x00124b0018efea37): HGZB-043 - Nue / 3A Smart light switch - 3 gang (Router)
6/24/2019, 8:22:24 PM - info: 0xd0cf5efffe4145b3 (0xd0cf5efffe4145b3): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0x000d6ffffe40f3f9 (0x000d6ffffe40f3f9): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0xd0cf5efffef12bdc (0xd0cf5efffef12bdc): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0xd0cf5efffef12cf9 (0xd0cf5efffef12cf9): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0xd0cf5efffe409f25 (0xd0cf5efffe409f25): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0x90fd9ffffe535eca (0x90fd9ffffe535eca): E1524 - IKEA TRADFRI remote control (EndDevice)
6/24/2019, 8:22:24 PM - info: 0x000d6ffffe179d61 (0x000d6ffffe179d61): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0xd0cf5efffef2988d (0xd0cf5efffef2988d): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0xd0cf5efffef13340 (0xd0cf5efffef13340): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0x00158d0002b95145 (0x00158d0002b95145): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
6/24/2019, 8:22:24 PM - info: 0x90fd9ffffe50681f (0x90fd9ffffe50681f): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
6/24/2019, 8:22:24 PM - info: 0xd0cf5efffe316ef2 (0xd0cf5efffe316ef2): E1524 - IKEA TRADFRI remote control (EndDevice)
6/24/2019, 8:22:24 PM - info: 0x00158d0002a8671f (0x00158d0002a8671f): HGZB-42-UK / HGZB-41 - Nue / 3A Smart switch 1 or 2 gang (Router)
6/24/2019, 8:22:24 PM - info: 0x00158d0002a86b26 (0x00158d0002a86b26): HGZB-42 - Nue / 3A Smart light switch - 2 gang v2.0 (Router)
6/24/2019, 8:22:24 PM - info: 0x00158d0002a86235 (0x00158d0002a86235): HGZB-42-UK / HGZB-41 - Nue / 3A Smart switch 1 or 2 gang (Router)
6/24/2019, 8:22:24 PM - info: 0x00158d000292b334 (0x00158d000292b334): HGZB-42-UK / HGZB-41 - Nue / 3A Smart switch 1 or 2 gang (Router)
6/24/2019, 8:22:24 PM - info: 0x00158d0002a85025 (0x00158d0002a85025): HGZB-42-UK / HGZB-41 - Nue / 3A Smart switch 1 or 2 gang (Router)
6/24/2019, 8:22:24 PM - info: 0x00158d0002b7a667 (0x00158d0002b7a667): HGZB-42 - Nue / 3A Smart light switch - 2 gang v2.0 (Router)
6/24/2019, 8:22:24 PM - info: 0x00124b001b92284a (0x00124b001b92284a): GL-S-007Z - Gledopto Smart RGBW GU10 (Router)
6/24/2019, 8:22:24 PM - warn: `permit_join` set to  `true` in configuration.yaml.
6/24/2019, 8:22:24 PM - warn: Allowing new devices to join.
6/24/2019, 8:22:24 PM - warn: Set `permit_join` to `false` once you joined all devices.
6/24/2019, 8:22:24 PM - info: Zigbee: allowing new devices to join.
[... cut ...]
6/24/2019, 8:22:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000292b334', payload '{"state":"OFF","linkquality":31}'
6/24/2019, 8:22:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a85025', payload '{"state":"ON","linkquality":36}'
6/24/2019, 8:22:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002b7a667', payload '{"state_top":"ON","linkquality":31}'
6/24/2019, 8:22:25 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001b92284a', payload '{"state":"ON","color":{"x":0.612,"y":0.374}}'

Koenkk, 2019-06-25:
Can you post more of the startup log?

CamHaug, 2019-06-25:
.... that is all of it...?

Koenkk, 2019-06-25:
Do you have `homeassistant: true` in your `configuration.yaml`?

CamHaug, 2019-06-25:
Yes I do...

Koenkk, 2019-06-26:
If some devices have been discovered successfully, I would at least expect publish to the `homeassistant/` MQTT topic in the logs.

Please provide more of the starting log, up till where you control devices.

closed 2019-09-01 by stale[bot]

saury, 2020-04-30:
@CamHaug 
Hi, I wonder if you make this 'FNB56-ZIR04FB1.2' (IR device) connect to HA?
Really appreciate it if you can help me with this IR device.

chrisjbawden, 2020-06-01:
> @CamHaug
> Hi, I wonder if you make this 'FNB56-ZIR04FB1.2' (IR device) connect to HA?
> Really appreciate it if you can help me with this IR device.

Not sure if you have this up and running. I just got one of these and looking to get it working. 

Here - https://github.com/Koenkk/zigbee2mqtt/issues/3649