Koenkk/zigbee2mqtt #1610: signal repeater doens't repeat

eeevs, 2019-06-08:
Hi,

I got one device with a linkquality of 0. So i bought a ikea signal repeater. The ikea repeater is connected to zigbee2mqtt but the device with the bad linkquality I have still problems with.
I updated zigbee2mqtt and I repaired the device. But still no luck.
Any suggestions ?

Thanks,
EVS

Koenkk, 2019-06-09:
What device is it?

eeevs, 2019-06-10:
The device is a xiaomi aqara temperature/humidity sensor.

I did some more debuging. The ikea repeater does nothing for my existing network. I will try adding new devices and see what happens.

eeevs, 2019-06-10:
Is this the problem?
Jun 10 10:07:26 hassbian npm[12979]:   zigbee2mqtt:info 6/10/2019, 10:07:26 AM repeater1 (0x000d6ffffeaa45c5): E1746 - IKEA TRADFRI signal repeater (Router)

AM No converter available for 'E1746' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"10":[69,49,55,52,54]}}'

Jun 10 10:07:42 hassbian npm[12979]:   zigbee2mqtt:warn 6/10/2019, 10:07:42 AM No converter available for 'E1746' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"lastMessageLqi":104,"lastMessageRssi":-74}}'

same issue? https://github.com/Koenkk/zigbee2mqtt/issues/1100

Koenkk, 2019-06-10:
No that's not an issue. 

- What do you see in your networkmap?
- Can you try re-pairing the Xiaomi device close the the repeater?

Note that sometimes xiaomi end devices are very picky about choosing their router.

eeevs, 2019-06-13:
I did some more testing:
I added the ikea router and set permit_join:  false again
That didn' t work work the ikea repeater.
So:
I did leave the network for a day or so with permit_join: true
 and then set permit_join: false again 
After that i move some sensors outside the range of the zigbee2mqtt and router  into the extended router router range.
And that seems to work.
 so somehow: permit_join: true is important for extend your zigbee network even if all the devices are already connected.

Koenkk, 2019-06-13:
Thanks for sharing this, I guess it depends on the devices used, if you issues are solved please close this issue.

closed 2019-06-15 by eeevs