Koenkk/zigbee2mqtt #1612: Pinging ZNCZ02LM - Xiaomi Mi power plug ZigBee  fails

lealoureiro, 2019-06-09:
Hello,

I'm running version of zigbee2mqtt 1.4.0 with CC2531 firmware 20190425 and i also have a router with firmware from file CC2531_router_2018_09.zip.

Keeps appear on logs that it can't ping the Xiaomi Power plug:

error: Failed to ping 0x00158d000xxxxxxx

And I believe because of that the state of plug on Home Assistant is not consistent with actual state of the plug. 

I have no extra configuration for this item in configuration.yaml

Koenkk, 2019-06-10:
Can you try if it also occurs with this firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing (`CC2531_SOURCE_ROUTING_20190610.zip`).

lealoureiro, 2019-06-10:
I flashed with suggested firmware still same problem.

I tried to repair and it fixed. Lets see if stays stable.

Thanks.

closed 2019-06-10 by lealoureiro