Koenkk/zigbee2mqtt #1617: Please add this to devices.js to support Hive Active Light Cool to Warm White TWBulb01US

mattaw, 2019-06-09:
I have got the  Hive Active Light Cool to Warm White TWBulb01US bulb working great with Zigbee2Mqtt.

Please add it to the official repo and homeassistant integration publishing.
Hive Active Light Cool to Warm White TWBulb01US 

```
    {
        zigbeeModel: ['TWBulb01US'],
        model: 'HV-GSCXZB269',
        vendor: 'Hive',
        description: 'Hive Active Light Cool to Warm White (E26) ',
        extend: generic.light_onoff_brightness_colortemp,
    },
```

Koenkk, 2019-06-09:
Thanks, supported in the dev branch now!

closed 2019-06-09 by Koenkk

mattaw, 2019-06-10:
Please could you add to the homeassistant.js as well? Or provide a neat way of managing those integrations in devices.js. I saw a homeassistant key, which I approve of (keep all device config in one file, but I couldn't quickly figure out how to use it to get something supported.

I think its a one liner in homeassistant.js, or could we move homeassistant.js to the zigbee-shepherd-converters project, since they both contain device specific information.

Koenkk, 2019-06-10:
This has already been done in the dev branch.

mattaw, 2019-06-10:
Could you share the commit? I am having trouble finding the code to do the publishing of the homeassistant/autodetect stuff. I am trying to get a dev environment running so I can do PRs directly for new devices and am very keen to include the homeassistant integration autodetect code (which I thought lived in zigbee2mqtt/lib/extension/homeassistant.js, and not zigbee-shepherd-converters) and if there is a newer way of doing this then I want to follow that. I couldn't find the code in your commit.

Thanks again, M

Koenkk, 2019-06-11:
@mattaw  see https://github.com/Koenkk/zigbee2mqtt/commit/852950fc09646bd8dc244e4f5e3bd845ad258518#diff-0c056233b68fd3653afb296f98947c8fR734

mattaw, 2019-06-13:
> @mattaw <https://github.com/mattaw> see 852950f
> #diff-0c056233b68fd3653afb296f98947c8fR734

My fault - I was looking for the Zigbee id, not model number
Thanks for the explaination