Koenkk/zigbee2mqtt #1618: Implement SBL Flashing Method as updatemanager inside Zigbee2Mqtt

tb-killa, 2019-06-10:
### This is a general issue post to discuss the introduction of a general SBL Flash method and to implement or test it.

As we already know, the firmware versions provided by @Koenkk contain the SBL (Serial Boot Loader), which allows communication via the serial interface during the boot process of the stick. This can also be used to write the firmware.

Normally (default) the USB stick waits 60 seconds after starting (plugging in or power supply of the processor on the stick) and then starts its own application (TI-ROS).
With the issue post: https://github.com/Koenkk/zigbee2mqtt/issues/1343 this behavior was statically skipped for the first time.

This option should be configurable to be able to trigger a restart during the update procedure where the bootloader waits for its time again.
1. Always skip default
2. Extend and implement MQTT Topic for Reset the Stick with the Value "waitonbootloader".

### And now the very good news :)

After a long search I found the flash tool sbl_tool.bin. 
A precompiled version for ARM is included in the **Z-Stack™Linux Ubuntu Gateway installer**, available at **http://www.ti.com/tool/CC2531EM-IOT-HOME-GATEWAY-RD**.
It runs fine on a Raspberry Pi. 
Just download the file from TI, and run it (under Linux). 
After unpacking, we find a tar archive at
```Z-Stack_Linux_Gateway-1.0.1-src/Precompiled_arm/z-stack_linux_gateway_arm_binaries_39793:39796M.tar```
After unpacking this archive, the Flash tool is available at
```Z-Stack_Linux_Gateway-1.0.1-src/Precompiled_arm/z-stack_linux_gateway_arm_binaries_39793:39796M/tools```
Here you can find the said "sbl_tool.bin"

With ``./sbl_tool.bin FILENAME SERIAL-PORT``  (./sbl_tool.bin CC2531ZNP-Prod.bin /dev/ttyACM0) you could flash them without any Problem.

How about providing a new MQTT-Topic **zigbee2mqtt/bridge/updatemanager** ?
You store the new firmware or firmware files in memory and can refer directly to a file via attributes "filename".

Return code or STDOUT of the Linux application should end up in the log and in the MQTT log topic, whereby one could consider whether an own ``log`` topic within the ``updatemanager`` might make sense.

For this in general I need @Koenkk help for implementation and installation.

Please lets Discuss and Bring in Ideas and Test if some Stuff is Implement.

zen2, 2019-08-15:
As I understand it, it means that we could flash firmware without CC debugger.
If it's the case that could be a huge improvement for new users since they don't need to buy a CC debugger.

That could permit to update firmware if needed when you update Zigbee2MQTT.
I think that could be a great enhancement since that should simplify firmware update for users.

tb-killa, 2019-10-14:
Keep open

closed 2019-12-20 by stale[bot]

spinpwr, 2021-04-28:
Link to download SerialBootTool from TI for Windows:
https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/538/SerialBootTool_5F00_1_5F00_3_5F00_2.zip