Koenkk/zigbee2mqtt #1619: Device pages, we need your help!

Koenkk, 2019-06-10:
From now on, each device will have a separate page on the zigbee2mqtt documentation. This page contains knowledge about the device (e.g. how to pair it). 

Examples:
- http://www.zigbee2mqtt.io/devices/E1603
- http://www.zigbee2mqtt.io/devices/E1524

The device page can be reached from the [Supported devices page](http://www.zigbee2mqtt.io/information/supported_devices.html) (click on the model)

These pages are automatically generated, information to the *Note* section can be added through: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docgen/device_page_notes.js

I would like to ask everybody to check the pages of their devices and contribute if applicable.

aktentasche, 2019-06-16:
Just to understand: The "Pairing" section is per vendor or per device?
I actually would like to update this section of http://www.zigbee2mqtt.io/devices/324131092621 (which seems to be a duplicate of http://www.zigbee2mqtt.io/devices/9290012573A) because the procedure is different. Maybe i just don't understand the format?

Koenkk, 2019-06-17:
Sections can be per device or per vendor. 

In your case you should edit it like:
- Add `notModel: ['324131092621'],` between here
https://github.com/Koenkk/zigbee2mqtt.io/blob/8fe364abe2119be2c05991cb599d0b8a193a3af8/docgen/device_page_notes.js#L162
- Add a section like https://github.com/Koenkk/zigbee2mqtt.io/blob/8fe364abe2119be2c05991cb599d0b8a193a3af8/docgen/device_page_notes.js#L273 with `model: ['324131092621'],`

closed 2019-08-16 by Koenkk