Koenkk/zigbee2mqtt #1620: Sudden Issue with Philips Hue Motion Sensor (No converter available)

RIMperator, 2019-06-10:
My Philips Hue Motion sensor used to work perfectly. However suddenly I'm getting errors and both of them stop working completely after a while. The error messages look as following:
No converter available for '9290012607' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":168}}'
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
or
No converter available for '9290012607' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"Philips","modelId":"SML001","powerSource":3}}'
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
or
No converter available for '9290012607' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"dateCode":"20190219","swBuildId":"6.1.1.27575"}}'
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

No converter would indicate that the device would be unsupported, however it has been running for weeks or even months flawlessly before and it's on the list of supported devices. At first my theory was that because I've recently added a second one, that motion sensor maybe had a different sw versions since this one was right out of the box and never added to the Hue bridge. So I paired it with the Hue Bridge and there was no sw update available, so I paired it back to z2m. Obviously I also tried removing and re-adding both sensors, but this didn't make a difference either. 
Both stop working after a few hours and I'm uncertain why. The update of z2m and the fw happened more or less at the same time as I've added the second Hue motion sensor, but so far I haven't come across a similar reported issue with Hue Motion sensors. Currently I will leave the newer one away and see what happens, but the longer I think about it, the longer I believe it could be just a coincidence.
Anyone seen this before or has any ideas?

## Debug Info
zigbee2mqtt version: 1.3.1 commit: 39d6a8e
CC253X firmware version: 20190223

Koenkk, 2019-06-12:
Those no converter available messages can be ignored (they are not the cause of the problem and will not show up in the next zigbee2mqtt version (1.5)).

The latest stable firmware might resolve your issue: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin

RIMperator, 2019-06-14:
Oddly since I've not rejoined the second (newer) Hue motion sensor the existing one keeps on working. Any idea why that is? Would you still recommend the upgrade? (I've been following some other issue with latest firmware release)

Koenkk, 2019-06-14:
I think your issue is #1408 which has been fixed in the latest stable firmware.

RIMperator, 2019-06-15:
Thank you @koenkk. In this case I will test firmware 20190319 and report back.

ZzetT, 2019-08-19:
> Those no converter available messages can be ignored (they are not the cause of the problem and will not show up in the next zigbee2mqtt version (1.5)).
> 
> The latest stable firmware might resolve your issue: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin

I'm running zigbee2mqtt version 1.5.1 with coordinator firmware 20190608, but I still see those "no converter messages" sometimes. However, only the two messages with cid "genBasic" appear in the log but not the one for genPowerCfg.

Koenkk, 2019-08-21:
I've added the ignore converters for this should be fixed in the dev branch.

closed 2019-08-21 by Koenkk

mabosch74, 2020-04-13:
I've the same problem:
debug 2020-04-13 15:19:48: No converter available for '9290012607' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4107,"imageType":269,"fileVersion":1107323831}'
My System:
Hardware: Texas Instruments CC1352P-2
Version of Zigbee2Mqtt: 1.12.0
Coordinator version: 20191106

Koenkk, 2020-04-13:
@mabosch74 fixed in the latest dev branch (messages are not interesting and therefore can be ignored).