Koenkk/zigbee2mqtt #1621: how to use "networkScan"

msadeghz, 2019-06-11:
Hi friends,
I want to scan network to find the devices want to join to network (when permit join is false) and then zigbee2mqtt permits the specific device to join. how can I use the available nodejs methods to do that? can I use networkScan to fine IEEE address of the possible devices?

Koenkk, 2019-06-11:
I'm not sure if zigbee has such a feature.

msadeghz, 2019-06-12:
for example, "mi home" app of Xiaomi, when you want to add device, you can put hub in scan mode to check which devices around the hub want to join and click on the device name in device name list and then join procedure is done.
this is applicable when there are some devices near your hub which aren't yours and you don't want to add them to your hub but they are in joining request stage.

renedis, 2019-06-12:
@msadeghz As far as I know the Mi Home software will only show Bluetooth/WiFi devices when scanning. Not Zigbee devices.

msadeghz, 2019-06-12:
BTW, I have seen this property in other brands. Infact, the hub list devices are sending "association request" or "device interview". I want to do this.

closed 2019-08-18 by stale[bot]