Koenkk/zigbee2mqtt #1623: Meaning of CC2530 router reported state and status

stepanov1975, 2019-06-11:
I have CC2530 flashed with router firmware. I didn't find any information about state and status values. Can anyone explain to me following?
What is the meaning of "state" true or false? Which is good and how to change it?
What is the meaning of "status" online/offline? Which value is good and how to change it? If router powered on and reporting state and status, why would it report "offline" status?
Thank you.

mabi, 2019-06-11:
"state" represents the LED state, see #1418

Koenkk, 2019-06-11:
In the next zigbee2mqtt version (1.5) state will represent the online state, for zigbee2mqtt 1.4 is represents the led state as mentioned by @mabi.

closed 2019-06-11 by Koenkk