Koenkk/zigbee2mqtt #1624: Z-stack

mquinn123, 2019-06-12:
Hi

Is there any alternative to IAR Embedded workbench for programming Z-stack for the CC2531/CC2530?  The price tag is high for small developers.  I can't find another compiler that would work.

With the current coordinator and router firmwares for the CC2531 and CC2530 is there any registry changes to the TXpower of the devices or are they kept as default?  

I was hoping to create firmware for changing the output power, frequency and channel, and also programming the I/O of the CC2530.  

Also these firmware are offered on a open licence, so you cant pretty much do what you please with the firmware?

My last question is:  If you set permit join to "no", and have a new encryption network key, do you regards Z-stack zigbee network to be secure?

Regards

Koenkk, 2019-06-12:
Is there any alternative to IAR Embedded workbench for programming Z-stack for the CC2531/CC2530? The price tag is high for small developers. I can't find another compiler that would work.
-> No, but for the CC2652R/CC1352P the free code composer studio can be used

With the current coordinator and router firmwares for the CC2531 and CC2530 is there any registry changes to the TXpower of the devices or are they kept as default?
@ptvoinfo can you answer this one?

Also these firmware are offered on a open licence, so you cant pretty much do what you please with the firmware?
Yes

My last question is: If you set permit join to "no", and have a new encryption network key, do you regards Z-stack zigbee network to be secure?
Yes

ptvoinfo, 2019-06-12:
@mquinn123 
Tx power: 
The firmware for standard CC2530 & CC2531 boards does not change this parameter.
The firmware for CC2530 & CC2531 with a power amplifier (CC2590, CC2591, CC2592, etc) sets the maximum Tx power for these amplifiers.

mquinn123, 2019-06-12:
Thanks for the quick replies.  This is an awesome project, great work.

closed 2019-06-12 by Koenkk