Koenkk/zigbee2mqtt #1625: [question] meaning of 'devChange' ?

suprimex, 2019-06-12:
Hello Gentlemen,
I have succeeded with setup of 6 Xiaomi different sensors/buttons/etc,  all of them working fine. 
But I noticed that in the log along with a normal payload like /...Received zigbee message of type 'attReport' with data..../  also often appears lines with /Received zigbee message of type 'devChange' with data.../    

What does it mean?  Should I just ignore them?

`
6/12/2019, 12:14:28 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,null,null,24]}}' of device 'lumi.sensor_cube' (0x00158d00029bba90) of endpoint 1

6/12/2019, 12:32:35 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65282":[null,null,null,{"elmVal":[0]},null,{"elmVal":86}]}}' of device 'lumi.sensor_magnet' (0x00158d0002b70c17) of endpoint 1

6/12/2019, 12:45:37 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3045,null,null,17320,28,[0,1],null,null,null,0]}}' of device 'lumi.sensor_switch.aq2' (0x00158d000201722a) of endpoint 1
`

roygbiv856, 2019-06-12:
I'd REALLY love to know the meanings of all the messages in the logs. That'd be really informative. I can't tell you what devchange means, but I can say it's very common and doesnt indicate an error or something going wrong

Koenkk, 2019-06-12:
`devChange` is a generated event when a value of a device changes (e.g. due to an `attReport`)

closed 2019-06-12 by Koenkk