Koenkk/zigbee2mqtt #1627: power off and then on ZNP & Virtual box running HASSIO

AjitKIPL, 2019-06-13:
hi,
After power off and then on ZNP & virtual box running HASSIO zigbee2mqtt addon, router device not able to operate from hassio. 
Is Router device not join the network after power off and on?
or 
is CC2652R ZNP firmware have an issue?

Can you please guide me to find and resolve the issue.

AjitKIPL, 2019-06-13:
hi,
Is CC2652R ZNP increment panid by 1 after power cycle(off and then on) when router device is always on? if yes then let me know how to disable this.

Koenkk, 2019-06-13:
CC2652R is known not to be stable yet, but the router should rejoin, what exact issues are you seeing?

The panid shouldn't be incremented (you can check if this happens in the log).

AjitKIPL, 2019-06-14:
hi,
when znp connected to hassio power off and then on, and also restart zigbee2mqtt addon. At that time router device is on. Panid in the log is shown the same. 
But router device not operate from hassio dashboard/lovelace.

below log in zigbee2mqtt addon when switch off button press-
zigbee2mqtt:info 6/14/2019, 9:42:10 AM Zigbee publish to device '0x00124b0018a67de7', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 4
2019-06-14T04:12:10.960Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 2
2019-06-14T04:12:10.990Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2019-06-14T04:12:11.600Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: {"status":205,"endpoint":1,"transid":2}
2019-06-14T04:12:11.601Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 2
  zigbee2mqtt:error 6/14/2019, 9:42:11 AM Zigbee publish to device '0x00124b0018a67de7', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 4 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 6/14/2019, 9:42:11 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x00124b0018a67de7","type":"device","friendlyName":"0x00124b0018a67de7"},"message":"OFF"}}'
2019-06-14T04:12:16.653Z zigbee-shepherd-converters:devices Configured '(cb) => ep1.bind('genOnOff', coordinator, cb)' with result 'Error: Timed out after 10000 ms'

also checked- 
Power off router device and then znp power off & then on, restart zigbee2mqtt addon.
after zigbee2mqtt restart power on router device, and it works fine.

problem observed only when znp power cycle.

Koenkk, 2019-06-14:
Does this issue fix itself after some time? When the znp is power cycled it needs to rebuild the routes.

AjitKIPL, 2019-06-14:
hi,
The issue is not fixed itself after some time. I checked it for 30mins.

Koenkk, 2019-06-15:
Strange, let's continue here: #1429

closed 2019-06-15 by Koenkk