Koenkk/zigbee2mqtt #1631: Not trying to pair device - Spamming Error: Timed out after 10000 ms

scuppasteve, 2019-06-14:
https://hastebin.com/mihoralixi.cs

# Bug Report
## What happened
Cannot get the Node Descriptor of the Device: 0xc8d2c10000004121 (Error: Timed out after 10000 ms)

## What did you expect to happen
For some reason on a new install it just spams this. I do not have a device with that info configured so i don't know, but basically whatever it is as far as i can tell is preventing everything else from pairing, i have no idea what it is that is doing it.

## How to reproduce it (minimal and precise)
Happens the second it boots when pairing is enabled.

## Debug Info
zigbee2mqtt version: 1.4
CC253X firmware version: CC2531ZNP-Prod.hex

Koenkk, 2019-06-15:
Which device is it?

scuppasteve, 2019-06-15:
Thats my point, i have no idea what it is. I have went and systematically unplugged Home Automation devices and i cant seem to figure it out. Is there a way to block that device, whatever it is, and maybe once i have got all my other zigbee items working i might be able to figure it out. But, there should be no powered zigbee devices available to spam connection like this. I would figure battery powered devices would not do this by itself as i think it would kill batteries.

Koenkk, 2019-06-15:
You can try to `ban` it (http://www.zigbee2mqtt.io/configuration/configuration.html)

scuppasteve, 2019-06-15:
So i have it banned, but it still shows up every time i enable pairing.

Koenkk, 2019-06-16:
I see, preventing this needs some changes in the underlying zigbee libraries, perhaps it might be easier to find out what device it is (it could also be a battery powered device).

closed 2019-08-22 by stale[bot]