Koenkk/zigbee2mqtt #1632: Aqara WXKG11LM wireless switch: missing infos on click in mqtt

leofalp, 2019-06-14:
# Bug Report
## What happened
When i click the switch the mqtt message report the battery percentage, link quality and voltage.
## What did you expect to happen
i need to see  the click action, sometingh like single,double, hold etc..

## How to reproduce it (minimal and precise)
All installed on Hass.io on ubuntu server with a CC2531 coordinator and Aqara WXKG11LM wireless switch.

## Debug Info
zigbee2mqtt version: zigbee2mqtt-1.4.0
CC253X firmware version: 20190425

I already tried to delete and rejoin the sensor, but nothing happend.

debug output:
```
  zigbee2mqtt:debug 6/14/2019, 11:13:49 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3035,"3":23,"4":5032,"5":21353,"6":[0,0],"10":0},"modelId":"lumi.sensor_switch.aq2"}}' of device 'lumi.sensor_switch.aq2' (0x00158d00023dc252) of endpoint 1
  zigbee2mqtt:info 6/14/2019, 11:13:49 AM MQTT publish: topic 'zigbee2mqtt/0x00158d00023dc252', payload '{"battery":100,"voltage":3035,"linkquality":73}'
  zigbee2mqtt:debug 6/14/2019, 11:13:49 AM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3035,null,23,5032,21353,[0,0],null,null,null,0],"modelId":"lumi.sensor_switch.aq2"}}' of device 'lumi.sensor_switch.aq2' (0x00158d00023dc252) of endpoint 1
```

Koenkk, 2019-06-14:
Looks the same as https://github.com/Koenkk/zigbee2mqtt/issues/376#issuecomment-461919228

leofalp, 2019-06-14:
So it is an hardware problem of the switch?

Koenkk, 2019-06-14:
Hardware or software problem of the switch. (Which cannot be fixed it seems, even not via the original xiaomi gateway).

leofalp, 2019-06-14:
Thank you so much!

closed 2019-06-14 by leofalp

DoraDeu, 2019-06-22:
Koenkk, I have exactly the same problem. Just battery percentage, link quality and voltage are shown but no button action.
For testing I linked the switch to the Mi-Gateway and there the click and double-click action works properly. Just the long-click cannot be realized. So I guess it can´t be hardware issue on my side.

UlrichFrank, 2019-06-22:
Same for me with 2 of 5 wireless switches.

Koenkk, 2019-06-23:
Please post debug logs if you have issues (when pressing the button) (`log_level: debug` http://www.zigbee2mqtt.io/configuration/configuration.html)

Luigi8723, 2019-11-25:
Same problem, Same switch:

{"friendly_name":"0x00158d000213252f","model":"WXKG11LM","vendor":"Xiaomi","description":"Aqara wireless switch","supported":true}}'
Nov 23 14:53:05 raspberrypi npm[7741]: zigbee2mqtt:info 2019-11-23T13:53:05: MQTT publish: topic 'zigbee2mqtt/0x00158d000213252f', payload '{"battery":100,"voltage":3055,"linkquality":114}'
Nov 23 14:53:23 raspberrypi npm[7741]: zigbee2mqtt:info 2019-11-23T13:53:23: MQTT publish: topic 'zigbee2mqtt/0x00158d000213252f', payload '{"battery":100,"voltage":3055,"linkquality":120}'
Nov 23 14:53:31 raspberrypi npm[7741]: zigbee2mqtt:info 2019-11-23T13:53:31: MQTT publish: topic 'zigbee2mqtt/0x00158d000213252f', payload '{"battery":100,"voltage":3055,"linkquality":99}'

No payload like single,double, hold etc. only Batt/ Volt/ Linkquality

How to solve this? i Would like to use the button.
Reset and reconnect??