Koenkk/zigbee2mqtt #1634: How to support new devices.

oemsys, 2019-06-14:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:1.4.0
CC253X firmware version:CC2530

HELP me 

How to support new devices.

[ABOX_sw_03.zip](https://github.com/Koenkk/zigbee2mqtt/files/3292314/ABOX_sw_03.zip)

JLFN, 2019-06-15:
paste logs online....

oemsys, 2019-06-15:
HI~~~~~

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1b311","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1560635110,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"9eyd675Fl0hALGyu"}

aktentasche, 2019-06-16:
I am sorry but you do not describe your problem, that's what the headers in the first post are for. We cannot help you. Also, no one will dig through your .zip file with at least a minimum amout of information. Your homework.

oemsys, 2019-06-25:
6/25/2019, 9:11:25 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00124b000904df4c"'
6/25/2019, 9:11:25 PM - info: Connecting with device...
6/25/2019, 9:11:25 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
6/25/2019, 9:11:25 PM - debug: Accept device incoming with ieeeAddr '0x00124b000904df4c' permit join is 'true'
6/25/2019, 9:11:25 PM - debug: Allowing device '0x00124b000904df4c' to join
6/25/2019, 9:11:25 PM - debug: Received zigbee message of type 'devIncoming' with data '"0x00124b000904df4c"' of device 'undefined' (0x00124b000904df4c) of endpoint 14
6/25/2019, 9:11:25 PM - info: Device incoming...
6/25/2019, 9:11:25 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
6/25/2019, 9:11:25 PM - info: New device 'undefined' with address 0x00124b000904df4c connected!
6/25/2019, 9:11:25 PM - debug: Saving state to file /app/data/state.json
6/25/2019, 9:11:25 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00124b000904df4c","meta":{}}'
6/25/2019, 9:11:25 PM - warn: Device with modelID 'undefined' is not supported.
6/25/2019, 9:11:25 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
6/25/2019, 9:11:25 PM - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'undefined' (0x00124b000904df4c) of endpoint 14
6/25/2019, 9:11:25 PM - warn: Device with modelID 'undefined' is not supported.
6/25/2019, 9:11:25 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
6/25/2019, 9:11:47 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x00124b000904df4c"' of device 'undefined' (0x00124b000904df4c) of endpoint 14
6/25/2019, 9:11:47 PM - warn: Device with modelID 'undefined' is not supported.
6/25/2019, 9:11:47 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

 
&mdash;

oemsys, 2019-07-30:
zigbee2mqtt:debug 7/30/2019, 9:31:45 PM Configuring 0x00124b0010013c59 (0x00124b0010013c59) ...                                                                                                           
  zigbee2mqtt:warn 7/30/2019, 9:31:45 PM Failed to configure 0x00124b0010013c59 (0x00124b0010013c5                                                                                                          
9) ('No device') (attempt #1)                                                                                                                                                                               
  zigbee2mqtt:warn 7/30/2019, 9:31:45 PM This can be ignored if the device is working properly                                                                                                              
  zigbee2mqtt:info 7/30/2019, 9:31:45 PM zigbee-shepherd ready                                                                                                                                              
  zigbee2mqtt:info 7/30/2019, 9:31:45 PM Connected to MQTT server                                                                                                                                           
  zigbee2mqtt:info 7/30/2019, 9:31:45 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '                                                                                                          
online'                                                                                                                                                                                                     
  zigbee2mqtt:info 7/30/2019, 9:31:45 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00010d0e09', pay                                                                                                          
load '{"temperature":33.7,"linkquality":15,"humidity":56.04}'                                                                                                                                               
  zigbee2mqtt:info 7/30/2019, 9:31:45 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload                                                                                                           
'{"version":"1.4.0","commit":"unknown","coordinator":20190223,"log_level":"debug","permit_join":tr                                                                                                          
ue}'                                                                                                                                                                                                        
  zigbee2mqtt:debug 7/30/2019, 9:31:45 PM Mounted the cieApp (epId 11)                                                                                                                                      
  zigbee2mqtt:debug 7/30/2019, 9:32:28 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1                                                                      
":2935,"4":5032,"5":121,"6":[0,10],"10":0,"100":3344,"101":5934}}}' of device 'lumi.sens' (0x00158d00010d0e09) of endpoint 1                                                                                
  zigbee2mqtt:info 7/30/2019, 9:32:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00010d0e09', payload '{"temperature":33.44,"linkqual                                                                      
ity":26,"humidity":59.34,"battery":62,"voltage":2935}'                                                                                                                                                      
  zigbee2mqtt:debug 7/30/2019, 9:32:28 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[nu                                                                      
ll,2935,null,null,5032,121,[0,10],null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,nul                                                                      
l,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,nu                                                                      
ll,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,n                                                                      
ull,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,3344,5934]}}' of device 'lumi.sens' (0x0                                                                      
0158d00010d0e09) of endpoint 1                                                                                                                                                                              
  zigbee2mqtt:debug 7/30/2019, 9:32:51 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00124b0010013c59"' of device                                                                       
'abox.sens' (0x00124b0010013c59) of endpoint 14                                                                                                                                                             
[... cut ...]
6/25/2019, 9:11:25 PM - warn: Device with modelID 'undefined' is not supported.
6/25/2019, 9:11:25 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
6/25/2019, 9:11:47 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x00124b000904df4c"' of device 'undefined' (0x00124b000904df4c) of endpoint 14
6/25/2019, 9:11:47 PM - warn: Device with modelID 'undefined' is not supported.
6/25/2019, 9:11:47 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

closed 2019-10-05 by stale[bot]